robotology/icub-main #83: dataSetPlayer unpredictably crashes with segmentation fault

arrenglover, 2015-01-16:
The segmentation fault occurs after selecting the dataset folder to load into the dataSetPlayer. The fault is unpredictable as sometimes the same actions result in successful loading of the dataset. The dataset was loading consistently with a iCub-main build from around November last year and the error has only been occurring since updating from the repository on the 15/1/2015. I have attempted loading the dataset from different folder levels above the .log file without finding a solution that works consistently.

The dataset I am using is [here](https://dl.dropboxusercontent.com/u/6582156/eBottle_objects_no14.zip)

I have the following output upon trying to load the data at 3 different folder levels, the fifth attempt is successful:
```
aglover@iiticublap031:~$ dataSetPlayer 
||| clearing context
||| adding context [dataSetPlayer]
||| configuring
||| no policy found
||| default config file specified as config.ini
||| checking [/home/aglover/config.ini] (pwd)
||| checking [/home/aglover/.config/yarp/robots/default] (robot YARP_CONFIG_HOME)
||| checking [/home/aglover/.local/share/yarp/robots/default] (robot YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/usr/share/ubuntu/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/ubuntu/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| found /usr/local/share/yarp/config/path.d
||| checking [/usr/local/share/ICUBcontrib/robots/default] (robot yarp.d)
||| checking [/usr/local/share/iCub/robots/default] (robot yarp.d)
||| checking [/home/aglover/.config/yarp/contexts/dataSetPlayer] (context YARP_CONFIG_HOME)
||| checking [/home/aglover/.local/share/yarp/contexts/dataSetPlayer] (context YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/contexts/dataSetPlayer] (context YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/contexts/dataSetPlayer] (context YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/contexts/dataSetPlayer] (context YARP_CONFIG_DIRS)
||| checking [/usr/share/ubuntu/yarp/contexts/dataSetPlayer] (context YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/contexts/dataSetPlayer] (context YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/contexts/dataSetPlayer] (context YARP_DATA_DIRS)
||| checking [/usr/share/yarp/contexts/dataSetPlayer] (context YARP_DATA_DIRS)
||| checking [/usr/local/share/ICUBcontrib/contexts/dataSetPlayer] (context yarp.d)
||| checking [/usr/local/share/iCub/contexts/dataSetPlayer] (context yarp.d)
||| found /usr/local/share/iCub/contexts/dataSetPlayer
||| checking [/usr/local/share/iCub/contexts/dataSetPlayer/config.ini] (context)
||| found /usr/local/share/iCub/contexts/dataSetPlayer/config.ini
yarp: Port /dataSetPlayer/rpc:i active at tcp://127.0.0.1:10002
window should be now visible
Segmentation fault (core dumped)
aglover@iiticublap031:~$ dataSetPlayer 
||| clearing context
||| adding context [dataSetPlayer]
||| configuring
||| no policy found
||| default config file specified as config.ini
||| checking [/home/aglover/config.ini] (pwd)
||| checking [/home/aglover/.config/yarp/robots/default] (robot YARP_CONFIG_HOME)
||| checking [/home/aglover/.local/share/yarp/robots/default] (robot YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/usr/share/ubuntu/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/ubuntu/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| found /usr/local/share/yarp/config/path.d
||| checking [/usr/local/share/ICUBcontrib/robots/default] (robot yarp.d)
||| checking [/usr/local/share/iCub/robots/default] (robot yarp.d)
||| checking [/home/aglover/.config/yarp/contexts/dataSetPlayer] (context YARP_CONFIG_HOME)
||| checking [/home/aglover/.local/share/yarp/contexts/dataSetPlayer] (context YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/contexts/dataSetPlayer] (context YARP_CONFIG_DIRS)
[... 119 lines cut ...]
||| checking [/usr/share/ubuntu/yarp/contexts/dataSetPlayer] (context YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/contexts/dataSetPlayer] (context YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/contexts/dataSetPlayer] (context YARP_DATA_DIRS)
||| checking [/usr/share/yarp/contexts/dataSetPlayer] (context YARP_DATA_DIRS)
||| checking [/usr/local/share/ICUBcontrib/contexts/dataSetPlayer] (context yarp.d)
||| checking [/usr/local/share/iCub/contexts/dataSetPlayer] (context yarp.d)
||| found /usr/local/share/iCub/contexts/dataSetPlayer
||| checking [/usr/local/share/iCub/contexts/dataSetPlayer/config.ini] (context)
||| found /usr/local/share/iCub/contexts/dataSetPlayer/config.ini
yarp: Port /dataSetPlayer/rpc:i active at tcp://127.0.0.1:10002
window should be now visible
the full path is /home/aglover/workspace/datasets/eBottle_objects_no14 
The size of the file is 3 
The size of the file is 3 
 /home/aglover/workspace/datasets/eBottle_objects_no14/pliershand_small/eBottles/info.log IS present adding it to the gui
the size of subDirs is: 1
opening file /home/aglover/workspace/datasets/eBottle_objects_no14/pliershand_small/eBottles/info.log
the name of the port is /cbottletoebottle/cte:o
opening file /home/aglover/workspace/datasets/eBottle_objects_no14/pliershand_small/eBottles/data.log
1417017145.917771
the biggest timestamp is: index 0 with value 1417017145.917771
need to create new port Bottle for pliershand_small_eBottles
creating and opening Bottle port for part pliershand_small_eBottles
yarp: Port /cbottletoebottle/cte:o active at tcp://127.0.0.1:10012
```

vtikha, 2015-01-16:
Ciao Arren, 
thank you for reporting. I have downloaded your dataset in order to do the various tests and will get back to you as soon as I have something. 
Ciao

vtikha, 2015-01-16:
Ciao Arren,

I have tested this and was able to reproduce it after some tries. I have just quickly committed a small patch to avoid this behaviour (as I am about to board on a plane in 5 minutes). I will check it out during the flight or the weekend for a nicer solution.
In the meantime please let me know if you still get the unpredictable crashes.

pattacini, 2015-01-16:
Extreme debugging :)

lornat75, 2015-01-19:
Don't forge to bring this fix in the qt version of the datasetplayer otherwiese we will have to debug it again in the future :)

arrenglover, 2015-01-20:
Ciao Vadim,

It's now opening the dataset correctly. Even better, when I purposefully select the wrong folder it's giving an error rather than crashing. Thanks and closed.

closed 2015-01-20 by arrenglover