robotology/icub-main #89: Error during compilation

robertocalandra, 2015-01-21:
When compiling icub-main (trought the codyco-superbuild) there is a compiling error.

[  7%] Performing build step for 'ICUB'
[  1%] Copying manager.py to /home/calandra/Software/codyco-superbuild/build/external/ICUB/bin/.
[  1%] Built target python-scripts
[  1%] Copying icub-cluster scripts to /home/calandra/Software/codyco-superbuild/build/external/ICUB/bin/.
[  1%] Built target cluster-scripts
Scanning dependencies of target iCubIDLClients
[  1%] Building CXX object src/libraries/iCubIDLClients/CMakeFiles/iCubIDLClients.dir/__/__/idl_generated_code/src/fingersTuner_IDL.cpp.o
[  2%] Building CXX object src/libraries/iCubIDLClients/CMakeFiles/iCubIDLClients.dir/__/__/idl_generated_code/src/depth2kin_IDL.cpp.o
[  2%] Building CXX object src/libraries/iCubIDLClients/CMakeFiles/iCubIDLClients.dir/__/__/idl_generated_code/src/PointReq.cpp.o
[  3%] Building CXX object src/libraries/iCubIDLClients/CMakeFiles/iCubIDLClients.dir/__/__/idl_generated_code/src/dataSetPlayer_IDL.cpp.o
Linking CXX static library ../../../lib/libiCubIDLClients.a
[  3%] Built target iCubIDLClients
Scanning dependencies of target iCubDev
[  3%] Building CXX object src/libraries/iCubDev/CMakeFiles/iCubDev.dir/src/LogpolarInterfaces.cpp.o
Linking CXX static library ../../../lib/libiCubDev.a
[  4%] Built target iCubDev
Scanning dependencies of target iCubInterfaceGuiLib
[  4%] Building CXX object src/libraries/iCubInterfaceGuiLib/CMakeFiles/iCubInterfaceGuiLib.dir/iCubBoardChannel.cpp.o
[  4%] Building CXX object src/libraries/iCubInterfaceGuiLib/CMakeFiles/iCubInterfaceGuiLib.dir/iCubBoard.cpp.o
[  5%] Building CXX object src/libraries/iCubInterfaceGuiLib/CMakeFiles/iCubInterfaceGuiLib.dir/iCubNetwork.cpp.o
[  5%] Building CXX object src/libraries/iCubInterfaceGuiLib/CMakeFiles/iCubInterfaceGuiLib.dir/iCubInterfaceGuiServer.cpp.o
Linking CXX static library ../../../lib/libiCubInterfaceGuiLib.a
[  5%] Built target iCubInterfaceGuiLib
Scanning dependencies of target logpolar
[  6%] Building CXX object src/libraries/logpolar/CMakeFiles/logpolar.dir/src/RC_DIST_FB_logpolar_mapper.cpp.o
Linking CXX static library ../../../lib/liblogpolar.a
[  6%] Built target logpolar
Scanning dependencies of target boostMIL
[  6%] Building CXX object src/libraries/boostMIL/CMakeFiles/boostMIL.dir/src/MILClassifier.cpp.o
[  7%] Building CXX object src/libraries/boostMIL/CMakeFiles/boostMIL.dir/src/SelectorClassifier.cpp.o
[  7%] Building CXX object src/libraries/boostMIL/CMakeFiles/boostMIL.dir/src/StrongClassifier.cpp.o
[  8%] Building CXX object src/libraries/boostMIL/CMakeFiles/boostMIL.dir/src/OnlineBoost.cpp.o
[  8%] Building CXX object src/libraries/boostMIL/CMakeFiles/boostMIL.dir/src/OnlineSupport.cpp.o
[  8%] Building CXX object src/libraries/boostMIL/CMakeFiles/boostMIL.dir/src/AdaBoost.cpp.o
[  9%] Building CXX object src/libraries/boostMIL/CMakeFiles/boostMIL.dir/src/AvgClassifier.cpp.o
Linking CXX static library ../../../lib/libboostMIL.a
[  9%] Built target boostMIL
Scanning dependencies of target ctrlLib
[  9%] Building CXX object src/libraries/ctrlLib/CMakeFiles/ctrlLib.dir/src/math.cpp.o
[  9%] Building CXX object src/libraries/ctrlLib/CMakeFiles/ctrlLib.dir/src/filters.cpp.o
[ 10%] Building CXX object src/libraries/ctrlLib/CMakeFiles/ctrlLib.dir/src/kalman.cpp.o
[ 10%] Building CXX object src/libraries/ctrlLib/CMakeFiles/ctrlLib.dir/src/pids.cpp.o
[ 11%] Building CXX object src/libraries/ctrlLib/CMakeFiles/ctrlLib.dir/src/tuning.cpp.o
[ 11%] Building CXX object src/libraries/ctrlLib/CMakeFiles/ctrlLib.dir/src/adaptWinPolyEstimator.cpp.o
[ 11%] Building CXX object src/libraries/ctrlLib/CMakeFiles/ctrlLib.dir/src/minJerkCtrl.cpp.o
[ 12%] Building CXX object src/libraries/ctrlLib/CMakeFiles/ctrlLib.dir/src/functionEncoder.cpp.o
[ 12%] Building CXX object src/libraries/ctrlLib/CMakeFiles/ctrlLib.dir/src/optimalControl.cpp.o
[ 13%] Building CXX object src/libraries/ctrlLib/CMakeFiles/ctrlLib.dir/src/neuralNetworks.cpp.o
[ 13%] Building CXX object src/libraries/ctrlLib/CMakeFiles/ctrlLib.dir/src/outliersDetection.cpp.o
Linking CXX static library ../../../lib/libctrlLib.a
[ 13%] Built target ctrlLib
Scanning dependencies of target skinDynLib
[ 14%] Building CXX object src/libraries/skinDynLib/CMakeFiles/skinDynLib.dir/src/skinContact.cpp.o
[ 14%] Building CXX object src/libraries/skinDynLib/CMakeFiles/skinDynLib.dir/src/skinContactList.cpp.o
[ 14%] Building CXX object src/libraries/skinDynLib/CMakeFiles/skinDynLib.dir/src/dynContact.cpp.o
[ 15%] Building CXX object src/libraries/skinDynLib/CMakeFiles/skinDynLib.dir/src/dynContactList.cpp.o
[ 15%] Building CXX object src/libraries/skinDynLib/CMakeFiles/skinDynLib.dir/src/common.cpp.o
Linking CXX static library ../../../lib/libskinDynLib.a
[ 15%] Built target skinDynLib
Scanning dependencies of target iKin
[ 16%] Building CXX object src/libraries/iKin/CMakeFiles/iKin.dir/src/iKinFwd.cpp.o
[ 16%] Building CXX object src/libraries/iKin/CMakeFiles/iKin.dir/src/iKinInv.cpp.o
[ 17%] Building CXX object src/libraries/iKin/CMakeFiles/iKin.dir/src/iKinHlp.cpp.o
Linking CXX static library ../../../lib/libiKin.a
[ 17%] Built target iKin
Scanning dependencies of target iDyn
[ 18%] Building CXX object src/libraries/iDyn/CMakeFiles/iDyn.dir/src/iDyn.cpp.o
[ 18%] Building CXX object src/libraries/iDyn/CMakeFiles/iDyn.dir/src/iDynInv.cpp.o
[... 168 lines cut ...]
     FilterModule() : gyroFilt(1,Vector(3,0.0)), magFilt(1,Vector(3,0.0)),
                      ^
/home/calandra/Software/codyco-superbuild/external/ICUB/src/tools/imuFilter/main.cpp:56:49: error: class ‘FilterModule’ does not have any field named ‘magFilt’
     FilterModule() : gyroFilt(1,Vector(3,0.0)), magFilt(1,Vector(3,0.0)),
                                                 ^
/home/calandra/Software/codyco-superbuild/external/ICUB/src/tools/imuFilter/main.cpp: In member function ‘virtual bool FilterModule::configure(yarp::os::ResourceFinder&)’:
/home/calandra/Software/codyco-superbuild/external/ICUB/src/tools/imuFilter/main.cpp:71:9: error: ‘gyroFilt’ was not declared in this scope
         gyroFilt.setOrder(gyro_order);
         ^
/home/calandra/Software/codyco-superbuild/external/ICUB/src/tools/imuFilter/main.cpp:72:9: error: ‘magFilt’ was not declared in this scope
         magFilt.setOrder(mag_order);
         ^
/home/calandra/Software/codyco-superbuild/external/ICUB/src/tools/imuFilter/main.cpp: In member function ‘virtual bool FilterModule::updateModule()’:
/home/calandra/Software/codyco-superbuild/external/ICUB/src/tools/imuFilter/main.cpp:106:21: error: ‘gyroFilt’ was not declared in this scope
         Vector gyro=gyroFilt.filt(imuData->subVector(6,8))-gyroBias;
                     ^
/home/calandra/Software/codyco-superbuild/external/ICUB/src/tools/imuFilter/main.cpp:107:20: error: ‘magFilt’ was not declared in this scope
         Vector mag=magFilt.filt(imuData->subVector(9,11));
                    ^
make[5]: *** [src/tools/imuFilter/CMakeFiles/imuFilter.dir/main.cpp.o] Error 1
make[4]: *** [src/tools/imuFilter/CMakeFiles/imuFilter.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [external/ICUB/CMakeFiles/YCMStamp/ICUB-build] Error 2
make[1]: *** [CMakeFiles/ICUB.dir/all] Error 2
make: *** [all] Error 2

traversaro, 2015-01-21:
Can you paste the content of the CMakeCache of your icub-main build?

robertocalandra, 2015-01-21:
# This is the CMakeCache file.
# For build in directory: /home/calandra/Software/codyco-superbuild/build/external/ICUB
# It was generated by CMake: /usr/bin/cmake
# You can edit this file to change values found and used by cmake.
# If you do not want to change any of the values, simply exit the editor.
# If you do want to change a value, simply edit, save, and exit the editor.
# The syntax for the file is as follows:
# KEY:TYPE=VALUE
# KEY is the name of a variable in the cache.
# TYPE is a hint to GUIs for the type of VALUE, DO NOT EDIT TYPE!.
# VALUE is the current value for the KEY.

########################
# EXTERNAL cache entries
########################

//Path to a library.
ACE_ACE_LIBRARY_DEBUG:FILEPATH=ACE_ACE_LIBRARY_DEBUG-NOTFOUND

//Path to a library.
ACE_ACE_LIBRARY_RELEASE:FILEPATH=/usr/lib/libACE.so

//Additional compiler flags for ACE
ACE_DEFINITIONS:STRING=

//Path to a library.
ACE_DL_LIBRARY_DEBUG:FILEPATH=ACE_DL_LIBRARY_DEBUG-NOTFOUND

//Path to a library.
ACE_DL_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libdl.so

//ACE include directory
ACE_INCLUDE_DIRS:PATH=

//Path to a library.
ACE_RT_LIBRARY_DEBUG:FILEPATH=ACE_RT_LIBRARY_DEBUG-NOTFOUND

//Path to a library.
ACE_RT_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/librt.so

//Allow YARP to (re)build IDL files as needed
ALLOW_IDL_GENERATION:BOOL=OFF

//No help, variable specified on the command line.
BUILD_SHARED_LIBS:BOOL=

//Path to a program.
CMAKE_AR:FILEPATH=/usr/bin/ar

//Choose the type of build, recommanded options are: Debug or Release
CMAKE_BUILD_TYPE:STRING=Release

//Enable/Disable color output during build.
CMAKE_COLOR_MAKEFILE:BOOL=ON

//CXX compiler.
CMAKE_CXX_COMPILER:FILEPATH=/usr/bin/c++

//Flags used by the compiler during all build types.
CMAKE_CXX_FLAGS:STRING=

//Flags used by the compiler during debug builds.
CMAKE_CXX_FLAGS_DEBUG:STRING=-g

//Flags used by the compiler during release minsize builds.
CMAKE_CXX_FLAGS_MINSIZEREL:STRING=-Os -DNDEBUG

//Flags used by the compiler during release builds (/MD /Ob1 /Oi
// /Ot /Oy /Gs will produce slightly less optimized but smaller
// files).
[... 3123 lines cut ...]
_PC_ACE_LIBDIR:INTERNAL=/usr/lib
_PC_ACE_LIBRARIES:INTERNAL=ACE;dl;rt
_PC_ACE_LIBRARY_DIRS:INTERNAL=
_PC_ACE_LIBS:INTERNAL=
_PC_ACE_LIBS_L:INTERNAL=
_PC_ACE_LIBS_OTHER:INTERNAL=
_PC_ACE_LIBS_PATHS:INTERNAL=
_PC_ACE_PREFIX:INTERNAL=/usr
_PC_ACE_STATIC_CFLAGS:INTERNAL=
_PC_ACE_STATIC_CFLAGS_I:INTERNAL=
_PC_ACE_STATIC_CFLAGS_OTHER:INTERNAL=
_PC_ACE_STATIC_INCLUDE_DIRS:INTERNAL=
_PC_ACE_STATIC_LDFLAGS:INTERNAL=-lACE;-ldl;-lrt
_PC_ACE_STATIC_LDFLAGS_OTHER:INTERNAL=
_PC_ACE_STATIC_LIBDIR:INTERNAL=
_PC_ACE_STATIC_LIBRARIES:INTERNAL=ACE;dl;rt
_PC_ACE_STATIC_LIBRARY_DIRS:INTERNAL=
_PC_ACE_STATIC_LIBS:INTERNAL=
_PC_ACE_STATIC_LIBS_L:INTERNAL=
_PC_ACE_STATIC_LIBS_OTHER:INTERNAL=
_PC_ACE_STATIC_LIBS_PATHS:INTERNAL=
_PC_ACE_VERSION:INTERNAL=6.0.3
__pkg_config_checked__PC_ACE:INTERNAL=1
//ADVANCED property for variable: board_tranceiver
board_tranceiver-ADVANCED:INTERNAL=1

traversaro, 2015-01-23:
Mhh sorry for the delay, I was busy with a deadline. 
You eventually solved? I have the impression that there was some inconsistency in the repo... 
Have you tried with `git pull && make update-all` ?

traversaro, 2015-01-23:
Anyway consider that here in IIT even for "codyco" setup we prefer to install `yarp` and `icub-main` "separately" from the superbuild (as mentioned in the [codyco-superbuild readme](https://github.com/robotology/codyco-superbuild/blob/master/README.md) ) for consitency with the usual way of installing icub software.

hu-yue, 2015-03-12:
@robertocalandra you solved the problem? Can we close the issue?

robertocalandra, 2015-03-12:
I did not solved the problem. But eventually with another fresh install, it disappeared.

closed 2015-03-12 by robertocalandra