robotology/icub-main #92: Problem with movement of the real iCub's head

Arikae, 2015-01-26:
Hi,

we've got a problem with the real iCub's head when accessing the joints.
In our project we are working with the Oculus Rift and want to use the headtracking information to mirror it to the iCub but he is only partially doing what we want him to.
In the first few seconds it seems that he is moving like the Oculus but then the movement slows down or even stops completly. 
By further random turning of the Oculus the iCub sometimes can shortly follow it but then it always slows down again.
We have plotted the calculated angles that we send to the robot (which we have limited too only those that are accessable by the real one)  and everything seems to be fine and the simulation within gazebo is running smoothly.
First we thought that the update rate may be too high with which we update the joints because the yarp ports crashed but reducing them from 1kHz to even 25Hz did solve the yarpserver-crash but the problem from above still occured.

We're sending the data via IPositionControl and have not idea why this is happening.
Do we have to access the robot differently?

randaz81, 2015-01-26:
Can you please move this post to the following location?
https://github.com/robotology/QA/issues

closed 2015-01-26 by randaz81

barbalberto, 2015-01-26:
When using high rate like 1KHz, it is suggested to send commands through the IPositionDirect interface, using either the "all joints" call: 

setPositions(const double *refs);  // where ref is a vector of references for all the joints

or the "group joint", if you are not interested in controlling the eyes:
setPositions(const int n_joint, const int *joints, double *refs);  

When using posidionDirect, which goes through streaming port instead of RPC, read also this topic.

paulfitz, 2015-01-26:
Also if you have a chance, a quick description (cut/paste of console messages) of the yarp port crash would be very welcome at https://github.com/robotology/yarp/issues/new