robotology/icub-main #93: Getting 3D skin data

MoritzN89, 2015-01-28:
I cant find data files with the positions and orientations from every sensor. I used google and searched in the icub contribution page but only get a page which refers to the icub 3D skin gui which points to such files (http://eris.liralab.it/iCub/contrib/dox/html/group__icub__iCubSkinGUI3D.html), that I didnt found too (maybe because I cub 3d skin gui doesent exists anymore?).

traversaro, 2015-01-28:
The skinManager uses taxel positions from https://github.com/robotology/icub-main/tree/master/app/skinGui/conf/positions .

traversaro, 2015-01-29:
@MoritzN89 please consider that those points are expressed in the reference frames defined by the [iKin library](http://eris.liralab.it/iCub/main/dox/html/group__iKin.html). 

Do you plan to use them for something related to this gazebo-yarp-plugins issue: https://github.com/robotology/gazebo-yarp-plugins/issues/168 ? 
In that case I suggest that you clearly state what you want to implement and the amount of the work that you can commit on it, so we can provide all the support needed.

MoritzN89, 2015-01-31:
I want to improve the icub.sdf file by adding cubes in the body-parts representing the several sensors. Every sensor sends its collision data to yarp where we can evaluate them. Ive allready done to programming a script which creates the cubes but the <pose> coordinates are as expected not compatible with the one from the reference frame. I looked at the iKin page but dont find anything which can helpt to compute the provided skin data into the pose format in the sdf file.

pattacini, 2015-01-31:
iKin convention is described [here](http://wiki.icub.org/wiki/ICubForwardKinematics).

MoritzN89, 2015-02-01:
Something on the data must be wrong or they are not compatible with gazebo or something. In the image attached you can see the cubes as sensors. They are for the most part right placed on the skin and you can see a triangle pattern. Nevertheless it looks strange
[image]

https://www.dropbox.com/s/nean48k6suq19f5/icub.sdf?dl=0

 and a bit sensors are under the skin. I considerd the reference frame too.

traversaro, 2015-02-01:
I guess there are two main reasons behind of this discrepancies: 
    * The points where obtained through a calibration procedure ( http://pasa.liralab.it/pasapdf/947_DelPrete_etal2011.pdf ) on the real robot. Hence they can be corrupted by sensor noise.
  * The Gazebo models meshes where obtained from a third-party (UPMC University in Paris, see for more details https://github.com/robotology-playground/icub-model-generator ) so we cannot guaranteed anything about their correctness. We setted up recently an automatic procedure to extract URDF from CAD files ( https://www.icub.org/wiki/Creo_Mechanism_to_URDF ) but a complete model for iCub obtained through this procedure is still not available.

alecive, 2015-02-02:
@MoritzN89 furthermore, the skin data (point 1 listed by @traversaro ) is supposedly better on the forearm. Try to visualize them instead of the upper arm.

matejhof, 2015-02-02:
Hi there,
I cannot judge the accuracy of the Gazebo models of meshes. I will speak to the accuracy of the taxel positions in the positions files that are a result of the calibration using force/torque sensing peformed by Andrea Del Prete. 
I plotted some of them (forearm) in matlab and they indeed are not perfect triangles - the position with respect to the position if one would project the triangular modules whose 2D geometry is known differs - the error is perhaps up to 1 cm in each taxel's position. 
The other problem is that this data is not available for other body parts (torso, legs).

As an alternative to the calibration of the type Andrea performed (which becomes more difficult and error-prone for some of the body parts due to the positioning of force/torque sensors), one can use data available from the robot's CAD models. For the palm data, I used the coordinates I got from our mechanical guys to create the positions file - assuming the palm is 2D - that was easy. For the other skin parts that are 3D, the 2D geometry of individual skin patches that cover the robot is known. In addition, certain attachment points of these patches in the 3D CAD model of the robot are also known (midpoints of triangular modules - middle taxel of every triangle). Taken together, these data form constraints that can be used for an optimization algorithm to find the most likely 3D positions of every tactile sensor. However, this is quite laborious.

@MoritzN89 ,  can we ask more information about you and your project? Perhaps we could see how we could jointly proceed foreward in this direction.
Equipping the Gazebo simulator with the skin would also be our interest. 

cheers,
Matej Hoffmann

lornat75, 2015-02-03:
I agree with @metejhof. On the other hand if we distribute the same amount of taxels of the real robot on the surfaces of the covers we can still get a very useful simulation even if triangles are not precise... so I would prioritize getting a realistic software interface rather than striving to get exactly the same positioning as the real robot.

iron76, 2015-02-03:
@matejhof has a point when saying that the procedure in http://ieeexplore.ieee.org/xpls/abs_all.jsp?arnumber=6094896 is not accurate enough. I have playing with that procedure when calibrating both feet (see results in http://ieeexplore.ieee.org/xpls/abs_all.jsp?arnumber=6943124). It is quite an elaborate procedure with poor results. As pointed out again by @matejhof, we should extract as much data as possible from the CAD. Then adjustment (i.e. calibration) can be performed on this preliminary positioning, trying to be realistic on the variables to be optimised (e.g. rotation and displacement of the skin patches). 

In my opinion, this can be done in two different ways. 

* **manually** : volunteers?
* **automatically** : with @traversaro we are considering automatic procedures to extract URDF from CAD, including sensors. At present, we are only considering accelerometers, gyros and force/torque sensors. Skin is not a priority but we might include it as well.

alecive, 2015-02-03:
Regarding the *manually*, We (me and @matejhof ) talked with alberto parmiggiani and he said that for a skilled CAD user it would have these costs:
 * a couple of hours to retrieve the taxels positions for the center of each triangle (the others 9 taxels will still be left out)
 * a couple of days to retrieve their normals wrt the covers. This might seem not that useful at a first glance, but it may be useful (e.g. it is something without which my sw does not work :stuck_out_tongue_winking_eye:  )

pattacini, 2015-02-03:
Two days?! Oh my goodness! :)
Maybe, they don't know all the magic behind the software :)

alecive, 2015-02-03:
Probably :)

traversaro, 2015-02-03:
Let me add that there is no contrast between *manually* and *automatically* extracting this skin information. 
Even if we extract this information *manually*, adding appropriate reference frames to the model (or perhaps the simplified model, see https://www.icub.org/wiki/Creo_Mechanism_to_URDF for more information) would drastically simplify even automatic extractions of this information. For example, having the iKin link reference frames in the CAD model would simplify a lot sensor extraction.

matejhof, 2015-02-04:
I just re-stumbled over this abandoned iCubSkinGUI3D: http://eris.liralab.it/iCub/contrib/dox/html/group__icub__iCubSkinGUI3D.html by Eric Sauser from EPFL. Has anybody seen it working or knows how to operate it? Perhaps this is relevant (perhaps not).

matejhof, 2015-05-13:
While playing with the positions files we have from Andrea Del Prete's method for forearms and upper arms, I noticed some new facts:
1. They were done on an older version of iCub skin - the one with 12-taxel triangles (confirmed by Andrea). The triangles there had the same size, but the individual taxel arrangement was different. At least, the majority of triangles had the same orientation w.r.t. the new skin versions.
2. For the left vs. right files, the positions and orientations are the same  - just remapped into the different FoR orientation. So calibration was probably performed only for one half of the body.
3. The triangle midpoints should roughly match with the holes in the covers where they are attached. The coordinates of these are available in CAD. When plotted against the positions files, there are systematic offsets - e.g. for the small patch on top of the forearm, the CAD data has the triangle centres shifted cca 1 cm distally.

So, the positions files should be used with care - the taxels of each triangle are more like a scrambled point cloud that may also be systematically shifted (corresponding to the picture posted by @MoritzN89 above). So my estimate - assuming the CAD data is correct - is that individual taxels can be off by 1, sometimes up to 2 cm.
(this is not true for the palm that was extracted directly from CAD)

At the same time, we now have some data from the CAD and some knowledge on how to formulate an optimization problem that could be used to recalibrate the skin. So, **volunteer needed**! :)

pattacini, 2016-09-01:
Hi @matejhof 

After a year, there's been any progress on this?

traversaro, 2016-09-01:
Actually, yes. : D 
But we can open new issue to track the progress on this activity.

closed 2016-09-01 by traversaro

matejhof, 2016-09-02:
yes, in fact the progress is happening at this very moment with @traversaro and @fiorisi 
We will update this very soon.

matejhof, 2016-10-21:
New forearm V2 files coming out of the new pipeline using CAD  (thanks to @traversaro and @fiorisi) have been committed here: 7d81dedfbffa4e7cc780267076a5d8478c49cf85
The optimization framework thanks to @traversaro is here: https://github.com/robotology-playground/icub-model-generator