robotology/icub-main #95: New Qt5 modules compilation failure

pattacini, 2015-01-28:
A fresh build (cmake cache purged) generates the following errors on my Windows:

```
Microsoft (R) Microsoft Visual Studio 2012 Version 11.0.61030.0.
Copyright (C) Microsoft Corp. All rights reserved.
1>------ Build started: Project: cluster-scripts, Configuration: Release Win32 ------
1>  Copying icub-cluster scripts to C:/dev/icub-main/build/bin/Release
2>------ Build started: Project: frameGrabberGui2-qt, Configuration: Release Win32 ------
3>------ Build started: Project: QtICubSkinGuiPlugin, Configuration: Release Win32 ------
2>dc1394thread.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::RemoteFrameGrabberControlsDC1394(void)" (__imp_??0RemoteFrameGrabberControlsDC1394@dev@yarp@@QAE@XZ) referenced in function "protected: virtual void __thiscall DC1394Thread::run(void)" (?run@DC1394Thread@@MAEXXZ)
2>dc1394thread.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: virtual __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::~RemoteFrameGrabberControlsDC1394(void)" (__imp_??1RemoteFrameGrabberControlsDC1394@dev@yarp@@UAE@XZ) referenced in function "public: virtual void * __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::`scalar deleting destructor'(unsigned int)" (??_GRemoteFrameGrabberControlsDC1394@dev@yarp@@UAEPAXI@Z)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual bool __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::close(void)" (?close@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAE_NXZ)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual bool __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::getActiveDC1394(int)" (?getActiveDC1394@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAE_NH@Z)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual double __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::getBrightness(void)" (?getBrightness@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAENXZ)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual unsigned int __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::getBytesPerPacketDC1394(void)" (?getBytesPerPacketDC1394@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAEIXZ)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual unsigned int __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::getColorCodingDC1394(void)" (?getColorCodingDC1394@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAEIXZ)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual unsigned int __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::getColorCodingMaskDC1394(unsigned int)" (?getColorCodingMaskDC1394@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAEII@Z)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual double __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::getExposure(void)" (?getExposure@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAENXZ)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual unsigned int __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::getFPSDC1394(void)" (?getFPSDC1394@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAEIXZ)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual unsigned int __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::getFPSMaskDC1394(void)" (?getFPSMaskDC1394@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAEIXZ)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual double __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::getFeatureDC1394(int)" (?getFeatureDC1394@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAENH@Z)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual bool __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::getFormat7MaxWindowDC1394(unsigned int &,unsigned int &,unsigned int &,unsigned int &,unsigned int &,unsigned int &)" (?getFormat7MaxWindowDC1394@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAE_NAAI00000@Z)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual bool __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::getFormat7WindowDC1394(unsigned int &,unsigned int &,int &,int &)" (?getFormat7WindowDC1394@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAE_NAAI0AAH1@Z)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual double __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::getGain(void)" (?getGain@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAENXZ)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual double __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::getGamma(void)" (?getGamma@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAENXZ)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual double __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::getHue(void)" (?getHue@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAENXZ)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual unsigned int __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::getISOSpeedDC1394(void)" (?getISOSpeedDC1394@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAEIXZ)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual double __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::getIris(void)" (?getIris@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAENXZ)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual bool __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::getModeDC1394(int)" (?getModeDC1394@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAE_NH@Z)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual bool __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::getOperationModeDC1394(void)" (?getOperationModeDC1394@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAE_NXZ)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual double __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::getSaturation(void)" (?getSaturation@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAENXZ)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual double __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::getSharpness(void)" (?getSharpness@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAENXZ)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual double __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::getShutter(void)" (?getShutter@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAENXZ)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual bool __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::getTransmissionDC1394(void)" (?getTransmissionDC1394@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAE_NXZ)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual unsigned int __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::getVideoModeDC1394(void)" (?getVideoModeDC1394@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAEIXZ)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual unsigned int __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::getVideoModeMaskDC1394(void)" (?getVideoModeMaskDC1394@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAEIXZ)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual bool __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::getWhiteBalance(double &,double &)" (?getWhiteBalance@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAE_NAAN0@Z)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual bool __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::getWhiteBalanceDC1394(double &,double &)" (?getWhiteBalanceDC1394@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAE_NAAN0@Z)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual bool __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::hasAutoDC1394(int)" (?hasAutoDC1394@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAE_NH@Z)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual bool __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::hasFeatureDC1394(int)" (?hasFeatureDC1394@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAE_NH@Z)
[... cut ...]
3>c:\dev\icub-main\src\tools\icubskingui-qt\plugin\include/TouchSensor.h(10): fatal error C1083: Cannot open include file: 'gsl/gsl_math.h': No such file or directory
3>  Triangle.cpp
3>c:\dev\icub-main\src\tools\icubskingui-qt\plugin\include/TouchSensor.h(10): fatal error C1083: Cannot open include file: 'gsl/gsl_math.h': No such file or directory
3>  QtICubSkinGuiPlugin_automoc.cpp
3>c:\dev\icub-main\src\tools\icubskingui-qt\plugin\include/TouchSensor.h(10): fatal error C1083: Cannot open include file: 'gsl/gsl_math.h': No such file or directory
3>  Generating Code...
========== Build: 2 succeeded, 2 failed, 90 up-to-date, 0 skipped ==========
```

pattacini, 2015-01-29:
Only one problem is solved: `frameGrabberGui2-qt` does not compile yet.

Here's the log:

```
Microsoft (R) Microsoft Visual Studio 2012 Version 11.0.61030.0.
Copyright (C) Microsoft Corp. All rights reserved.
1>------ Build started: Project: cluster-scripts, Configuration: Release Win32 ------
1>  Copying icub-cluster scripts to C:/dev/icub-main/build/bin/Release
2>------ Build started: Project: frameGrabberGui2-qt, Configuration: Release Win32 ------
2>dc1394thread.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::RemoteFrameGrabberControlsDC1394(void)" (__imp_??0RemoteFrameGrabberControlsDC1394@dev@yarp@@QAE@XZ) referenced in function "protected: virtual void __thiscall DC1394Thread::run(void)" (?run@DC1394Thread@@MAEXXZ)
2>dc1394thread.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: virtual __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::~RemoteFrameGrabberControlsDC1394(void)" (__imp_??1RemoteFrameGrabberControlsDC1394@dev@yarp@@UAE@XZ) referenced in function "public: virtual void * __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::`scalar deleting destructor'(unsigned int)" (??_GRemoteFrameGrabberControlsDC1394@dev@yarp@@UAEPAXI@Z)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual bool __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::close(void)" (?close@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAE_NXZ)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual bool __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::getActiveDC1394(int)" (?getActiveDC1394@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAE_NH@Z)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual double __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::getBrightness(void)" (?getBrightness@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAENXZ)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual unsigned int __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::getBytesPerPacketDC1394(void)" (?getBytesPerPacketDC1394@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAEIXZ)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual unsigned int __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::getColorCodingDC1394(void)" (?getColorCodingDC1394@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAEIXZ)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual unsigned int __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::getColorCodingMaskDC1394(unsigned int)" (?getColorCodingMaskDC1394@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAEII@Z)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual double __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::getExposure(void)" (?getExposure@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAENXZ)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual unsigned int __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::getFPSDC1394(void)" (?getFPSDC1394@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAEIXZ)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual unsigned int __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::getFPSMaskDC1394(void)" (?getFPSMaskDC1394@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAEIXZ)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual double __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::getFeatureDC1394(int)" (?getFeatureDC1394@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAENH@Z)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual bool __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::getFormat7MaxWindowDC1394(unsigned int &,unsigned int &,unsigned int &,unsigned int &,unsigned int &,unsigned int &)" (?getFormat7MaxWindowDC1394@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAE_NAAI00000@Z)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual bool __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::getFormat7WindowDC1394(unsigned int &,unsigned int &,int &,int &)" (?getFormat7WindowDC1394@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAE_NAAI0AAH1@Z)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual double __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::getGain(void)" (?getGain@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAENXZ)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual double __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::getGamma(void)" (?getGamma@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAENXZ)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual double __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::getHue(void)" (?getHue@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAENXZ)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual unsigned int __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::getISOSpeedDC1394(void)" (?getISOSpeedDC1394@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAEIXZ)
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual double __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::getIris(void)" (?getIris@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAENXZ)
[... cut ...]
2>dc1394thread.obj : error LNK2001: unresolved external symbol "public: virtual bool __thiscall yarp::dev::RemoteFrameGrabberControlsDC1394::setWhiteBalanceDC1394(double,double)" (?setWhiteBalanceDC1394@RemoteFrameGrabberControlsDC1394@dev@yarp@@UAE_NNN@Z)
2>C:\dev\icub-main\build\bin\Release\frameGrabberGui2.exe : fatal error LNK1120: 63 unresolved externals
3>------ Build started: Project: python-scripts, Configuration: Release Win32 ------
3>  Copying manager.py to C:/dev/icub-main/build/bin/Release
========== Build: 2 succeeded, 1 failed, 91 up-to-date, 0 skipped ==========
```

drdanz, 2015-01-30:
@pattacini Should be fixed, but I don't have windows to test right now, can you please confirm or re-open if it is not fixed?

closed 2015-01-30 by drdanz

pattacini, 2015-01-30:
it compiles, thanks!

commit 1f274113 (2015-01-29): Fix QtICubSkinGuiPlugin build (GSL is required)

Fixes #95
commit d8f95016 (2015-01-30): [frameGrabberGui2-qt] Link with YARP_dev

Should fix #95 part 2