robotology/icub-main #96: iCubSkinGui-qt fails to open

alecive, 2015-01-30:
I tried on different linux machines (both mine and the cluster of the blue robot), but apparently the result is the same: `iCubSkinGui` (qt version) does not open. Here is the terminal output:

```
alecive@malakim:~/Programmazione/yarp/build (master)$ iCubSkinGui
QQmlApplicationEngine failed to load component 
qrc:/main.qml:3 module "QtQuick.Window" version 2.2 is not installed
```

gsaponaro, 2015-01-30:
Dunno about your cluster, but on your own machine you may want to check whether ```qtquick1-5-dev``` is installed (or equivalent package, according to http://wiki.icub.org/yarpdoc/dependencies.html).

alecive, 2015-02-02:
@gsaponaro thanks for suggestion. I have installed `qtquick1-5-dev (5.2.1-1ubuntu1)` on my machine but the problem persists.

Maybe this can be useful: with @vtikha we then noticed the qt5 directories are correctly found, but cmake finds an uncorrect `qmake` executable (because it still relies on `qmake-qt4`):

```
QT_QMAKE_EXECUTABLE              /usr/bin/qmake-qt4                                                                              
 Qt5Concurrent_DIR                /usr/lib/x86_64-linux-gnu/cmake/Qt5Concurrent                                                   
 Qt5Core_DIR                      /usr/lib/x86_64-linux-gnu/cmake/Qt5Core                                                         
 Qt5Gui_DIR                       /usr/lib/x86_64-linux-gnu/cmake/Qt5Gui                                                          
 Qt5Network_DIR                   /usr/lib/x86_64-linux-gnu/cmake/Qt5Network                                                      
 Qt5OpenGL_DIR                    /usr/lib/x86_64-linux-gnu/cmake/Qt5OpenGL                                                       
 Qt5Qml_DIR                       /usr/lib/x86_64-linux-gnu/cmake/Qt5Qml                                                          
 Qt5Quick_DIR                     /usr/lib/x86_64-linux-gnu/cmake/Qt5Quick                                                        
 Qt5Widgets_DIR                   /usr/lib/x86_64-linux-gnu/cmake/Qt5Widgets                                                      
 Qt5_DIR                          /usr/lib/x86_64-linux-gnu/cmake/Qt5     
```

@drdanz is it possible for this to be a `cmake` issue? If not, how can I solve this?

As a side note, in the process of installing qt5 on ubuntu trusty, I simply copy-pasted this command (from http://wiki.icub.org/yarpdoc/dependencies.html):
```
sudo apt-get install qtbase5-dev qtdeclarative5-dev qtmultimedia5-dev \
  qtdeclarative5-qtquick2-plugin qtdeclarative5-window-plugin \
  qtdeclarative5-qtmultimedia-plugin qtdeclarative5-controls-plugin
```
but qtquick1-5-dev was out from this list of packages. Is there something missing in the documentation?

drdanz, 2015-02-02:
Which Ubuntu version are you using? Maybe with the packages installed you have a QtQuick.Windows older than 2.2.
I don't know if this can be true, and if there is a reason why 2.2 is required, though.

alecive, 2015-02-02:
As I said, I'm on ubuntu trusty 14.04 and the version installed is `qtquick1-5-dev (5.2.1-1ubuntu1)`, but the exact same error appears on the debian `icub16` and `icub12` machines (as far as I remember) on `iCubGenova04`. If you're in IIT, you can directly test them :)

gsaponaro, 2015-02-02:
He's using Ubuntu Trusty (14.04). I'm using the same version, and I confirm the error when trying to launch iCubSkinGui (qt) on my machine.

Some more information:

```
$ dpkg --get-selections | grep libqt5
libqt5clucene5:amd64				install
libqt5concurrent5:amd64				install
libqt5core5a:amd64				install
libqt5dbus5:amd64				install
libqt5declarative5:amd64			install
libqt5designer5:amd64				install
libqt5designercomponents5:amd64			install
libqt5feedback5:amd64				install
libqt5gui5:amd64				install
libqt5help5:amd64				install
libqt5multimedia5:amd64				install
libqt5multimedia5-plugins:amd64			install
libqt5multimedia5-touch:amd64			deinstall
libqt5multimedia5-touch-plugins:amd64		deinstall
libqt5multimediaquick-p5:amd64			install
libqt5multimediaquick-p5-touch:amd64		deinstall
libqt5multimediawidgets5:amd64			install
libqt5network5:amd64				install
libqt5opengl5:amd64				install
libqt5opengl5-dev:amd64				install
libqt5organizer5:amd64				install
libqt5positioning5:amd64			install
libqt5printsupport5:amd64			install
libqt5qml-graphicaleffects:amd64		install
libqt5qml5:amd64				install
libqt5quick5:amd64				install
libqt5quickparticles5:amd64			install
libqt5quicktest5:amd64				install
libqt5script5:amd64				install
libqt5sensors5:amd64				install
libqt5sql5:amd64				install
libqt5sql5-sqlite:amd64				install
libqt5svg5:amd64				install
libqt5test5:amd64				install
libqt5webkit5:amd64				install
libqt5webkit5-qmlwebkitplugin:amd64		install
libqt5widgets5:amd64				install
libqt5xml5:amd64				install
```

```
$ ldd iCubSkinGui
	linux-vdso.so.1 =>  (0x00007fffb11f3000)
	libQt5Widgets.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5 (0x00007fe85ff12000)
	libQt5Qml.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Qml.so.5 (0x00007fe85f90a000)
	libQt5Quick.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Quick.so.5 (0x00007fe85f37f000)
	libQt5Gui.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5 (0x00007fe85ed32000)
	libQt5Core.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Core.so.5 (0x00007fe85e68b000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fe85e387000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fe85e171000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fe85ddaa000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fe85db8c000)
	libgobject-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0 (0x00007fe85d93b000)
	libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007fe85d632000)
	libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007fe85d2fd000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fe85cff7000)
	libQt5Network.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Network.so.5 (0x00007fe85ccb2000)
	libGL.so.1 => /usr/lib/nvidia-304/libGL.so.1 (0x00007fe85c995000)
	libpng12.so.0 => /lib/x86_64-linux-gnu/libpng12.so.0 (0x00007fe85c76f000)
	libharfbuzz.so.0 => /usr/lib/x86_64-linux-gnu/libharfbuzz.so.0 (0x00007fe85c519000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007fe85c300000)
	libicui18n.so.52 => /usr/lib/x86_64-linux-gnu/libicui18n.so.52 (0x00007fe85bef8000)
	libicuuc.so.52 => /usr/lib/x86_64-linux-gnu/libicuuc.so.52 (0x00007fe85bb7e000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fe85b97a000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007fe85b772000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fe860770000)
	libffi.so.6 => /usr/lib/x86_64-linux-gnu/libffi.so.6 (0x00007fe85b569000)
	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007fe85b32b000)
	libxcb.so.1 => /usr/lib/x86_64-linux-gnu/libxcb.so.1 (0x00007fe85b10c000)
	libnvidia-tls.so.304.125 => /usr/lib/nvidia-304/tls/libnvidia-tls.so.304.125 (0x00007fe85af08000)
	libnvidia-glcore.so.304.125 => /usr/lib/nvidia-304/libnvidia-glcore.so.304.125 (0x00007fe858b1e000)
	libXext.so.6 => /usr/lib/x86_64-linux-gnu/libXext.so.6 (0x00007fe85890c000)
	libfreetype.so.6 => /usr/lib/x86_64-linux-gnu/libfreetype.so.6 (0x00007fe858668000)
	libgraphite2.so.3 => /usr/lib/x86_64-linux-gnu/libgraphite2.so.3 (0x00007fe85844c000)
	libicudata.so.52 => /usr/lib/x86_64-linux-gnu/libicudata.so.52 (0x00007fe856bde000)
	libXau.so.6 => /usr/lib/x86_64-linux-gnu/libXau.so.6 (0x00007fe8569da000)
	libXdmcp.so.6 => /usr/lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007fe8567d4000)
```

alecive, 2015-02-02:
Ok, apparently version 2.2 is not even available in the 14.04 official ubuntu repositories. For 2.2 version qt 5.4. is needed.

drdanz, 2015-02-02:
This is not good, we should depend on qt 5.2... @lornat75 should we forward this?

lornat75, 2015-02-02:
Let's poke @aitek4iit to investigate this.

alecive, 2015-02-03:
Apparently, it was an easy fix. I just changed one line in `mail.qml`. On my machine everything works fine, can you please double check on yours @gsaponaro ?

closed 2015-02-03 by alecive

gsaponaro, 2015-02-03:
Confirmed working, thanks!

commit b1ef3e44 (2015-02-03): [iCubSkinGui-QT] Fixed #96 : changed dependencty from qtquick.window in main.qml (was 2.2, now is 2.1)