robotology/icub-main #97: [MAC-OS] error in compiling iCubSkinGui-qt

iron76, 2015-01-30:
Trying to compile on Os X 10.10.2 I am getting this weird error.

```
[ 48%] Building CXX object src/tools/iCubSkinGui-qt/plugin/CMakeFiles/QtICubSkinGuiPlugin.dir/qticubskinguiplugin.cpp.o
cd /Users/iron/Code/icub/build/src/tools/iCubSkinGui-qt/plugin && /usr/bin/c++   -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_NO_DEBUG -DQT_PLUGIN -DQT_QML_LIB -DQT_QUICK_LIB -DQtICubSkinGuiPlugin_EXPORTS -D_REENTRANT -O3 -DNDEBUG -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk -fPIC -I/Users/iron/Code/icub/build/src/tools/iCubSkinGui-qt/plugin -I/Users/iron/Code/icub/src/tools/iCubSkinGui-qt/plugin -I/Users/iron/Code/opencv/build -I/Users/iron/Code/opencv/include -I/Users/iron/Code/opencv/include/opencv -I/Users/iron/Code/opencv/modules/core/include -I/Users/iron/Code/opencv/modules/flann/include -I/Users/iron/Code/opencv/modules/imgproc/include -I/Users/iron/Code/opencv/modules/highgui/include -I/Users/iron/Code/opencv/modules/features2d/include -I/Users/iron/Code/opencv/modules/calib3d/include -I/Users/iron/Code/opencv/modules/ml/include -I/Users/iron/Code/opencv/modules/video/include -I/Users/iron/Code/opencv/modules/legacy/include -I/Users/iron/Code/opencv/modules/objdetect/include -I/Users/iron/Code/opencv/modules/photo/include -I/Users/iron/Code/opencv/modules/gpu/include -I/Users/iron/Code/opencv/modules/ocl/include -I/Users/iron/Code/opencv/modules/nonfree/include -I/Users/iron/Code/opencv/modules/contrib/include -I/Users/iron/Code/opencv/modules/stitching/include -I/Users/iron/Code/opencv/modules/superres/include -I/Users/iron/Code/opencv/modules/ts/include -I/Users/iron/Code/opencv/modules/videostab/include -I/Users/iron/Code/bin/include -I/usr/local/include -F/Users/iron/Code/qt/5.4/clang_64/lib -isystem /Users/iron/Code/qt/5.4/clang_64/lib/QtQuick.framework/Headers -isystem /Users/iron/Code/qt/5.4/clang_64/lib/QtGui.framework/Headers -isystem /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk/System/Library/Frameworks/OpenGL.framework/Headers -isystem /Users/iron/Code/qt/5.4/clang_64/lib/QtCore.framework/Headers -isystem /Users/iron/Code/qt/5.4/clang_64/mkspecs/macx-clang -isystem /Users/iron/Code/qt/5.4/clang_64/lib/QtQml.framework/Headers -I/Users/iron/Code/qt/5.4/clang_64/lib/QtNetwork.framework/Headers    -o CMakeFiles/QtICubSkinGuiPlugin.dir/qticubskinguiplugin.cpp.o -c /Users/iron/Code/icub/src/tools/iCubSkinGui-qt/plugin/qticubskinguiplugin.cpp
In file included from /Users/iron/Code/icub/src/tools/iCubSkinGui-qt/plugin/qticubskinguiplugin.cpp:103:
In file included from /Users/iron/Code/icub/src/tools/iCubSkinGui-qt/plugin/qticubskinguiplugin.h:4:
In file included from /Users/iron/Code/qt/5.4/clang_64/lib/QtQuick.framework/Headers/QtQuick:3:
In file included from /Users/iron/Code/qt/5.4/clang_64/lib/QtQuick.framework/Headers/QtQuickDepends:5:
In file included from /Users/iron/Code/qt/5.4/clang_64/lib/QtQml.framework/Headers/QtQml:6:
In file included from /Users/iron/Code/qt/5.4/clang_64/lib/QtQml.framework/Headers/qjsengine.h:42:
/Users/iron/Code/qt/5.4/clang_64/lib/QtQml.framework/Headers/qjsvalue.h:132:21: error: expected ';' at end of declaration list
    QJSValue(void *) Q_DECL_EQ_DELETE;
```

pattacini, 2015-03-26:
hoping this has disappeared...
...reopen it if not.

closed 2015-03-26 by pattacini

alecive, 2015-06-26:
This issue is tracked in https://github.com/robotology/icub-main/issues/134 .