robotology/icub-main #99: Doubling of code between gtk and qt versions

alecive, 2015-02-07:
Right now, any interface that has been ported to qt is doubling the code base, because most of the code gtk-independent was copied into the qt sw. Here is a quick demonstration:

```
/iCub/icub-main/src/tools$ ls -al `find | grep Fingertip | grep cpp`
-rw-rw-r-- 1 alecive alecive 272 gen 30 13:15 ./iCubSkinGui-gtk/src/Fingertip2Left.cpp
-rw-rw-r-- 1 alecive alecive 273 gen 30 13:15 ./iCubSkinGui-gtk/src/Fingertip2Right.cpp
-rw-rw-r-- 1 alecive alecive 273 gen 30 13:15 ./iCubSkinGui-gtk/src/Fingertip.cpp
-rw-rw-r-- 1 alecive alecive 272 gen 30 13:15 ./iCubSkinGui-qt/plugin/Fingertip2Left.cpp
-rw-rw-r-- 1 alecive alecive 273 gen 30 13:15 ./iCubSkinGui-qt/plugin/Fingertip2Right.cpp
-rw-rw-r-- 1 alecive alecive 273 gen 30 13:15 ./iCubSkinGui-qt/plugin/Fingertip.cpp
```

This appears also in `yarp`. Wouldn't it be possible to have the same codebase with the two GUI on top of it? By doing so, we would be able to better maintain the code as well as reduce the number of bugs. I do understand that eventually the gtk versions are going to be dropped, but I am not expecting it to happen soon.

pattacini, 2015-02-07:
Dropping `gtk` guis shall happen soon instead.

alecive, 2015-02-07:
Ok, I'll forget the gtk code. Even though probably it will not work any more :)

closed 2015-02-07 by alecive