robotology/icub-main #105: canmotioncontrol compilation failure

gsaponaro, 2015-02-12:
Hi, trying to compile icub-main with ENABLE_icubmod_canmotioncontrol=ON on our pc104 (yarp & icub-main are up-to-date) we get the following error:

```
Scanning dependencies of target canmotioncontrol
[ 29%] Building CXX object src/libraries/icubmod/canBusMotionControl/CMakeFiles/canmotioncontrol.dir/__/__/__/__/generated_code/yarpdev_add_canmotioncontrol.cpp.o
[ 29%] Building CXX object src/libraries/icubmod/canBusMotionControl/CMakeFiles/canmotioncontrol.dir/CanBusMotionControl.cpp.o
/usr/local/src/robot/icub-main/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.cpp: In member function ‘unsigned char yarp::dev::CanBusMotionControl::from_modevocab_to_modeint(int)’:
/usr/local/src/robot/icub-main/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.cpp:3747:16: warning: large integer implicitly truncated to unsigned type [-Woverflow]
/usr/local/src/robot/icub-main/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.cpp: In member function ‘virtual bool yarp::dev::CanBusMotionControl::setFilterTypeRaw(int, int)’:
/usr/local/src/robot/icub-main/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.cpp:5619:23: error: ‘ICUBCANPROTO_POL_MC_CMD__SET_TCFILTER_TYPE’ was not declared in this scope
make[2]: *** [src/libraries/icubmod/canBusMotionControl/CMakeFiles/canmotioncontrol.dir/CanBusMotionControl.cpp.o] Error 1
make[1]: *** [src/libraries/icubmod/canBusMotionControl/CMakeFiles/canmotioncontrol.dir/all] Error 2
make: *** [all] Error 2
```

pattacini, 2015-02-12:
Try to update [`icub-firmware-shared`](https://github.com/robotology/icub-firmware-shared) as well.

randaz81, 2015-02-12:
You ALWAYS need to update icub-firmware-shared every time you update the
pc104. Typically also a FW update is required (not in this case, but the
new build is going to be released next week).

gsaponaro, 2015-02-12:
OK, thanks.
I confirm that updating icub-firmware-shared was enough to fix our problem.

closed 2015-02-12 by gsaponaro