robotology/icub-main #107: Add a version check in CMake to verify that icub-main is linking against correct version of icub-firmware-shared

lornat75, 2015-02-16:
@drdanz ?

drdanz, 2015-02-16:
icub-firmware-shared already has a version number that is checked in icub-main, it is just a question of changing it...

lornat75, 2015-02-16:
Ok, we should just make sure it is maintained up to date: @randaz81 @marcoaccame @barbalberto

lornat75, 2015-02-16:
For completeness:

Version check in icub-main is done here: https://github.com/robotology/icub-main/blob/master/conf/iCubFindDependencies.cmake#L31
Version check of icub-firmware-shared is set here: https://github.com/robotology/icub-firmware-shared/blob/master/CMakeLists.txt#L11

randaz81, 2015-02-16:
It works, but the error message that I get is: "cannot find icub_firmware_shared library, turn off device ecc"" which can be somehow confusing and does not print the required vs found version (like in yarp)

lornat75, 2015-02-24:
@drdanz

drdanz, 2015-02-24:
This is coherent with what the icub-main repository does for the other dependencies, what kind of behaviour do you want? And do you want to change it for all the dependencies?

barbalberto, 2015-02-24:
the point was simply to change the error message to be like:

found icub_firmware_version is too old, found version xxx, required version yyy 

like we do on yarp instead of telling not found (which is not entirely true, it is found but wrong version)

commit 661f23c6 (2015-02-25): Do not find icub_firmware_shared quietly

Fixes: #107
commit 13aaaa1d (2015-02-26): Do not warn if icub_firmware_shared is not available.

This is a workaround to enable the warning only if icub_firmware_shared
was found, but the version is not compatible with# the requested one.

Fixes #107
closed 2015-02-26 by drdanz