robotology/icub-main #108: robotMotorGui and PID tuning

arocchi, 2015-02-16:
robotMotorGui casts to int the values sent as PID references, so that it is not possible to set gain lower than 1

iron76, 2015-02-16:
@randaz81 should know the reason for this choice. I guess it has to do with the current firmware implementation but it should be definitively fixed.

traversaro, 2015-03-02:
Dealing with the same issue: a scale parameter is used at the moment to scale those values. 
On the iCub firmware it is meant to be a left shift, so for example: k_p : 2 scale : 2  means: k_p = 2 >> 2 = 2/2^2 = 0.5 . 
However the behaviour on your robot depends on how setPid / setTorquePid are implemented in coman/walkman controlBoard driver.

rlober, 2015-05-12:
Hey guys, just wanted to see if I could bring this back to the top of the stack.

I really need to tune the control gains on Paris02 so I was trying to build a PID tuning tool for this purpose. I have looked over the documentation on the different control modes, but found very little on interfacing directly with the various control gains. My goal ultimately, would be to be able to look at one joint at a time, and tune the pos, vel and torque pids.

1. Does this tool already exist somewhere?
2. What is the best way of reading and writing PID gains during operation? Is there a tutorial somewhere?

cc @traversaro @randaz81 @iron76

barbalberto, 2015-05-12:
Not sure if your issue is related to this topic, however if you want to build a tool for PID tuning, you need to follow the tutorial http://wiki.icub.org/brain/icub_motor_control_tutorial.html and modify it by taking the relevant interfaces.
For position pid use 
yarp::dev::IPidControl, the call get/setPID (http://wiki.icub.org/yarpdoc/classyarp_1_1dev_1_1IPidControl.html)
for velocity pid use:
yarp::dev::IVelocityControl2 and use get/setVelPid (http://wiki.icub.org/yarpdoc/classyarp_1_1dev_1_1IVelocityControl2.html)
for torque pid use
yarp::dev::ITorqueControl and use get/setTorquePid (http://wiki.icub.org/yarpdoc/classyarp_1_1dev_1_1ITorqueControl.html)

rlober, 2015-05-13:
@barbalberto 

Definitely related! Thank you for the info. I will get started and let you guys know if I come up with anything useful. 

Ryan

pattacini, 2015-05-13:
Some info related to this issue:

- `robotMotorGui` has been finally declared superseded! Try to launch it, you'll get this :wink: 
[image]
- Regarding the PID tuning, this https://github.com/robotology/QA/issues/2#issuecomment-69908682 can be helpful too.

rlober, 2015-05-13:
@pattacini 

Thank you for the info, I will read through the document you linked to on the other thread.

Just FYI: it should read *depreciated* not deprecated. And I really like the new yarpmotorgui btw :+1:

iron76, 2015-05-13:
@pattacini not sure why you were so anxious about getting rid of the former version of the robotMotorGui :smile: It served the community for more than 8 years and I think we should have a proper retirement ceremony to celebrate the end of a glorious application that I guiltily started to code. I will always be thankful to @randaz81 who took the burden of its badly-thought structure, elevating it to levels of absolute excellence :)

randaz81, 2015-05-13:
@rlober
it seems that deprecate is the correct way to say: http://en.wikipedia.org/wiki/Deprecation

pattacini, 2015-05-13:
It was not me... @randaz81 decided for this.

randaz81, 2015-05-13:
@arocchi sorry for looking at issue only now, Kp,Kd, Ki gains are now floating point on yarpmotorgui-qt (https://github.com/robotology/yarp/commit/9281fd95c25cedf1ef70f691d644ceb6f0a214e1)

rlober, 2015-05-13:
@randaz81 

You are right! Apparently the software community has decided to adopt old English! I have to admit though that as an English speaker seeing "deprecate" in this sense, seems odd... but then again who cares!

randaz81, 2015-05-13:
@iron76 
>we should have a proper retirement ceremony to celebrate

That's why we are organizing the great BBQ this we!
Joking aside, the original code of robotMotorgui is still available in the repo, but it has been moved to yarp with different name (yarpmotorgui-gtk) and is no more supported. On the other side yarpmotorgui-qt is now default target for the yarpmotorgui 'link': it has been in testing for several months, and I think that now is mature enough for the official release. Nevertheless if the qt-libraries are not found on your system, yarpmotorgui will point to yarpmotorgui-gtk instead yarpmotorgui-qt.

randaz81, 2015-05-13:
@barbalberto 
just an additional note for icub-users:
> for velocity pid use: yarp::dev::IVelocityControl2 and use get/setVelPid 

This method is *currently* not used on the *icub platform*: in velocity mode, the controller generates a position stepwise trajectory and uses the same *position control* gains.
This may change in the future.

alecive, 2015-05-13:
@rlober :
 * isn't _depreciated_ something that loses value over time?
 * shall we close this issue? It seems to quite diverge from its original purpose in many ways :)

rlober, 2015-05-13:
@alecive 

Yes, to *depreciate* means to lose value or to reduce something's value. *Deprecate* means to discourage or disapprove and also, *to depreciate* something. I have seen *depreciate* used to indicate the obsolescence of something, more so than *deprecate*, but *deprecate* is an older word that has fallen out of oral usage in comparison to *depreciate* so that my be the root of my confusion. According to wikipedia *deprecate* has been adopted by the software community although I am not sure if this is because coders are English enthusiasts or because of typos. Either way it is on wikipedia so you are pretty much safe, but I would recommend using depreciate when not talking specifically about software.

closed 2015-05-13 by randaz81