robotology/icub-main #111: [iCubParis01] icubheadcalibrator not found

rlober, 2015-02-24:
I have just updated yarp and icub-main. When I launch the robotInterface on iCubParis01 I ge the following errors: 
~~~
794		3.673413		ERROR		Cannot find "icubheadcalibrator" plugin (not built in, and no .ini file found for it)
795		3.673466		ERROR		Check that YARP_DATA_DIRS leads to at least one directory with plugins/icubheadcalibrator.ini or share/yarp/plugins/icubheadcalibrator.ini in it
806		3.754097		ERROR		One or more devices failed opening... see previous log messages for more info 
807		4.188140		ERROR		Error in startup phase... see previous messages for more info 
~~~

Apparently icubheadcalibrator device was removed from icub-main, should we migrate the configuration file to parametric calibrator?

traversaro, 2015-02-24:
cc @randaz81

randaz81, 2015-02-24:
icubheadcalibrator has been deprecated months ago. Looking at your xml files, everything seems already configured properly. Try just to change the string icubheadcalibrator with parametricCalibrator in your icub-main\app\robots\iCubParis01\calibrators\head_calib.xml. If everything works properly, we can commit the fix.

traversaro, 2015-02-24:
Thanks @randaz81 , checking as soon as possible.

traversaro, 2015-02-25:
Hi @randaz81,
  we tried to just change the
```
    <device name="head_calibrator" type="icubheadcalibrator">         
```
line to
```
    <device name="head_calibrator" type="parametricCalibrator">         
```
but on startup the head is moving to the right, before we fault the robot for avoiding the head to crash on the shoulder.  I'll send a video of the failing calibration in your mail. 

cc @DanielePucci

randaz81, 2015-02-25:
I'll wait for the video to better understand. However, if icubParis01 has a V1 head, I expect the head to touch the shoulder during the calibration (there are no absolute encoders in the V1 head: the calibration is relative to the hardware limit)

DanielePucci, 2015-02-25:
Ok @randaz81, we were a little bit afraid of that movement, so we'll try to let it go.

@rlober @traversaro

DanielePucci, 2015-02-25:
It does not seem to work properly. Currently shooting another video.

randaz81, 2015-02-25:
Hi, I pushed a modified version (attempt) of head_calib.xml. Can you give it a try?
Also check if the head firmware is updated (if i remember correctly, latest build is of few months ago)

DanielePucci, 2015-02-25:
Yep. Trying right now.g

traversaro, 2015-02-25:
Just flashed the firmware (it seemed to be up to date but just in case) and using the new configuration file, but we still get the same behaviour: the head is reaching the left shoulder and then switching from calibrating mode to idle.

The output of the robotInterface filtered by `HeadV1_Calibrator` is:

~~~
65		0,740710		INFO		HeadV1_Calibrator : starting calibration 
70		0,741007		DEBUG		HeadV1_Calibrator : Joints calibration order: ( 0 1 2 ) ( 3 ) ( 4 ) ( 5 )  
89		0,821145		DEBUG		HeadV1_Calibrator : skipping maxPwm=0 of joint  0 
91		0,902305		DEBUG		HeadV1_Calibrator : skipping maxPwm=0 of joint  1 
94		0,983523		DEBUG		HeadV1_Calibrator : skipping maxPwm=0 of joint  2 
95		0,983769		DEBUG		In calibration  HeadV1_Calibrator : enabling joint  0  to test hardware limit 
98		0,994117		DEBUG		In calibration  HeadV1_Calibrator : enabling joint  1  to test hardware limit 
101		1,004169		DEBUG		In calibration  HeadV1_Calibrator : enabling joint  2  to test hardware limit 
116		1,125304		DEBUG		HeadV1_Calibrator : Calling calibrateJoint on joint  0  with params:  0 500 20 0 
117		1,135421		DEBUG		HeadV1_Calibrator : Calling calibrateJoint on joint  1  with params:  0 1333 20 0 
118		1,135582		DEBUG		HeadV1_Calibrator : Calling calibrateJoint on joint  2  with params:  0 1333 20 0 
145		1,236489		DEBUG		HeadV1_Calibrator : set 1 j 0 : Calibrating... enc values AFTER calib:  -12.2922 
146		1,236647		DEBUG		HeadV1_Calibrator : set 1 j 1 : Calibrating... enc values AFTER calib:  -51.4499 
147		1,236728		DEBUG		HeadV1_Calibrator : set 1 j 2 : Calibrating... enc values AFTER calib:  -54.5018 
710		22,362613		ERROR		HeadV1_Calibrator :Timeout while calibrating  0 
 
711		22,362793		ERROR		HeadV1_Calibrator : set 1 : Calibration went wrong! Disabling axes and keeping safe pid limit
 
715		22,473752		DEBUG		HeadV1_Calibrator : skipping maxPwm=0 of joint  3 
716		22,473941		DEBUG		In calibration  HeadV1_Calibrator : enabling joint  3  to test hardware limit 
719		22,584743		DEBUG		HeadV1_Calibrator : Calling calibrateJoint on joint  3  with params:  0 600 -20 0 
725		22,685537		DEBUG		HeadV1_Calibrator : set 2 j 3 : Calibrating... enc values AFTER calib:  33.8618 
735		24,688337		DEBUG		HeadV1_Calibrator : calib joint ended 
736		24,688538		DEBUG		HeadV1_Calibrator : set 2 : Calibration ended, going to zero!
 
737		25,193904		DEBUG		HeadV1_Calibrator : Sending positionMove to joint 3  (desired pos:  0 desired speed:  10  ) 
738		26,205043		DEBUG		HeadV1_Calibrator: checkGoneToZeroThreshold: joint: 3 curr: 45.416 des: 0.000 -> delta: 45.416 threshold: 2.000 output: 0.000 mode: pos
739		26,700024		DEBUG		HeadV1_Calibrator: checkGoneToZeroThreshold: joint: 3 curr: 39.977 des: 0.000 -> delta: 39.977 threshold: 2.000 output: 0.000 mode: pos
740		27,205579		DEBUG		HeadV1_Calibrator: checkGoneToZeroThreshold: joint: 3 curr: 31.982 des: 0.000 -> delta: 31.982 threshold: 2.000 output: 0.000 mode: pos
744		27,710545		DEBUG		HeadV1_Calibrator: checkGoneToZeroThreshold: joint: 3 curr: 22.677 des: 0.000 -> delta: 22.677 threshold: 2.000 output: 0.000 mode: pos
751		28,214059		DEBUG		HeadV1_Calibrator: checkGoneToZeroThreshold: joint: 3 curr: 13.718 des: 0.000 -> delta: 13.718 threshold: 2.000 output: 0.000 mode: pos
758		28,713895		DEBUG		HeadV1_Calibrator: checkGoneToZeroThreshold: joint: 3 curr: 6.408 des: 0.000 -> delta: 6.408 threshold: 2.000 output: 0.000 mode: pos
765		29,215131		DEBUG		HeadV1_Calibrator: checkGoneToZeroThreshold: joint: 3 curr: 1.885 des: 0.000 -> delta: 1.885 threshold: 2.000 output: 0.000 mode: pos
769		29,720418		DEBUG		HeadV1_Calibrator: checkGoneToZeroThreshold: joint: 3 curr: 0.138 des: 0.000 -> delta: 0.138 threshold: 2.000 output: 0.000 mode: pos
770		30,225818		DEBUG		HeadV1_Calibrator: checkGoneToZeroThreshold: joint: 3 curr: -0.002 des: 0.000 -> delta: 0.002 threshold: 2.000 output: 0.000 mode: pos
771		30,226021		DEBUG		HeadV1_Calibrator: checkGoneToZeroThreshold: joint: 3 completed with delta: 0.002 over: 2.000
772		30,226108		DEBUG		HeadV1_Calibrator : set 2 : Reached zero position!
 
773		30,307060		DEBUG		HeadV1_Calibrator : skipping maxPwm=0 of joint  4 
774		30,307302		DEBUG		In calibration  HeadV1_Calibrator : enabling joint  4  to test hardware limit 
777		30,418586		DEBUG		HeadV1_Calibrator : Calling calibrateJoint on joint  4  with params:  0 500 20 0 
780		30,519646		DEBUG		HeadV1_Calibrator : set 3 j 4 : Calibrating... enc values AFTER calib:  -5.08361 
781		32,525165		DEBUG		HeadV1_Calibrator : calib joint ended 
782		32,525374		DEBUG		HeadV1_Calibrator : set 3 : Calibration ended, going to zero!
 
783		33,031889		DEBUG		HeadV1_Calibrator : Sending positionMove to joint 4  (desired pos:  0 desired speed:  10  ) 
790		34,035382		DEBUG		HeadV1_Calibrator: checkGoneToZeroThreshold: joint: 4 curr: -20.309 des: 0.000 -> delta: 20.309 threshold: 180.000 output: 0.000 mode: pos
797		34,540634		DEBUG		HeadV1_Calibrator: checkGoneToZeroThreshold: joint: 4 curr: -20.307 des: 0.000 -> delta: 20.307 threshold: 180.000 output: 0.000 mode: pos
804		35,046416		DEBUG		HeadV1_Calibrator: checkGoneToZeroThreshold: joint: 4 curr: -20.307 des: 0.000 -> delta: 20.307 threshold: 180.000 output: 0.000 mode: pos
811		35,542559		DEBUG		HeadV1_Calibrator: checkGoneToZeroThreshold: joint: 4 curr: -20.307 des: 0.000 -> delta: 20.307 threshold: 180.000 output: 0.000 mode: pos
[... cut ...]
826		38,982349		DEBUG		HeadV1_Calibrator : set 4 : Calibration ended, going to zero!
 
827		39,487152		DEBUG		HeadV1_Calibrator : Sending positionMove to joint 5  (desired pos:  0 desired speed:  10  ) 
840		40,498517		DEBUG		HeadV1_Calibrator: checkGoneToZeroThreshold: joint: 5 curr: -0.001 des: 0.000 -> delta: 0.001 threshold: 180.000 output: 0.000 mode: pos
841		40,498740		DEBUG		HeadV1_Calibrator: checkGoneToZeroThreshold: joint: 5 completed with delta: 0.001 over: 180.000
842		40,498843		DEBUG		HeadV1_Calibrator : set 4 : Reached zero position!
~~~

pattacini, 2015-03-29:
Any update on this issue?
I'll leave this open for a short while, then I'll close it.

Consider https://github.com/robotology/icub-support/issues from now onward as proper place where to post such issues.

traversaro, 2015-03-29:
The issue is yet to be solved, but it definitely belongs to the 'icub-support' repo. I will open an issue there to cover the future updates.

closed 2015-03-29 by traversaro