robotology/icub-main #112: Matlab Function to get the Regressor Matrix ''Y( . )''

DiegoRomeres, 2015-02-24:
Considering the model 

f = Y( v , a ) \times \alpha

where:
'f' is the spatial net force acting on a rigid body
'v,a' are the spatial velocity and acceleration of the body, respectively
'Y( . , . )' the regressor matrix
'\alpha' the body's interial parameters,

is it possible to have a Matlab function which given 'v,a' returns 'Y( . , .)' ?

thanks

iron76, 2015-02-25:
@DiegoRomeres: currently @ommac is trying the C++ interface. I think that after the IROS deadline we could have a look at how much work is needed to create a simple Matlab interface. Two persons have done this in the past: @jeljaik and @naveenoid.

traversaro, 2015-05-04:
Initial implementation of matlab bindings for iDynTree was merged in https://github.com/robotology-playground/idyntree/pull/28 . Regressor generation is still not covered, but I thinks we can discuss about this in the iDynTree issues.

closed 2015-05-04 by traversaro

traversaro, 2015-06-22:
Just to "actually" close the issue. The first example for computation of regressor matrices using Matlab is available in https://github.com/robotology-playground/idyntree/tree/master/examples/matlab/iCubLeftArmDynamicRegressors .