robotology/icub-main #114: robotInterface segmentation fault on Vizzy

plinioMoreno, 2015-02-26:
I just pulled the most recent versions of yarp, icub-firmware-shared and icub-main, and when I run robotInterface on the Vizzy robot, I get a segmentation fault right after the startup phases are finished.

The printouts from the robotInterface as follows:

[DEBUG]Left_Arm_Calibrator : Sending positionMove to joint 3  (desired pos:  0 desired speed:  10  ) 
[DEBUG]Left_Arm_Calibrator : Sending positionMove to joint 4  (desired pos:  0 desired speed:  10  ) 
[DEBUG]Right_Arm_Calibrator : Sending positionMove to joint 3  (desired pos:  0 desired speed:  10  ) 
[DEBUG]Right_Arm_Calibrator : Sending positionMove to joint 4  (desired pos:  0 desired speed:  10  ) 
[DEBUG]Left_Arm_Calibrator: checkGoneToZeroThreshold: joint: 3 curr: -37.936 des: 0.000 -> delta: 37.936 threshold: 1.000 output: 0.000 mode: pos
[DEBUG]Right_Arm_Calibrator: checkGoneToZeroThreshold: joint: 3 curr: 36.059 des: 0.000 -> delta: 36.059 threshold: 1.000 output: 0.000 mode: pos
[DEBUG]Left_Arm_Calibrator: checkGoneToZeroThreshold: joint: 3 curr: -29.209 des: 0.000 -> delta: 29.209 threshold: 1.000 output: 0.000 mode: pos
[DEBUG]Right_Arm_Calibrator: checkGoneToZeroThreshold: joint: 3 curr: 29.028 des: 0.000 -> delta: 29.028 threshold: 1.000 output: 0.000 mode: pos
[DEBUG]Left_Arm_Calibrator: checkGoneToZeroThreshold: joint: 3 curr: -19.782 des: 0.000 -> delta: 19.782 threshold: 1.000 output: 0.000 mode: pos
[DEBUG]Right_Arm_Calibrator: checkGoneToZeroThreshold: joint: 3 curr: 19.789 des: 0.000 -> delta: 19.789 threshold: 1.000 output: 0.000 mode: pos
[DEBUG]Left_Arm_Calibrator: checkGoneToZeroThreshold: joint: 3 curr: -10.473 des: 0.000 -> delta: 10.473 threshold: 1.000 output: 0.000 mode: pos
[DEBUG]Right_Arm_Calibrator: checkGoneToZeroThreshold: joint: 3 curr: 10.591 des: 0.000 -> delta: 10.591 threshold: 1.000 output: 0.000 mode: pos
[DEBUG]Left_Arm_Calibrator: checkGoneToZeroThreshold: joint: 3 curr: -3.487 des: 0.000 -> delta: 3.487 threshold: 1.000 output: 0.000 mode: pos
[DEBUG]Right_Arm_Calibrator: checkGoneToZeroThreshold: joint: 3 curr: 3.718 des: 0.000 -> delta: 3.718 threshold: 1.000 output: 0.000 mode: pos
[DEBUG]Left_Arm_Calibrator: checkGoneToZeroThreshold: joint: 3 curr: -0.049 des: 0.000 -> delta: 0.049 threshold: 1.000 output: 0.000 mode: pos
[DEBUG]Right_Arm_Calibrator: checkGoneToZeroThreshold: joint: 3 curr: 0.464 des: 0.000 -> delta: 0.464 threshold: 1.000 output: 0.000 mode: pos
[DEBUG]Left_Arm_Calibrator: checkGoneToZeroThreshold: joint: 3 curr: 0.345 des: 0.000 -> delta: 0.345 threshold: 1.000 output: 0.000 mode: pos
[DEBUG]Left_Arm_Calibrator: checkGoneToZeroThreshold: joint: 3 completed with delta: 0.345 over: 1.000
[DEBUG]Left_Arm_Calibrator: checkGoneToZeroThreshold: joint: 4 curr: -0.007 des: 0.000 -> delta: 0.007 threshold: 1.000 output: 0.000 mode: pos
[DEBUG]Left_Arm_Calibrator: checkGoneToZeroThreshold: joint: 4 completed with delta: 0.007 over: 1.000
[DEBUG]Left_Arm_Calibrator : set 2 : Reached zero position!
 
[DEBUG]left_arm_calibrator finished calibration of device left_arm_mc 
[DEBUG]Right_Arm_Calibrator: checkGoneToZeroThreshold: joint: 3 curr: -0.005 des: 0.000 -> delta: 0.005 threshold: 1.000 output: 0.000 mode: pos
[DEBUG]Right_Arm_Calibrator: checkGoneToZeroThreshold: joint: 3 completed with delta: 0.005 over: 1.000
[DEBUG]Right_Arm_Calibrator: checkGoneToZeroThreshold: joint: 4 curr: 0.006 des: 0.000 -> delta: 0.006 threshold: 1.000 output: 0.000 mode: pos
[DEBUG]Right_Arm_Calibrator: checkGoneToZeroThreshold: joint: 4 completed with delta: 0.006 over: 1.000
[DEBUG]Right_Arm_Calibrator : set 2 : Reached zero position!
 
[DEBUG]right_arm_calibrator finished calibration of device right_arm_mc 
[INFO]All actions for action level 11 of startup phase finished. 
[INFO]startup phase finished. 

[2]+  Segmentation fault      (core dumped) robotInterface

plinioMoreno, 2015-02-26:
Just to add more information, I'm running the same config files at https://github.com/robotology/icub-main/tree/master/app/robots/vizzy

lornat75, 2015-02-27:
Can you recompiled with debug symbols and run it under gdb?

plinioMoreno, 2015-02-27:
Hi Lorenzo,

this is the output from backtrace when running robotInterface under gdb:
#0  0x00007ffff689bd75 in malloc_consolidate (av=0x7ffff6bd5720) at malloc.c:4254
#1  0x00007ffff689d436 in malloc_consolidate (av=0x7ffff6bd5720) at malloc.c:4215
#2  _int_malloc (av=0x7ffff6bd5720, bytes=1024) at malloc.c:3532
#3  0x00007ffff689ffb5 in __GI___libc_malloc (bytes=1024) at malloc.c:2924
#4  0x00007ffff714e8f8 in operator new(unsigned long) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff714e9a9 in operator new[](unsigned long) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff74c9f72 in yarp::os::ManagedBytes::ManagedBytes(unsigned long) () from /home/vizzy/github/yarp/build/lib/libYARP_OS.so.1
#7  0x00007ffff7499832 in yarp::os::impl::BufferedConnectionWriter::addPool(yarp::os::Bytes const&) () from /home/vizzy/github/yarp/build/lib/libYARP_OS.so.1
#8  0x00007ffff7499c35 in yarp::os::impl::BufferedConnectionWriter::push(yarp::os::Bytes const&, bool) () from /home/vizzy/github/yarp/build/lib/libYARP_OS.so.1
#9  0x00007ffff7495c32 in yarp::os::impl::BottleImpl::synch() () from /home/vizzy/github/yarp/build/lib/libYARP_OS.so.1
#10 0x00007ffff74ffa5a in yarp::os::impl::PortCore::sendHelper(yarp::os::PortWriter&, int, yarp::os::PortReader*, yarp::os::PortWriter*) ()
   from /home/vizzy/github/yarp/build/lib/libYARP_OS.so.1
#11 0x00007ffff7517ec4 in yarp::os::Port::write(yarp::os::PortWriter&, yarp::os::PortWriter*) const () from /home/vizzy/github/yarp/build/lib/libYARP_OS.so.1
#12 0x00007ffff74c9599 in yarp::os::LogForwarder::forward(std::string) () from /home/vizzy/github/yarp/build/lib/libYARP_OS.so.1
#13 0x00007ffff74c74d1 in yarp::os::impl::LogImpl::forward_callback(yarp::os::Log::LogType, char const*, char const*, unsigned int, char const*) ()
   from /home/vizzy/github/yarp/build/lib/libYARP_OS.so.1
#14 0x00000000004be072 in yarp::os::LogStream::~LogStream (this=0x7fffffffd770, __in_chrg=<optimized out>) at /home/vizzy/github/yarp/src/libYARP_OS/include/yarp/os/LogStream.h:59
#15 0x00000000004c9ac7 in RobotInterface::Robot::enterPhase (this=0x8875d8, phase=RobotInterface::ActionPhaseStartup) at /home/vizzy/github/icub-main/src/core/robotInterface/Robot.cpp:602
#16 0x00000000004c4ef4 in RobotInterface::Module::configure (this=0x7fffffffda50, rf=...) at /home/vizzy/github/icub-main/src/core/robotInterface/Module.cpp:71
#17 0x00007ffff753f90a in yarp::os::RFModule::runModule(yarp::os::ResourceFinder&) () from /home/vizzy/github/yarp/build/lib/libYARP_OS.so.1
#18 0x00000000004bdcf5 in main (argc=1, argv=0x7fffffffdbd8) at /home/vizzy/github/icub-main/src/core/robotInterface/main.cpp:53

plinioMoreno, 2015-02-27:
In the meantime I can confirm that when switching to commit 8ea9512867fbc9b13906daa678fbf621af1c29eb just for the robotInterface folder, the segfault does not happen.

lornat75, 2015-02-28:
@drdanz, ideas?

drdanz, 2015-02-28:
Looks like a crash in the log forwarding callback, perhaps @randaz81 has seen it before?

pattacini, 2015-03-29:
@plinioMoreno any update on that?

plinioMoreno, 2015-03-30:
I did not update yet robotInterface to the master branch of icub-main because I did not see any messages here (I was expecting a reply from @randaz81 to @drdanz question). Should i try again with the master branch and see if the error is gone?

plinioMoreno, 2015-07-06:
Hello all,

I tried the most recent version of yarp and icub-main (from last saturday July 04,2015), and the robotInterface works properly.

closed 2015-07-06 by plinioMoreno

pattacini, 2015-07-07:
cool :smiley: