robotology/icub-main #121: Head control problem with new experimental FW

pattacini, 2015-03-11:
Apparently, the new mechanical protection designed and installed by mechies to counter magnetic interference does not solve the problem we have been experiencing for the gaze control with the new firmware, especially when running IOL.

@alecive has seen the problem also with the red-ball demo: it happened once.

This is a reminder, but we should address this issue asap.

cc @randaz81, @vtikha

pattacini, 2015-07-03:
Lately, I've been spotting a misbehavior of the `iKinGazeCtrl` when running on `Blacky`, where we have the **experimental FW** running, and immediately after the **head maintenance**.

Briefly, the head largely oscillates when controlled by `iKinGazeCtrl`. The same does not happen on the standard FW on `Bluey`.

We need **urgent investigation** on this, otherwise `Blacky` cannot be used for demos and research!

/cc @randaz81 @robotology/icub-support-admin

pattacini, 2015-07-03:
Side note: I've checked that the gains of PIDs are unchanged with respect to the case when everything was working nice also with the experimental FW.

pattacini, 2015-07-04:
It is very likely that the original problem described in this issue is due to the fact that the user level sees a discrepancy between the commanded set-points to the neck and the actual feedback read from `icub/head/state:o`. This final discrepancy is never brought to zero.

pattacini, 2015-07-14:
Any update on this? It's quite urgent in view of VVV'15.
If possible, I'd prefer to go back to standard FW on Blacky and then investigate one problem at time.

pattacini, 2015-07-31:
We have news on this.

@randaz81 "sneakily" flashed the standard FW back and we spotted again control problem on the head; therefore, I feel to exclude that FW (either standard or exp.) can be the cause of the big oscillations (we actually broke the head today!).

From this point onward we keep this issue just to track the request for fixing the `exp. FW` in that we need to attain the commanded set-point, whereas I'm going to open up a new issue for the mechanical inspection.

/cc @super-ste @robotology/icub-support-admin

lornat75, 2015-08-17:
Should we close this?

pattacini, 2015-08-17:
Nope, 'cause `exp. FW` still needs to be tuned up.
@randaz81 will be working on this soon.

pattacini, 2016-09-02:
Closing since this `exp. FW` seems to be abandoned.

closed 2016-09-02 by pattacini