robotology/icub-main #123: [embObjMotionControl] robotInterface hangs if launched with motor switch set to off

traversaro, 2015-03-13:
If the user launches the robotInterface with motor switch set to off on the ethernet-based iCubs, it is then impossible to close the `robotInterface` with a Ctrl-C .

pattacini, 2016-09-01:
@marcoaccame did we solve this problem?

marcoaccame, 2016-09-01:
Hi @pattacini, ... I think that nobody worked on this issue.

However, I have just tested the behaviour on iCubGenova02 and:

* `yarprobotinterface` can be closed at the third CTRL-C.

[image]

[image]

pattacini, 2016-09-01:
@marcoaccame thanks for testing it.

@traversaro with 3 ctrl+c in a row didn't you manage to stop `yarprobotinterface`? In other words, was the module outright stuck?

Though needing 3 stop requests is not graceful, perhaps we can simply get away with that and close the issue. What do you think @traversaro ?

Otherwise, I think we would require to asynchronously _unlock a mutex_ or _exit from a while loop_ that might be there while querying the boards status.

traversaro, 2016-09-01:
Yes, I opened the issue because it was impossible to close it with 3 Ctrl-C, so I think it can be closed.

pattacini, 2016-09-01:
Great then.
Closing.

closed 2016-09-01 by pattacini