robotology/icub-main #129: Gaze control problem in position-direct mode

pattacini, 2015-03-26:
We've been spotting problem with `iKinGazeCtrl` set in `position-direct` mode.

/cc @randaz81 @vtikha @alecive @lornat75

lornat75, 2015-03-26:
I believe we should first write and run a test for position-direct, then proceed with debugging the specific problems with iKinGazeCtrl. What do you think?

pattacini, 2015-03-26:
It's fine with me; I can help debugging either way.
The weird thing is that:
- `cartesiancontrollerserver` is still working in `position-direct`;
- `iKinGazeCtrl` has not changed since months and it was fully functional in `position-direct` until let's say two weeks ago;
- this problem emerged first on ethernet robots while Blacky was ok with the same SW release, then _infected_ the Blacky;

lornat75, 2015-03-26:
Strange... Maybe running a generic test for position-direct on the arms and the head can reveal something...

alecive, 2015-03-27:
@pattacini it might be that the problems on the blackie are different from the problems on the ETH robots (with similar outcomes) :)

pattacini, 2015-03-27:
Uhm... it might certainly be but, actually, I've seen exactly the same behavior both on `iCubLisboa01` and `Blacky`/`Bluey`: they are different in terms of ETH/CAN and FW std/exp.

pattacini, 2015-03-31:
I've added up some messages to `iKinGazeCtrl` to investigate the issue. Suddenly, on `Blacky` the `position-direct` started over working fine. We have to repeat the experiment on `Reddy` or `Bluey`: stay tuned.

At the same time, @randaz81 performed single-joint test on the neck in `position-direct` mode; found nothing special: it worked. Next experiment will involve the control of multiple joints.

pattacini, 2015-04-03:
Update on this.

I managed to run some tests on the gaze controller and I found it is behaving correctly.

The figure below shows, for each joint of the neck, the `position-direct` commands in blue, the `feedback` in red and the `gate` in orange, the latter accounting for the time periods when the controller is active.
[image]

Circled in red the weird event happening on the yaw joint is zoomed in below:
[image]

As you can see, there is an unwanted snap stop while the joint is moving. The controller is however producing a smooth varying reference.

Attached the [archive](https://files.zenhub.io/551e67b6a5fa455769d2d176) containing the logged data along with the MATLAB script used to generate the graphs.

Further investigations are thus required to see whether those commands are actually received at the PC104 side.

/cc @randaz81 @lornat75

lornat75, 2015-04-03:
Friction?

pattacini, 2015-04-03:
I've already thought friction might play its role, but how likely can it be happening all of a sudden and on many different robots?

pattacini, 2015-04-03:
I tend to exclude `stiction` because the snap stop is at the middle of trajectory.
It would be interesting to understand whether we have this snap stop always in correspondence of roughly the same yaw angle.

pattacini, 2015-06-16:
/cc @francesco-romano @DanielePucci

pattacini, 2015-06-20:
**I got it!**

With the aim of narrowing down the problem, I've designed this [**`test program`**](https://files.zenhub.io/55858c7fa5e87d7a13418be2) that basically takes as input the data logged from the port `iKinGazeCtrl/dbg:o` and then sends the same commands to the robot. This way I managed to fully decouple from the gaze controller.

###### Preamble
The gaze controller opens up the device driver making use of the **`writeStrict`** option (see [here](https://github.com/robotology/icub-main/blob/master/src%2Fmodules%2FiKinGazeCtrl%2Fsrc%2Fmain.cpp#L1034)), because it has to send both direct positions (to the neck) and velocities (to the eyes) at each control slot of `10 ms`, therefore we don't want to lose data.

###### Then, we have two cases
1. If I keep sending `poss`+`vels` each slot, I can see the problem appearing along with the test program.
2. If I send just the `poss` and skip `vels` (i.e. launch: `txTarget --vel off` that corresponds to commenting out [this line](https://github.com/robotology/icub-main/blob/master/src%2Fmodules%2FiKinGazeCtrl%2Fsrc%2Fcontroller.cpp#L646)), the movements remain smooth!

###### To sum up
The `writeStrict` was used to function in the past, and in fact `iKinGazeCtrl` was running fine in `position-direct` mode; what happened?
Can we look into it asap, please?

/cc @lornat75 @barbalberto @randaz81 @alecive @vtikha

alecive, 2015-06-21:
Great! This seem to fit also with some issues I had with the gazeStabilizer during its early stages of development. I don't remember exactly the context, but it was related to the fact that I was sending commands at the neck decoupled from those at the eyes.

traversaro, 2015-06-21:
Greate indeed! I am curious to check if is a problem of the `writeStrict` option (i.e. the streaming messages are not sent from the `RemoteControlBoard`) or a problem of the `Contro;BoardWrapper` callback (i.e. the messages too close are not properly received).

pattacini, 2015-06-21:
Once this issue will be solved, I hope to see regression tests running overnight to prevent the same bother from happening again :smirk: 

[Some developers](https://github.com/robotology/icub-tests#contributers) worked hard to design and implement those [tests](https://github.com/robotology/icub-tests), it's then a pity they are not used yet.

lornat75, 2015-06-22:
I think the problem could be the callback of the buffered port discarding messages that are too close. Can you try calling setStrict():

http://eris.liralab.it/yarpdoc/classyarp_1_1os_1_1BufferedPort.html#a66c1b8b92b88fbb1a9bc329819788806

In this case delay may accumulate if the ControlBoardWrapper cannot read messages quickly enough but let's see what happens.

pattacini, 2015-06-23:
Thanks to @barbalberto this issue seems to be finally solved!

closed 2015-06-23 by pattacini

pattacini, 2015-06-23:
@robotology/qa-users 
you are warmly suggested to restore **`position-direct`** mode of `iKinGazeCtrl` as per instructions outlined [here](https://github.com/robotology/icub-main/issues/129#issue-64464506).

Remember to update `yarp`.