robotology/icub-main #131: yarpdataplayer Repeat behaviour when switch datasets

arrenglover, 2015-03-27:
It seems the repeat option is not persistent between loading two different datasets. Example:
1. I load a dataset and tick the repeat option dropdown list. This dataset loops as expected
2. I stop the dataset and open a new dataset in the same yarpdataplayer instance. 'Repeat' is still ticked in the dropdown menu but the dataset does not loop (see figure).
3. As a workaround I must turn off repeat and then turn it back on again.

[image]

I think proper behaviour should be: 
(a) that the repeat remains on when the dataset is switched -or-
(b) repeat is turned off, but the gui is updated so I only need to turn it back on (not off and then on)

vtikha, 2015-03-27:
Ciao @arrenglover, thanks for noting this, I have implemented and pushed the (b) scenario. (new dataset = new settings)

arrenglover, 2015-03-27:
Okay sorry about the wrong repository. It now works as you say. Grazie.

closed 2015-03-27 by arrenglover