robotology/icub-main #132: Stiction up/down params for position control on ETH firmware

pattacini, 2015-03-28:
As agreed with @randaz81 we should check whether the stiction `up` and `down` parameters used in position control mode have been correctly ported to be used along with ETH firmware.

These parameters are relevant for [`fingersTuner`](http://wiki.icub.org/iCub/main/dox/html/group__icub__fingersTuner.html) tool.

randaz81, 2015-07-20:
looking in robotology\firmware\*, searching for:
`ICUBCANPROTO_POL_MC_CMD__SET_POS_STICTION_PARAMS`
`ICUBCANPROTO_POL_MC_CMD__GET_POS_STICTION_PARAMS`
`ICUBCANPROTO_POL_MC_CMD__SET_TORQUE_STICTION_PARAMS`
`ICUBCANPROTO_POL_MC_CMD__GET_TORQUE_STICTION_PARAMS`
and double checking with the contents of the file:
`robotology\icub-firmware\emBODY\eBcode\arch-arm\board\ems004\appl\v2\cfg\eoprot-callbacks`
I can state that those parameters are not transmitted to MC4 boards on the ETH architecture. They have to be implemented.

valegagge, 2016-07-28:
@randaz81 Is it still necessary to implement these messages?

pattacini, 2016-07-28:
@valegagge yes :smile:

pattacini, 2016-09-01:
@valegagge any rough estimate on when we could have this implemented?

valegagge, 2016-09-05:
I hope to close this issue before of  23 September.

valegagge, 2016-09-07:
Now ems firmware sends and reads stiction params from/to mc4 (CAN).

(See icub-firmware@ce2a729d92321cfb5e4e744b759e953b79c4a12a)

I tested firmware on purple iCub by setting and raeding stinction params by yarprobotgui. It seems work fine.

closed 2016-09-07 by valegagge

pattacini, 2016-09-07:
Thanks @valegagge 🌹 
Sooner or later, I'll have the chance to test the `fingersTuner` module over this update.