robotology/icub-main #133: Avoid hitting mechanical limits while controlling joints equipped with relative encoders

pattacini, 2015-03-28:
Some time ago we discussed with @randaz81 and @lornat75 about the possibility to introduce in the firmware safety bands around mechanical stop of joints equipped with relative encoders (e.g. eyes, prono-supination ...). These bands already exist for joints with absolute encoders.

This way, high-level modules will drive those joints always within some allowed range to prevent effects such as elasticity during collisions with stops from negatively affecting the overall knowledge of the joints configuration.

Deciding proper bands is a matter of tuning an heuristic, but could be still beneficial. Whether to implement such safe margins in the firmware needs to be clarified. This issue is a sort of reminder.

randaz81, 2015-07-20:
Is this issue caused by the the mechanical collision of the joint with hardware limits? If so, what is the difference respect to narrowing the joint range in the xml file? In any case, I think that the safe band should be properly set in the configuration file.

pattacini, 2015-07-21:
Imagine that the physical span of a joint (that will be calibrated at start-up) is `-90...90 [deg]`, then what would be good to have is the joint ranging at run-time in the set `-89...89 [deg]` with let's say `1 [deg]` of safety margin. Therefore, the `min` and `max` values read from the robot using the `IControlLimits` interface will be actually `-89 [deg]` and `89 [deg]`, respectively.

Of course, the safety margins can be conveniently chosen from within the configuration files.

closed 2021-03-27 by pattacini