robotology/icub-main #134: problem in installing icub-main

azadm, 2015-03-30:
Here is the error that I get when I compile icub-main in the build directory:
````
Scanning dependencies of target QtICubSkinGuiPlugin
[ 49%] Building CXX object src/tools/iCubSkinGui-qt/plugin/CMakeFiles/QtICubSkinGuiPlugin.dir/Fingertip2Left.cpp.o
[ 49%] Building CXX object src/tools/iCubSkinGui-qt/plugin/CMakeFiles/QtICubSkinGuiPlugin.dir/Fingertip2Right.cpp.o
[ 49%] Building CXX object src/tools/iCubSkinGui-qt/plugin/CMakeFiles/QtICubSkinGuiPlugin.dir/Fingertip.cpp.o
[ 49%] Building CXX object src/tools/iCubSkinGui-qt/plugin/CMakeFiles/QtICubSkinGuiPlugin.dir/PalmLeft.cpp.o
[ 50%] Building CXX object src/tools/iCubSkinGui-qt/plugin/CMakeFiles/QtICubSkinGuiPlugin.dir/PalmRight.cpp.o
[ 50%] Building CXX object src/tools/iCubSkinGui-qt/plugin/CMakeFiles/QtICubSkinGuiPlugin.dir/qticubskinguiplugin.cpp.o
In file included from /Users/azadm/Documents/icub-main/src/tools/iCubSkinGui-qt/plugin/qticubskinguiplugin.cpp:103:
In file included from /Users/azadm/Documents/icub-main/src/tools/iCubSkinGui-qt/plugin/qticubskinguiplugin.h:4:
In file included from /Applications/Qt/5.3/clang_64/lib/QtQuick.framework/Headers/QtQuick:3:
In file included from /Applications/Qt/5.3/clang_64/lib/QtQuick.framework/Headers/QtQuickDepends:4:
In file included from /Applications/Qt/5.3/clang_64/lib/QtQml.framework/Headers/QtQml:6:
In file included from /Applications/Qt/5.3/clang_64/lib/QtQml.framework/Headers/qjsengine.h:50:
/Applications/Qt/5.3/clang_64/lib/QtQml.framework/Headers/qjsvalue.h:140:21: error: expected ';' at end of declaration list
    QJSValue(void *) Q_DECL_EQ_DELETE;
                    ^
In file included from /Users/azadm/Documents/icub-main/src/tools/iCubSkinGui-qt/plugin/qticubskinguiplugin.cpp:103:
In file included from /Users/azadm/Documents/icub-main/src/tools/iCubSkinGui-qt/plugin/qticubskinguiplugin.h:4:
In file included from /Applications/Qt/5.3/clang_64/lib/QtQuick.framework/Headers/QtQuick:3:
In file included from /Applications/Qt/5.3/clang_64/lib/QtQuick.framework/Headers/QtQuickDepends:4:
In file included from /Applications/Qt/5.3/clang_64/lib/QtQml.framework/Headers/QtQml:9:
/Applications/Qt/5.3/clang_64/lib/QtQml.framework/Headers/qqml.h:114:9: error: use of undeclared identifier 'qRegisterNormalizedMetaType'
        qRegisterNormalizedMetaType<T *>(pointerName.constData()),
        ^
/Applications/Qt/5.3/clang_64/lib/QtQml.framework/Headers/qqml.h:114:37: error: 'T' does not refer to a value
        qRegisterNormalizedMetaType<T *>(pointerName.constData()),
                                    ^
/Applications/Qt/5.3/clang_64/lib/QtQml.framework/Headers/qqml.h:106:19: note: declared here
template<typename T>
                  ^
/Applications/Qt/5.3/clang_64/lib/QtQml.framework/Headers/qqml.h:114:40: error: expected expression
        qRegisterNormalizedMetaType<T *>(pointerName.constData()),
                                       ^
/Applications/Qt/5.3/clang_64/lib/QtQml.framework/Headers/qqml.h:115:9: error: use of undeclared identifier 'qRegisterNormalizedMetaType'
        qRegisterNormalizedMetaType<QQmlListProperty<T> >(listName.constData()),
        ^
/Applications/Qt/5.3/clang_64/lib/QtQml.framework/Headers/qqml.h:115:57: error: expected '(' for function-style cast or type construction
        qRegisterNormalizedMetaType<QQmlListProperty<T> >(listName.constData()),
                                    ~~~~~~~~~~~~~~~~~~~ ^
/Applications/Qt/5.3/clang_64/lib/QtQml.framework/Headers/qqml.h:147:9: error: use of undeclared identifier 'qRegisterNormalizedMetaType'
        qRegisterNormalizedMetaType<T *>(pointerName.constData()),
        ^
/Applications/Qt/5.3/clang_64/lib/QtQml.framework/Headers/qqml.h:147:37: error: 'T' does not refer to a value
        qRegisterNormalizedMetaType<T *>(pointerName.constData()),
                                    ^
/Applications/Qt/5.3/clang_64/lib/QtQml.framework/Headers/qqml.h:139:19: note: declared here
template<typename T>
                  ^
/Applications/Qt/5.3/clang_64/lib/QtQml.framework/Headers/qqml.h:147:40: error: expected expression
        qRegisterNormalizedMetaType<T *>(pointerName.constData()),
                                       ^
/Applications/Qt/5.3/clang_64/lib/QtQml.framework/Headers/qqml.h:148:9: error: use of undeclared identifier 'qRegisterNormalizedMetaType'
        qRegisterNormalizedMetaType<QQmlListProperty<T> >(listName.constData()),
        ^
/Applications/Qt/5.3/clang_64/lib/QtQml.framework/Headers/qqml.h:148:57: error: expected '(' for function-style cast or type construction
        qRegisterNormalizedMetaType<QQmlListProperty<T> >(listName.constData()),
                                    ~~~~~~~~~~~~~~~~~~~ ^
/Applications/Qt/5.3/clang_64/lib/QtQml.framework/Headers/qqml.h:178:9: error: use of undeclared identifier 'qRegisterNormalizedMetaType'
        qRegisterNormalizedMetaType<T *>(pointerName.constData()),
        ^
/Applications/Qt/5.3/clang_64/lib/QtQml.framework/Headers/qqml.h:178:37: error: 'T' does not refer to a value
        qRegisterNormalizedMetaType<T *>(pointerName.constData()),
                                    ^
/Applications/Qt/5.3/clang_64/lib/QtQml.framework/Headers/qqml.h:170:19: note: declared here
template<typename T, int metaObjectRevision>
                  ^
/Applications/Qt/5.3/clang_64/lib/QtQml.framework/Headers/qqml.h:178:40: error: expected expression
        qRegisterNormalizedMetaType<T *>(pointerName.constData()),
                                       ^
/Applications/Qt/5.3/clang_64/lib/QtQml.framework/Headers/qqml.h:179:9: error: use of undeclared identifier 'qRegisterNormalizedMetaType'
        qRegisterNormalizedMetaType<QQmlListProperty<T> >(listName.constData()),
        ^
/Applications/Qt/5.3/clang_64/lib/QtQml.framework/Headers/qqml.h:179:57: error: expected '(' for function-style cast or type construction
        qRegisterNormalizedMetaType<QQmlListProperty<T> >(listName.constData()),
                                    ~~~~~~~~~~~~~~~~~~~ ^
/Applications/Qt/5.3/clang_64/lib/QtQml.framework/Headers/qqml.h:209:9: error: use of undeclared identifier 'qRegisterNormalizedMetaType'
        qRegisterNormalizedMetaType<T *>(pointerName.constData()),
        ^
/Applications/Qt/5.3/clang_64/lib/QtQml.framework/Headers/qqml.h:209:37: error: 'T' does not refer to a value
        qRegisterNormalizedMetaType<T *>(pointerName.constData()),
                                    ^
/Applications/Qt/5.3/clang_64/lib/QtQml.framework/Headers/qqml.h:201:19: note: declared here
template<typename T>
                  ^
/Applications/Qt/5.3/clang_64/lib/QtQml.framework/Headers/qqml.h:209:40: error: expected expression
        qRegisterNormalizedMetaType<T *>(pointerName.constData()),
                                       ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make[2]: *** [src/tools/iCubSkinGui-qt/plugin/CMakeFiles/QtICubSkinGuiPlugin.dir/qticubskinguiplugin.cpp.o] Error 1
make[1]: *** [src/tools/iCubSkinGui-qt/plugin/CMakeFiles/QtICubSkinGuiPlugin.dir/all] Error 2
make: *** [all] Error 2
````

It would be great if someone can help me out.

azadm, 2015-03-30:
solved!

vtikha, 2015-03-30:
@azadm could you please write the various steps you did to solve this?

azadm, 2015-03-30:
I am actually not sure what the problem was because I tried to recompile icub for a few times. But I guess  it was due to two different versions of Qt that I had. I uninstalled the one that I installed a while ago by using homebrew and then it worked fine.

francesco-romano, 2015-03-31:
can you please tell us which version of Qt are you currently using (and if you remember the older one)?

azadm, 2015-03-31:
the current version if 5.3 and the old one was 4.6 I guess.

vtikha, 2015-03-31:
This issue has been re-opened in order to keep track of this as the solution is just a patch and not a complete fix. This behaviour can be seen in various OS X machines that have used homebrew for the QT installation. We still need to figure out why the problem is only with iCubSkinGui-qt and not the other guis (what is the difference between them, includes/cmakeLists/etc?) and why the error is seen only when compiling in Xcode.

kt10aan, 2015-03-31:
Just some notes/thoughts:

* The qt4 and qt5 formulas in Homebrew cannot coexist happily right now. If someone has already installed qt4 through Homebrew and then installs qt5, qt5 remains unlinked by default (and then the Qt5_DIR has to be pointed to /usr/local/Cellar/qt5.....). 

* Issues can (and usually do) arise from someone upgrading his Homebrew installation packages without having upgraded Xcode to the latest version. Then, someone can end up with some package compiled with a previous Xcode version and then some other packages with another version. Running ``brew doctor`` often is recommended. 

* when trying to debug OS X issues, it is always useful (or rather, mandatory) to know the Homebrew and Xcode versions. The output of 
```brew config```
will have all the details listed.

pattacini, 2015-06-23:
Any update on that? Otherwise will be closed soon.

francesco-romano, 2015-06-24:
I still have problems compiling icub-main with Xcode.

kt10aan, 2015-06-24:
I just compiled icub-main with Xcode from a fresh git pull.
It compiled fine, with no errors.
To be sure, I did:
```rm -rf $ICUB_ROOT/build/*```
```cd $ICUB_ROOT/build```
```cmake -G Xcode ..``

Then I opened the resulting ``xcode`` project file, accepted the "Autogenerate Schemes" and hit the Build button 

Are you sure there is not a leftover variable/setting that messes up with your build environment?

My brew config output:
```
mbp:build kostas$ brew config
HOMEBREW_VERSION: 0.9.5
ORIGIN: https://github.com/Homebrew/homebrew
HEAD: 68779e0f71b9d74a6bdff852cbafd9f819b88b89
Last commit: 9 hours ago
HOMEBREW_PREFIX: /usr/local
HOMEBREW_CELLAR: /usr/local/Cellar
CPU: quad-core 64-bit broadwell
OS X: 10.10.3-x86_64
Xcode: 6.3.2
CLT: 6.3.2.0.1.1431401888
Clang: 6.1 build 602
X11: 2.7.7 => /opt/X11
System Ruby: 2.0.0-p481
Perl: /usr/bin/perl
Python: /usr/local/bin/python => /usr/local/Cellar/python/2.7.10/Frameworks/Python.framework/Versions/2.7/bin/python2.7
Ruby: /usr/bin/ruby
Java: 1.8.0_45
```

Tobias-Fischer, 2015-09-22:
Hi all,
I've just run into the same issue, as we've installed both qt4 and qt5 via homebrew. We cannot simply uninstall qt4, as we have other projects depending on it.

However, on a blog I found the solution (and explanation) to our problem: http://blog.glehmann.net/2015/01/06/Investigating-build-failure-with-both-Qt4-and-Qt5-installed-on-Mac-OS-X/

So all what needs to be done is setting `-I/usr/local/opt/qt5/include` as `CMAKE_CXX_FLAGS`.

Hope that helps some people in the future :)

Best, Tobi

iron76, 2015-10-09:
Simply executing `brew unlink qt` worked for me. Thanks for the help.

vtikha, 2015-10-22:
The fact that there seems to be a compatibility problem when qt4 and qt5 are both installed on your machine is clear. There are a few ways to go around this: 
 -  do a `brew unlink qt` compile iCub and then `brew link qt` but this is quite ugly. 
 - As @Tobias-Fischer mentioned in his post, and with the link provided this is resolvable but unfortunately again in a non elegant way as the path is hardcoded and might work for some installations but not for others. 
 - There seems to be a bug in the compiler as it looks like, that, by adding a useless flag in cmake clang is able to successfully compile modules requiring qt5 having qt4 still linked in the system. eg: in `CMAKE_CXX_FLAGS -print-search-dirs`
Honestly this is a strange issue but we wil test it on other systems and try to figure it out.
Leaving this open for now as a reminder. 

cc @iron76 @claudia-lat @ChiaraBartolozzi @kt10aan @francesco-romano

drdanz, 2016-08-04:
Is this still an issue?

claudiofantacci, 2016-08-04:
I don't honestly know whether this is still an issue or not.

What I do to compile iCub_main having homebrew qt4 (namely **qt**) and **qt5** is to do as @vtikha and @iron76 suggested:

> do a `brew unlink qt`, `brew link qt5` and compile iCub

I don't recall any compiling error using Xcode by doing so.
Maybe @francesco-romano can tell us something about it.

What I suggest is to put a remark on the wiki installation guide for Mac about
`brew unlink qt` -> `brew link qt5` -> `compile iCub`

traversaro, 2016-08-04:
I think that doing `brew unlink qt` could cause problem if you are using `qt4` applications such as Gazebo.  I think it is more conservative to just set the `Qt5_DIR` enviromental variable so that CMake can find the Qt5 installed by brew: 
~~~
export Qt5_DIR=/usr/local/opt/qt5/lib/cmake
~~~
as documented in http://www.yarp.it/install_yarp_mac.html .

claudiofantacci, 2016-08-04:
@traversaro that's correct, but I remember I had some problem with that and in the end I just do
`brew unlink qt` -> `brew link qt5` -> compile iCub -> `brew unlink qt5` -> `brew link qt`
and everything seems to run smoothly. I agree that this is not an elegant solution, but seems to be sufficient for program to compile and work properly.

Anyway, beside how you menage to compile and run things with qt4/qt5, I think that this issue can be closed.

francesco-romano, 2016-08-04:
@traversaro unfortunately that was not sufficient.
For some bug (which we did not explain to ourselves) on some machine on some particular planet alignment, icub-main was failing to build and the only solution we found was to unlink qt and relink it after.
Having said that, this has not happened to me since last year.

traversaro, 2016-08-04:
@francesco-romano  Uau.

pattacini, 2016-09-01:
@francesco-romano @traversaro @vtikha 

Problems solved?

claudiofantacci, 2016-09-01:
As long as qt5 will not be the mainstream qt package on brew, I don't think that the problem can be 100% solved and in a clean way.

Having said that, I think that this issue can be closed.

francesco-romano, 2016-09-02:
Let's close this and reopen in case it happens again

closed 2016-09-02 by francesco-romano