robotology/icub-main #135: ETH robots: update encoders @ 10 ms

pattacini, 2015-04-07:
We have to make sure that all ETH robots will update encoders at a period of `10 ms`.

In particular, all `xml` files of those robots located in `wrappers/motorControl` will have to contain the option `period` set to `10`.

@davidetome please check this and update the files as said.
@apaikan is currently validating this change with the testing suite.

/cc @randaz81 @lornat75

apaikan, 2015-04-07:
I have added a test case to **icub-tests** repository to tests the `robotInterface` wrappers frequency. To run the test and check that all the wrappers are running on 10ms (100Hrz) you can run the `PortsFrequency` test case as follows: 

```
$ cd icub-tests/build/plugins
$ testrunner -v -t PortsFrequency.so -p "--from robinterface_stream.ini" -e "--robotname icub"

``` 

You can add a new config file which contains a list of the ports to check any other streaming ports you need. for example : 
```INI
name "My Ports Frequency"
// check every port for <time> seconds.
time 2 

[PORTS]
//port-name   frequency(Hrz)   tolerance
/myport_1:o   100                  5
/myport_2:o   100                  5
...
```

davidetome, 2015-04-08:
changed period value to 10 in all xml files inside wrapper/motorcontrol

pattacini, 2015-04-08:
thanks @davidetome 
two things: 
- my name is Pattacini, not Pataccini as referenced in the commit :)
- I've seen that the script you run practically changed the indentation of all files, which is quite bad in the end; could we fix this up?

closed 2015-04-08 by pattacini