robotology/icub-main #136: iCub Genova04 (Blue) Wrappers frequency

apaikan, 2015-04-13:
Some of the robotInterface wrappers' threads of the iCub04 (Blue) are still running at 50Hrz. Should we change them to 100Hrz? 

Here is the result of the PortFrequency test using Robot Testing Framework (100Hrz ports are not shown):
 
```
[INFO]  (Interface Frequency) reports: Checking port /icub/left_arm/state:o ...
[INFO]  (Interface Frequency) reports: Source frequency 50 hrz. (min: 33, max: 65)
[FAIL]  (Interface Frequency) checking (diff < ports[i].tolerance): Calculated frequency is outside the desired range [95 .. 105]

[INFO]  (Interface Frequency) reports: Checking port /icub/left_arm/stateExt:o ...
[INFO]  (Interface Frequency) reports: Source frequency 50 hrz. (min: 40, max: 65)
[FAIL]  (Interface Frequency) checking (diff < ports[i].tolerance): Calculated frequency is outside the desired range [95 .. 105]

[FAIL]  (Interface Frequency) checking (connected): could not connect to remote port /icub/left_foot/analog:o.
[INFO]  (Interface Frequency) reports: Checking port /icub/right_arm/state:o ...
[INFO]  (Interface Frequency) reports: Source frequency 49 hrz. (min: 27, max: 62)
[FAIL]  (Interface Frequency) checking (diff < ports[i].tolerance): Calculated frequency is outside the desired range [95 .. 105]

[INFO]  (Interface Frequency) reports: Checking port /icub/right_arm/stateExt:o ...
[INFO]  (Interface Frequency) reports: Source frequency 49 hrz. (min: 24, max: 57)
[FAIL]  (Interface Frequency) checking (diff < ports[i].tolerance): Calculated frequency is outside the desired range [95 .. 105]

[INFO]  (Interface Frequency) reports: Checking port /icub/torso/state:o ...
[INFO]  (Interface Frequency) reports: Source frequency 50 hrz. (min: 42, max: 59)
[FAIL]  (Interface Frequency) checking (diff < ports[i].tolerance): Calculated frequency is outside the desired range [95 .. 105]

[INFO]  (Interface Frequency) reports: Checking port /icub/torso/stateExt:o ...
[INFO]  (Interface Frequency) reports: Source frequency 50 hrz. (min: 39, max: 66)
[FAIL]  (Interface Frequency) checking (diff < ports[i].tolerance): Calculated frequency is outside the desired range [95 .. 105]

```

pattacini, 2015-04-13:
The configuration files have been already updated in the repo by @davidetome.
It's just a matter of pulling them locally.

closed 2015-04-13 by pattacini

apaikan, 2015-04-13:
ok!