robotology/icub-main #140: Reaching accuracy with ETH robots is low

pattacini, 2015-04-25:
The **reaching accuracy** of a `ETH robot` is significantly lower than a `CAN robot`.
This is likely due to mis-tuned integral parts of the joints PID. A back-to-back
comparison needs to be done to better investigate this issue which is somewhat
relevant for the application software. For example, in `depth2kin` we are able
to achieve an accuracy of `0.001 m` with `iCubGenova01`, while an accuracy of
`0.02` must be set for `iCubSheffield01` (see 4afc00b1a6eadd8072e2af23662054afb56b0e2f).

pattacini, 2017-03-28:
Hi @ale-git 

This is a long lasting issue related to the tuning of PID on ETH robots.
I'll assign it to you as discussed.

randaz81, 2017-05-05:
I added test `positionControl-accuracy` in icub-tests repo. The test plots the PID response of a step trajectory (joint controlled in positionDirect mode). In this way, we can compare the accuracy of iCubgenova01 (CAN) with ETH iCubs.

closed 2021-03-27 by pattacini