robotology/icub-main #146: robotInterface DTD file not found

gsaponaro, 2015-04-30:
The file http://www.icub.org/DTD/robotInterfaceV1.0.dtd, referred to in the doctype declaration of many configuration files (for example ```icub-main/app/robots/iCubLisboa01/general.xml```), is giving ERROR 404: Not Found.

pattacini, 2015-11-11:
@gsaponaro I think we can close this, right?

gsaponaro, 2015-11-11:
The problem still exists. Ideally this file, which describes the structure of the XML files used by **robotInterface**, should be (re-)uploaded to the location that is being advertised at the top of all the XMLs.

```
$ wget http://www.icub.org/DTD/robotInterfaceV1.0.dtd
--2015-11-11 23:10:37--  http://www.icub.org/DTD/robotInterfaceV1.0.dtd
Resolving www.icub.org (www.icub.org)... 90.147.26.147
Connecting to www.icub.org (www.icub.org)|90.147.26.147|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2015-11-11 23:10:37 ERROR 404: Not Found.
```

/cc @lornat75 @randaz81 @drdanz @barbalberto

pattacini, 2015-11-12:
I see ... I thought it was solved by your recent PR which changed the content of the robot conf files. Let's keep this open then.

randaz81, 2015-11-12:
It is likely that this kind of format will be deprecated in the very near future (with xsd replacing dtd on our robot configuration files). So I think that it is not worthwhile to work on this now...

closed 2015-11-12 by randaz81

gsaponaro, 2015-11-12:
OK, thanks.