robotology/icub-main #147: [ymanager.ini] not found when yarpmanager is launched

rlober, 2015-05-05:
When I launch yarpmanager it doesn't find an .ini file:
```bash
ryan@ISIR2014057:~$ yarpmanager
||| did not find ymanager.ini
yarp: cannot read from ymanager.ini
||| did not find ymanager.ini
```
and it doesn't seem to be looking in any of the usual locations ie: $YARP_DATA_DIRS. When I run a `locate ymanager.ini` I get the following results...
```bash
ryan@ISIR2014057:~$ locate ymanager.ini
/home/ryan/Code/iCub_Software/icub-main/app/ymanager.ini
/home/ryan/Code/iCub_Software/icub-main/app/robots/coman/ymanager.ini
/home/ryan/Code/iCub_Software/icub-main/app/robots/vizzy/ymanager.ini
/home/ryan/Code/iCub_Software/yarp/src/yarpmanager/tests/ymanager.ini
```

Which means that nothing is getting copied to my build dir when I make. This is odd because all of the modules and whatnot build properly. I was wondering if maybe I am missing some CMake options or something like that. I put the CMakeCache at the end. Here is my `.bash_iCubrc`
```bash
export ROBOT_CODE=/home/ryan/Code/iCub_Software
export YARP_ROOT=$ROBOT_CODE/yarp
export YARP_DIR=$ROBOT_CODE/yarp/build
export ICUB_ROOT=$ROBOT_CODE/icub-main
export ICUB_DIR=$ROBOT_CODE/icub-main/build

export YARP_DATA_DIRS=$YARP_DIR/share/yarp:$ICUB_DIR/share/iCub
export YARP_ROBOT_NAME=iCubParis02

export PATH=$YARP_DIR/bin:$PATH
export PATH=$ICUB_DIR/bin:$PATH
export LD_LIBRARY_PATH=$YARP_DIR/lib:$LD_LIBRARY_PATH
export LD_LIBRARY_PATH=$ICUB_DIR/lib:$LD_LIBRARY_PATH

export GAZEBO_MODEL_PATH=${GAZEBO_MODEL_PATH}:$ROBOT_CODE/icub-gazebo
export GAZEBO_PLUGIN_PATH=${GAZEBO_PLUGIN_PATH}:$ROBOT_CODE/gazebo-yarp-plugins/build

CODYCO_SUPERBUILD_ROOT=$ROBOT_CODE/codyco-superbuild
export PATH=$CODYCO_SUPERBUILD_ROOT/build/install/bin:$PATH
export LD_LIBRARY_PATH=$CODYCO_SUPERBUILD_ROOT/build/install/lib:$LD_LIBRARY_PATH
export YARP_DATA_DIRS=$YARP_DATA_DIRS:$CODYCO_SUPERBUILD_ROOT/build/install/share/codyco

export Qt5_DIR=/opt/Qt/5.4/gcc_64/lib/cmake/Qt5/
```

**`CmakeCache.txt`**
```cmake
# This is the CMakeCache file.
# For build in directory: /home/ryan/Code/iCub_Software/icub-main/build
# It was generated by CMake: /usr/bin/cmake
# You can edit this file to change values found and used by cmake.
# If you do not want to change any of the values, simply exit the editor.
# If you do want to change a value, simply edit, save, and exit the editor.
# The syntax for the file is as follows:
# KEY:TYPE=VALUE
# KEY is the name of a variable in the cache.
# TYPE is a hint to GUI's for the type of VALUE, DO NOT EDIT TYPE!.
# VALUE is the current value for the KEY.

########################
# EXTERNAL cache entries
########################

//Path to a library.
ACE_ACE_LIBRARY_DEBUG:FILEPATH=ACE_ACE_LIBRARY_DEBUG-NOTFOUND

//Path to a library.
ACE_ACE_LIBRARY_RELEASE:FILEPATH=/usr/lib/libACE.so

//Additional compiler flags for ACE
ACE_DEFINITIONS:STRING=

[... 2343 lines cut ...]
_PC_ACE_LIBRARY_DIRS:INTERNAL=
_PC_ACE_LIBS:INTERNAL=
_PC_ACE_LIBS_L:INTERNAL=
_PC_ACE_LIBS_OTHER:INTERNAL=
_PC_ACE_LIBS_PATHS:INTERNAL=
_PC_ACE_PREFIX:INTERNAL=/usr
_PC_ACE_STATIC_CFLAGS:INTERNAL=
_PC_ACE_STATIC_CFLAGS_I:INTERNAL=
_PC_ACE_STATIC_CFLAGS_OTHER:INTERNAL=
_PC_ACE_STATIC_INCLUDE_DIRS:INTERNAL=
_PC_ACE_STATIC_LDFLAGS:INTERNAL=-lACE;-ldl;-lpthread;-lrt
_PC_ACE_STATIC_LDFLAGS_OTHER:INTERNAL=
_PC_ACE_STATIC_LIBDIR:INTERNAL=
_PC_ACE_STATIC_LIBRARIES:INTERNAL=ACE;dl;pthread;rt
_PC_ACE_STATIC_LIBRARY_DIRS:INTERNAL=
_PC_ACE_STATIC_LIBS:INTERNAL=
_PC_ACE_STATIC_LIBS_L:INTERNAL=
_PC_ACE_STATIC_LIBS_OTHER:INTERNAL=
_PC_ACE_STATIC_LIBS_PATHS:INTERNAL=
_PC_ACE_VERSION:INTERNAL=6.0.1
__pkg_config_checked__PC_ACE:INTERNAL=1
//ADVANCED property for variable: board_tranceiver
board_tranceiver-ADVANCED:INTERNAL=1

```

traversaro, 2015-05-05:
The search for `ymanager.ini` is not printing the usual messages because the verbosity of the ResourceFinder is setted to false in https://github.com/robotology/yarp/blob/master/src/yarpmanager-qt/main.cpp#L52 .

rlober, 2015-05-05:
Ok just made the change to that file and I get the following output:

```bash
ryan@ISIR2014057:~$ yarpmanager
||| clearing context
||| adding context [yarpmanager]
||| configuring
||| no policy found
||| default config file specified as ymanager.ini
||| checking [/home/ryan/Code/iCub_Software/icub-tutorials/src/motorControlAdvanced/build/ymanager.ini] (pwd)
||| checking [/home/ryan/.config/yarp/robots/icubSim] (robot YARP_CONFIG_HOME)
||| checking [/home/ryan/.local/share/yarp/robots/icubSim] (robot YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/robots/icubSim] (robot YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/robots/icubSim] (robot YARP_CONFIG_DIRS)
||| checking [/home/ryan/Code/iCub_Software/yarp/build/share/yarp/robots/icubSim] (robot YARP_DATA_DIRS)
||| checking [/home/ryan/Code/iCub_Software/icub-main/build/share/iCub/robots/icubSim] (robot YARP_DATA_DIRS)
||| checking [/home/ryan/Code/iCub_Software/codyco-superbuild/build/install/share/codyco/robots/icubSim] (robot YARP_DATA_DIRS)
||| checking [/home/ryan/Code/iCub_Software/yarp/build/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/home/ryan/Code/iCub_Software/icub-main/build/share/iCub/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/home/ryan/Code/iCub_Software/codyco-superbuild/build/install/share/codyco/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/home/ryan/.config/yarp/contexts/yarpmanager] (context YARP_CONFIG_HOME)
||| checking [/home/ryan/.local/share/yarp/contexts/yarpmanager] (context YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/contexts/yarpmanager] (context YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/contexts/yarpmanager] (context YARP_CONFIG_DIRS)
||| checking [/home/ryan/Code/iCub_Software/yarp/build/share/yarp/contexts/yarpmanager] (context YARP_DATA_DIRS)
||| checking [/home/ryan/Code/iCub_Software/icub-main/build/share/iCub/contexts/yarpmanager] (context YARP_DATA_DIRS)
||| checking [/home/ryan/Code/iCub_Software/codyco-superbuild/build/install/share/codyco/contexts/yarpmanager] (context YARP_DATA_DIRS)
||| checking [/home/ryan/.config/yarp/ymanager.ini] (YARP_CONFIG_HOME)
||| checking [/home/ryan/.local/share/yarp/ymanager.ini] (YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/ymanager.ini] (YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/ymanager.ini] (YARP_CONFIG_DIRS)
||| checking [/home/ryan/Code/iCub_Software/yarp/build/share/yarp/ymanager.ini] (YARP_DATA_DIRS)
||| checking [/home/ryan/Code/iCub_Software/icub-main/build/share/iCub/ymanager.ini] (YARP_DATA_DIRS)
||| checking [/home/ryan/Code/iCub_Software/codyco-superbuild/build/install/share/codyco/ymanager.ini] (YARP_DATA_DIRS)
||| did not find ymanager.ini
yarp: cannot read from ymanager.ini
||| finding file [from]
||| did not find ymanager.ini
||| finding paths [applications]
||| checking [/home/ryan/Code/iCub_Software/icub-tutorials/src/motorControlAdvanced/build/applications] (pwd)
||| checking [/home/ryan/.config/yarp/applications] (YARP_CONFIG_HOME)
||| checking [/home/ryan/.local/share/yarp/applications] (YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/applications] (YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/applications] (YARP_CONFIG_DIRS)
||| checking [/home/ryan/Code/iCub_Software/yarp/build/share/yarp/applications] (YARP_DATA_DIRS)
||| found /home/ryan/Code/iCub_Software/yarp/build/share/yarp/applications
||| checking [/home/ryan/Code/iCub_Software/icub-main/build/share/iCub/applications] (YARP_DATA_DIRS)
||| checking [/home/ryan/Code/iCub_Software/codyco-superbuild/build/install/share/codyco/applications] (YARP_DATA_DIRS)
||| finding paths [scripts]
||| finding paths [modules]
||| checking [/home/ryan/Code/iCub_Software/icub-tutorials/src/motorControlAdvanced/build/modules] (pwd)
||| checking [/home/ryan/.config/yarp/modules] (YARP_CONFIG_HOME)
||| checking [/home/ryan/.local/share/yarp/modules] (YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/modules] (YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/modules] (YARP_CONFIG_DIRS)
||| checking [/home/ryan/Code/iCub_Software/yarp/build/share/yarp/modules] (YARP_DATA_DIRS)
||| found /home/ryan/Code/iCub_Software/yarp/build/share/yarp/modules
||| checking [/home/ryan/Code/iCub_Software/icub-main/build/share/iCub/modules] (YARP_DATA_DIRS)
||| found /home/ryan/Code/iCub_Software/icub-main/build/share/iCub/modules
||| checking [/home/ryan/Code/iCub_Software/codyco-superbuild/build/install/share/codyco/modules] (YARP_DATA_DIRS)
||| finding paths [resources]
||| checking [/home/ryan/Code/iCub_Software/icub-tutorials/src/motorControlAdvanced/build/resources] (pwd)
||| checking [/home/ryan/.config/yarp/resources] (YARP_CONFIG_HOME)
||| checking [/home/ryan/.local/share/yarp/resources] (YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/resources] (YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/resources] (YARP_CONFIG_DIRS)
||| checking [/home/ryan/Code/iCub_Software/yarp/build/share/yarp/resources] (YARP_DATA_DIRS)
||| checking [/home/ryan/Code/iCub_Software/icub-main/build/share/iCub/resources] (YARP_DATA_DIRS)
||| checking [/home/ryan/Code/iCub_Software/codyco-superbuild/build/install/share/codyco/resources] (YARP_DATA_DIRS)
||| finding paths [templates/applications]
||| checking [/home/ryan/Code/iCub_Software/icub-tutorials/src/motorControlAdvanced/build/templates/applications] (pwd)
||| checking [/home/ryan/.config/yarp/templates/applications] (YARP_CONFIG_HOME)
[... cut ...]
||| checking [/home/ryan/Code/iCub_Software/icub-main/build/share/iCub/templates/applications] (YARP_DATA_DIRS)
||| found /home/ryan/Code/iCub_Software/icub-main/build/share/iCub/templates/applications
||| checking [/home/ryan/Code/iCub_Software/codyco-superbuild/build/install/share/codyco/templates/applications] (YARP_DATA_DIRS)
||| found /home/ryan/Code/iCub_Software/codyco-superbuild/build/install/share/codyco/templates/applications
ryan@ISIR2014057:~/Code/iCub_Software/icub-tutorials/src/motorControlAdvanced/build$ 

```

alecive, 2015-05-11:
@rlober  `ymanager.ini` is not strictly necessary for the functioning of the GUI. It happens also in my machine but I have no problem whatsoever during my general use.

rlober, 2015-05-11:
@alecive 

You are totally right, it doesn't stop it from opening, but none of the compiled modules show up as a result. They are all in my `icub-main/build` dirs but they don't show up in `yarpmanager`. It is really just a convenience thing.

pattacini, 2015-05-11:
This info might be of help:

Straight out of the repo we don't have any `.xml` application in the `icub-main/build`, but rather `.xml.template` files the users need to customize to make them work for their specific architecture. The customized `.xml` are meant to be located in `~/.local/share/yarp/applications`, which should be in turn read at start-up by the `yarpmanager`.

rlober, 2015-05-11:
Ok that makes sense. I went ahead tested it out by customizing `iCubGui` and everything worked fine. Thanks for the help everyone.

closed 2015-05-11 by rlober