robotology/icub-main #150: iCubGui wrong name port

gregoire-pointeau, 2015-05-20:
Hello.

I try to launch several iCubGui with several names.

When I change the name through the config file (eg: name guiMentalOPC), the windows of the gui takes the good name (/guiMentalOPC), but the port created keep the default name (eg: /iCubGui/base:i instead of /guiMentalOPC/base:i). This happens even with only one OPC and one iCubGui launched.

[image]

Is it possible to fix it ? (I don't have the right for icub-main) 
Thank you

Grégoire Pointeau

pattacini, 2015-05-20:
Hi @gregoire-pointeau 

Apparently, ports names are hard coded, but you can use the `YARP_PORT_PREFIX` way. Just set up the env variable `YARP_PORT_PREFIX=/stem` prior to launching `iCubGui` and you'll get ports opened like in the following: `/stem/iCubGui/forces`, `/stem/iCubGui/head:i` ...

Importantly, within `yarpmanager` you have the `YARP_PORT_PREFIX` field available to fill in for each module.

gregoire-pointeau, 2015-05-20:
Ok, thanks Ugo, I'll use the yarp_port_prefix
Quick question, how do you fill it in an application xml file ?
I tried with <environment>YARP_PORT_PREFIX=/stem</environment> but it didn't work ?

pattacini, 2015-05-20:
You should use the **`<prefix> /stem </prefix>`** option as described in the [yarpmanager guide](http://wiki.icub.org/yarpdoc/yarpmanager.html).

gregoire-pointeau, 2015-05-20:
Ok, perfect !
Thank you Ugo

closed 2015-05-20 by gregoire-pointeau