robotology/icub-main #152: iKinCartesianSolver very slow

arturoribes, 2015-05-27:
Hi,

I am having problems with the cartesian solver being too slow. While using the cartesian controller to move the hand, I observed some jerky movements, so I checked the output of iKinCartesianSolver. I saw that the "computed in [s]" time was too high, from 0.05 to 0.5 seconds. 
This, for my kind of application, is unacceptably slow. So as sanity check, I executed iKinCartesianSolver in a different machine in our lab and, even though both machines have the same ipopt version, the other machine runs below 0.01 seconds. It is not an issue of machine power nor computational load. Both simulator and real icub (iCubLondon01) exhibit same behaviour.

Now I am back in my lab in Barcelona and I am running experiments there in simulator. I updated yarp and icub, and installed everything. I observed similar behaviour. Now I installed the newest version of ipopt and the problem is still the same, even worse! 

Is there any known version problem with ipopt and iKinCartesianSolver? Which versions work well? What about ipopt dependencies in icub cmake? I see that only the coin* ones are satisfied, no dl, gfortran, m, pthread or quadmath are found, even though my system has these.

pattacini, 2015-05-27:
Currently, we have never spotted any particular problem with `Ipopt` versions in our frameworks.

1. What are the steps did you follow to compile/install `Ipopt`?
2. Could you please report on the results of the smoke tests located in the [`icub-tutorials`](https://github.com/robotology/icub-tutorials)?

For (2), just clone the `icub-tutorials` repository, compile and then launch the `test_ipopt` binary. How long does `Ipopt` take to solve the smoke test?

arturoribes, 2015-05-27:
1) The steps in http://wiki.icub.org/wiki/Installing_IPOPT everything compiled ok and dependencies downloaded ok.
2) The results of the test are as follows:
<pre>
Ipopt: testing correct installation...

******************************************************************************
This program contains Ipopt, a library for large-scale nonlinear optimization.
 Ipopt is released as open source code under the Eclipse Public License (EPL).
         For more information visit http://projects.coin-or.org/Ipopt
******************************************************************************

target position = (-0.30000  0.10000  0.10000) [m]
solved position = (-0.29990  0.10002  0.10009) [m]
distance to target = 0.000140781 [m] ... test successful!
</pre>

However, this is a fragment of iKinCartesianSolver on my computer with iCub_SIM. 
<pre>
   Request type       = go
  Target rxPose   [m] = -0.397970	 0.141525	 0.049357	 0.000000	 1.000000	 0.000000	 3.141593
  Target txPose   [m] = -0.397376	 0.141280	 0.050130	 0.011737	 0.999927	 0.003016	 3.081985
Target txJoints [deg] =  3.604898	-48.936944	 10.000000	 100.000001	 80.000000	 62.489009	-66.431634	-50.171350	-1.912264
  norm(rxPose-txPose) = pos [m]: 0.00100499
                        ang [*]: 0.0608275
    computed in   [s] = 0.028301

   Request type       = go
  Target rxPose   [m] = -0.398649	 0.135173	 0.049394	 0.000000	 1.000000	 0.000000	 3.141593
  Target txPose   [m] = -0.398061	 0.134942	 0.050175	 0.010896	 0.999937	 0.002642	 3.081489
Target txJoints [deg] =  3.360570	-49.885375	 10.000000	 100.000001	 80.000000	 63.295215	-66.081698	-50.631139	-0.586207
  norm(rxPose-txPose) = pos [m]: 0.00100499
                        ang [*]: 0.061141
    computed in   [s] = 0.0690701

   Request type       = go
  Target rxPose   [m] = -0.399831	 0.124133	 0.049458	 0.000000	 1.000000	 0.000000	 3.141593
  Target txPose   [m] = -0.400056	 0.124208	 0.050435	-0.002487	 0.999988	 0.004231	 3.066469
Target txJoints [deg] =  0.000000	-32.706739	-49.407858	 56.105833	 68.801170	 15.647518	 7.134902	-15.928362	 1.280023
  norm(rxPose-txPose) = pos [m]: 0.001005
                        ang [*]: 0.0752834
    computed in   [s] = 0.379997

   Request type       = go
  Target rxPose   [m] = -0.409762	 0.031329	 0.050000	 0.000000	 1.000000	 0.000000	 3.141593
  Target txPose   [m] = -0.409250	 0.031940	 0.050612	-0.053358	 0.998327	-0.022273	 3.058806
Target txJoints [deg] =  9.373638	-50.000000	 10.000000	 100.000001	 39.606297	 77.072869	-17.924544	-35.392021	 40.000000
  norm(rxPose-txPose) = pos [m]: 0.00100499
                        ang [*]: 0.100993
    computed in   [s] = 0.0505941

   Request type       = go
  Target rxPose   [m] = -0.409762	 0.031329	 0.050000	 0.000000	 1.000000	 0.000000	 3.141593
  Target txPose   [m] = -0.409263	 0.031927	 0.050636	-0.054855	 0.998313	-0.019007	 3.058729
Target txJoints [deg] =  9.364534	-50.000000	 10.000000	 100.000001	 39.615507	 77.047492	-17.619830	-35.121028	 40.000000
  norm(rxPose-txPose) = pos [m]: 0.0010052
                        ang [*]: 0.101191
    computed in   [s] = 0.0229149
</pre>

Commands are sent as waypoints in a trajectory (controller code adapted from slidingController), so they are evenly distributed in a line and with the same trajectory time (0.2-0.3 seconds).
Notice how some are solved in 0.02-0.06 seconds, while one is solved in 0.38 seconds. 

How can I make that icub software install ipopt as dependency? That is, without me having to manually do it. Maybe I can try to remove every single trace of ipopt from my system and start from scratch...

pattacini, 2015-05-27:
I reckon that `Ipopt` is correctly installed on your machine. However, if you feel it's worth giving a try to the automatic installation of `Ipopt` as a dependency, you have to refer to this [page](http://wiki.icub.org/wiki/Linux:Installation_from_sources#Getting_all_dependencies) in our wiki. Be aware that this procedure will install all the dependencies and conflicts might arise with pre-installed/compiled packages.

Can we switch to a more standard test we have already available within the `icub-tutorials` bundle? It's called `tutorial-cartesian-interface` and does pretty much what you're trying to implement: setting up a trajectory defined as a collection of via-points, which in this case outline a circular path.

I've made a quick test on my windows laptop getting the results in the file [computation-time.log](https://files.zenhub.io/5565f10a345477bc7e657dcd).
And here comes the statistics about the solver's computation time:
[image]

As you can see we end up with a distribution of execution time, which is quite expected. Can you run exactly this test and report on your distribution?

I'm asking for this because it might be that the problem is not in the solver but in how you configured the control part and the generation of via-points. In this respect, you might find useful having a look at https://github.com/robotology/QA/issues/30.

arturoribes, 2015-05-27:
How did you get the computational time log?

pattacini, 2015-05-27:
I redirected the `iKinCartesianSolver`'s output to a file; something like the following:
```
$> iKinCartesianSolver --context simCartesianControl --part left_arm > computation-time.log
```

Then, I parsed that file seeking for "`computed in`" to extract the relevant information. A bit cumbersome in the end but it worked. If you like, you can send me directly the file and I'll do the rest.

arturoribes, 2015-05-28:
Well, I ran some tests and the results with the tutorial-cartesian-interface look pretty much the same as your histogram, although the range of values reaches 0.5 seconds.
Regarding my controller, results are not exactly the same. However, I realized that the problems arise when I command a long trajectory time (1 to 1.5 seconds) and for some configurations. 

Finally, after the whole morning testing, I found out that the problem was the elbow height. It seems that had some impact in the solver performance for some configurations (arm request positions + trajectory time). The weird thing is that in one of the computers of the lab it did not affect negatively.

Conclusion: it was some minor bug in my code but it is not 100% the cause, as the bug only affected to  when the code is executed in the simulator and there is a computer where the bug does not have any negative effect.

Anyway, thanks for the help. I will leave the issue as "currently overcame but too difficult to find the cause".

closed 2015-05-28 by pattacini