robotology/icub-main #153: Creating icub-basic-demo repository

pattacini, 2015-06-09:
Since the eager @alecive and @randaz81 - strongly motivated by @super-ste - made deep modifications to our **red-ball demo**, I'd like to streamline a bit the code of the demo and its organization before revising the upcoming PR.

What about then creating the `icub-basic-demo` repository where we can move the following modules (the repo name shouldn't be too general to give the impression that all the demos will be contained herein)?

`icub-main` old name | `icub-basic-demo` new name
:---: | :---:
`demoGraspManager` | `demoRedBall`
`demoForceControl` | `demoForceControl`
`demoForceImitation` | `demoForceImitation`
`iCubDemoY3` | `demoMovements`

/cc @lornat75 @vtikha @iron76 @traversaro @francesco-romano

vtikha, 2015-06-09:
:+1: :+1:

alecive, 2015-06-09:
:+1: :+1: :+1:

pattacini, 2015-06-09:
Started porting in [icub-basic-demos](https://github.com/robotology/icub-basic-demos).

gsaponaro, 2015-06-09:
Just curious: what is the motivation for having this separate repository as opposed to committing to **icub-main** the updated version of said demos?

I'm sure there are advantages. I don't know the details of these modifications to the demos, which perhaps warrant the need for a separate repository.

On the other hand, applications like demoGrasp will now be "sliced" between different repos, with the manager module inside **icub-basic-demos**, the other ones inside **icub-main**. Maintaining, installing or committing changes to this application will be slightly harder than before. Sorry to break the enthusiasm :smile: and please correct me if I'm wrong.

pattacini, 2015-06-09:
Maintaining smaller repositories is much easier and we have already populated `robotology` with a large number of them: we expect this number will still increase.

Applications made of components located in different repositories is perfectly normal and then we can rely on tools like [`ycm`](https://github.com/robotology/ycm) to perform the so-called **superbuild**. This is the current vision.

Finally, `icub-basic-demos` collects demos showing some basic capabilities of `icub-main` (as per its manifest), therefore no surprise that we are moving there the `demoGrasp` module since it does make use of the `Cartesian Interface` and the `pf3DTracker`.

pattacini, 2015-06-10:
Porting to [**`icub-basic-demos`**](https://github.com/robotology/icub-basic-demos) concluded.

I'd give a test run to `demoRedBall` and then remove old software in `icub-main`.

@alecive @randaz81 you can then start contributing with your PR!

pattacini, 2015-06-10:
A general announcement about the porting will follow soon on rc-hackers and [**QA**](https://github.com/robotology/QA/issues).

pattacini, 2015-06-11:
Old software removed from `icub-main`.

closed 2015-06-11 by pattacini

francesco-romano, 2015-06-25:
cc @reafrancesco