robotology/icub-main #154: INSTALL_WITH_RPATH and icub-cluster-run.sh : moving executable and linking error

maxime-petit, 2015-06-10:
Hi,

We are using the `INSTALL_WITH_RPATH` at `ON` in iCubLondon01 in order to have all the executable/lib/xml in separate, non-standard folders (/usr/local/src/install/robot/yarp or icub-main, ...).

However, when running the `icub-cluster-run.sh` script used with `icub-cluster.py`, the `yarprun` executable is copied to the `/tmp` folder, producing a link error:
`./yarprun: error while loading shared libraries: libYARP_OS.so.1: cannot open shared object file: No such file or directory`

The executables work fine when launched from their `CMAKE_INSTALL_PREFIX` directory.

Is there any reason why `yarprun` is copied to `/tmp`? If not, we are happy to create a pull request changing `icub-cluster-run.sh` to not copy `yarprun`, and instead run it in `CMAKE_INSTALL_PREFIX`.

By the way, this issue just emerged recently, as we did not use to need `INSTALL_WITH_RPATH` (although we always used non-standard installation directories).

Best :dancers: ,
Max and Tobi

Tobias-Fischer, 2015-06-23:
Hi all,
we just encountered this problem again, and as a temporary solution we put the `yarp` `lib` folder in `LD_LIBRARY_PATH`. It would be nice if that wouldn't be needed thought.

Best, Tobias

pattacini, 2015-06-23:
@Tobias-Fischer I think you could go ahead with the PR and put @lornat75 in cc's.
If I'm not mistaken, he's the author of that script and he knows all the details.

traversaro, 2015-06-23:
cc @rlober this reminds be of the strange behaviour we saw on iCubParis02 setup..

rlober, 2015-06-23:
@traversaro 
Yeah this reminds me of some of the difficulties we were having on the iCubParis02 cluster. In the end the easiest fix was to simply append the `LD_LIBRARY_PATH` in a shared `.bash_iCubrc`. Actually, @maxime-petit @Tobias-Fischer I am a little surprised that you managed to use `yarp`, installed in a non-standard directory, *without* having redirected both `LD_LIBRARY_PATH` and `PATH`? Can anyone clarify what `INSTALL_WITH_RPATH` does exactly?

Tobias-Fischer, 2015-06-24:
@rlober: It seems like setting `LD_LIBRARY_PATH` in the `.bashrc` is the easiest solution, however I usually don't like messing around with `LD_LIBRARY_PATH` (and it wasn't necessary up until recently). We always had to set the `PATH` variable accordingly.
As far as I understand, `INSTALL_WITH_RPATH` hard-codes the path to the libraries in the executables. Therefore, the executables can find the libraries without setting `LD_LIBRARY_PATH`. However, the path seems to be relative rather than absolute, which then causes trouble if you move the executable around, e.g. in the way it is done in `icub-cluster-run.sh`.
@lornat75: Can you remember why the `yarprun` executable is copied to `/tmp` before running it? I am happy to change it in a way that it just runs the executable in the `PATH`, however I don't want to break anything mysterious :P.

Best, Tobias

francesco-romano, 2015-06-24:
During the discussion I missed the point that the executable is moved around.
So, yes. RPATH is now relative (but anyway it should be used only during install and not build), because it assumes a proper installation (e.g. installdir/lib, installdir/bin, etc.) If this is no longer true because some bins are moved (and not the libs) then of course rpath does not work anymore.

As a side note the idea of rpath is to avoid to hardcode the library path in the executable, but to defer the path resolution at run time by giving some "hints".

lornat75, 2015-06-25:
There was no particular need for using tmp as far as I remember (there was maybe at some point the need to log some output in an easy way or something like that).

closed 2015-06-30 by pattacini