robotology/icub-main #155: Problem with yarpmotorgui and iKinCartesianSolver

arturoribes, 2015-06-10:
Hi,

We are experiencing issues when running iKinCartesianSolver in a remote machine. After the initialisation of the joint limits it blocks. Also, yarpmotorgui fails when trying to access to getEncoders(). Looking at the source code of iKin, I can see that getEncoders() is also called in a while loop, so I guess it is also trapped there. 
However, we found out that running them in ikart or pc104 they run correctly, so I don't know which may be the problem. Maybe it is somehow related to the RPATH issue from Max and Toby (issue #154).

Arturo.

Tobias-Fischer, 2015-06-10:
We just tried to use static libraries rather than shared libraries, and installed with `RPATH` off (in yarp and icub-main), and used `LD_LIBRARY_PATH` instead. The problem still persists ..

randaz81, 2015-06-10:
No, you simply have icub-main and yarp out of sync between the pc104 and your remote pc.
Please put again shared libraries, double check that all the machines are updated to the last version and recompile.

Clone of:
* https://github.com/robotology/yarp/pull/495
* https://github.com/robotology/yarp/issues/491

arturoribes, 2015-06-10:
It was that, needed to update ikart and pc104. Thanks!

closed 2015-06-10 by arturoribes