robotology/icub-main #164: Slow and delayed finger movement with direct control

Arikae, 2015-06-17:
Hi,

we are experiencing very slow finger movement even when using direct control with the iCub. 
There is a huge delay until our commands are executed and the movement itself is just a fraction of what our control signal tells it to do.
Does anyone know possible reasons for such a behavior?

Regards
Lars

pattacini, 2015-06-18:
Hi @Arikae 

Unfortunately, this might be happening whenever slack and friction start playing their fatal role in the actuation chain of the fingers.

You may find helpful these [SW countermeasures](https://github.com/robotology/QA/issues/2#issuecomment-69908682).
In this regard, in case you have a `ETH` iCub, there is a related https://github.com/robotology/icub-main/issues/132.

If the problem persists, it is likely that a maintenance of the tendons is required.

Arikae, 2015-06-18:
Thanks for the reply.
Do you know if iCub Darmstadt is ETH?

traversaro, 2015-06-18:
Yes, `iCubDarmstadt01` is an `ETH` iCub.

robertocalandra, 2015-06-18:
Hi @pattacini 

I can exclude that it is delay due to slack/friction.
In normal position control (with higher velocities) we achieve a much more responsive tracking.

We are doing some more tests and will get back to you soon with more questions ;)

pattacini, 2015-06-18:
An increased level of slack and/or friction usually causes those problems you're experiencing.

Basically, higher velocities make the moving reference (which follows a minimum-jerk profile) reach the final set-point earlier. As result, the PID seems to be more responsive only because the error gets amplified (the proportional gain is dominant here). This is an hint of the fact that the PID needs to be re-tuned because of the changed operative conditions (due to stiction/slack and whatsoever).

pattacini, 2015-09-02:
Closing for the meantime.
Feel free to reopen it once got fresh news.

closed 2015-09-02 by pattacini