robotology/icub-main #165: Hardware Fault on switch to Torque Mode

rlober, 2015-06-19:
On `iCubParis02` when I try to switch a joint into `torque mode` I get a `HARDWARE FAULT` error. Here is a snapshot of this:
[image]

Here is the terminal output (I tested on multiple joints):
```
[DEBUG]Opening interfaces...
joint: 0 in TORQUE mode!
joint: 0 in POSITION mode!
joint: 3 in TORQUE mode!
joint: 3 in POSITION mode!
joint: 3 in TORQUE mode!
joint: 3 in POSITION mode!
joint: 4 in TORQUE mode!
joint: 4 in POSITION mode!
joint: 3 in TORQUE mode!
joint: 3 in POSITION mode!
```

I don't get any error messages in `yarplogger`.

Any idea why this may be happening?

Ryan

traversaro, 2015-06-19:
You probably don't have a working torque feedback. Have you launched the `wholeBodyDynamics(Tree)` with the `--autoconnect` option?

rlober, 2015-06-19:
[image]

Is this what you are referring to? On little note: When I try to run `wholeBodyDynamics` on a cluster blade I get the following: 

```
[ERR] (iCubStartup) cannot run wholeBodyDynamics on icub-b1 : cannot run wholeBodyDynamics on /icub-b1
```

but when I run it on `icub-desktop` it works.

rlober, 2015-06-19:
Ok... Sorry, wholeBodyDynamics is not the same as wholeBodyDynamics(Tree). I am trying to get it to work right now, but I have deactivated the left leg so I have to reconfigure the .ini 

What all should I comment out to make it ignore the left leg?
Everything with "left_leg"?

rlober, 2015-06-19:
@traversaro @francesco-romano 

Just for my info, What is the difference between `wholeBodyDynamics` and `wholeBodyDynamicsTree`?

traversaro, 2015-06-19:
They are totally different. `wholeBodyDynamics` is the legacy module that does not support floating-base estimation of the forces (it assume that the robot is fixed base) while `wholeBodyDynamicsTree` is the new module that compute the floating base whole body dynamics.

rlober, 2015-06-22:
@traversaro 

Thank you for the clarification on the two modules. 

Can you help me with the following problem:
I have deactivated the left leg because of CAN issues, so I have reconfigured the `icub_all.xml` with all of the left_leg/foot entries commented out.

What all should I comment out in `wholeBodyDynamicsTree.ini` to make it ignore the left leg?

I tried changing just the `yarpWholeBodyInterface.ini` lines with left_leg 
ie: `ROBOT_LEFT_LEG_JOINTS = () `
and I get the following assertion error:

```
[INFO]Launching wholeBodyDynamicsThread with name :  wholeBodyDynamicsTree  and robotName  icub  and period  10 
[INFO] TorqueEstimationTree constructor: loaded urdf with 26dofs and 4 fts ( 4) 
[FATAL]Assertion failure (dof_serialization.size() == serial.getNrOfDOFs())
```

traversaro, 2015-06-22:
Sorry, the configuration of the `wholeBodyDynamicsTree` is a bit under documented. 
To remove a part you should:
  * Remove the joints of that part from the `ROBOT_DYNAMIC_MODEL_JOINTS` in the `yarpWholeBodyInterface.ini` (this are the joint considered for the dynamical model used for the torque estimation)
  * Remove the joints of the part from the list of published  torque ports in `wholeBodyDynamicsTree.ini`. This mean that in the `[WBD_OUTPUT_TORQUE_PORTS]` group, you can comment out the lines relative to the `left_leg` .

rlober, 2015-06-22:
@traversaro 

Great thanks man, I have some meetings this morning but I will give it a shot in the afternoon and let you know.

randaz81, 2015-06-22:
@traversaro Is a `--no_legs` flag available? (like in legacy WBD)

traversaro, 2015-06-22:
@randaz81 not at the moment.. the problem is wholeBodyDynamicsTree is agnostic of the underlyng model, that could have any structure. Having an option to disable just some part is useful for sure, but I have to think about it for how to correct implement it.

rlober, 2015-06-23:
Ok, so I tried commenting out the lines you mentioned @traversaro but I am still getting the same error. I have gone ahead and copied the relevent `.ini`'s and the output:

**`icub_all.xml`**
```xml
<?xml version="1.0" encoding="UTF-8" ?>
<robot name="icub">

    <!-- cartesian -->
    <devices file="cartesian/left_arm_cartesian.xml" />
    <devices file="cartesian/right_arm_cartesian.xml" />

    <!-- motor controllers wrappers -->
    <devices file="wrappers/motorControl/left_arm_mc_wrapper.xml" />
    <devices file="wrappers/motorControl/right_arm_mc_wrapper.xml" />
    <!-- <devices file="wrappers/motorControl/left_leg_mc_wrapper.xml" />  -->
    <devices file="wrappers/motorControl/right_leg_mc_wrapper.xml" />
    <devices file="wrappers/motorControl/head_mc_wrapper.xml" />
    <devices file="wrappers/motorControl/torso_mc_wrapper.xml" />
    <devices file="hardware/motorControl/icub_left_arm.xml" />
    <devices file="hardware/motorControl/icub_left_hand.xml" />
    <devices file="hardware/motorControl/icub_right_arm.xml" />
    <devices file="hardware/motorControl/icub_right_hand.xml" />
    <!-- <devices file="hardware/motorControl/icub_left_leg.xml" />  -->
    <devices file="hardware/motorControl/icub_right_leg.xml" />
    <devices file="hardware/motorControl/icub_head.xml" />
    <devices file="hardware/motorControl/icub_torso.xml" />

    <!-- VIRTUAL ANALOG SERVERs -->
    <devices file="wrappers/VFT/left_arm_VFT_wrapper.xml" />
    <!-- <devices file="wrappers/VFT/left_leg_VFT_wrapper.xml" />  -->
    <devices file="wrappers/VFT/right_arm_VFT_wrapper.xml" />
    <devices file="wrappers/VFT/right_leg_VFT_wrapper.xml" />
    <devices file="wrappers/VFT/torso_VFT_wrapper.xml" />

    <devices file="hardware/VFT/left_arm_virtual_strain.xml" />
    <!-- <devices file="hardware/VFT/left_leg_virtual_strain.xml" />  -->
    <devices file="hardware/VFT/right_arm_virtual_strain.xml" />
    <devices file="hardware/VFT/right_leg_virtual_strain.xml" />
    <devices file="hardware/VFT/torso_virtual_strain.xml" />

    <!-- REAL ANALOG SENSORS -->
    <devices file="wrappers/FT/left_arm_FT_wrapper.xml" />
    <!-- <devices file="wrappers/FT/left_leg_FT_wrapper.xml" />  -->
    <devices file="wrappers/FT/right_arm_FT_wrapper.xml" />
    <devices file="wrappers/FT/right_leg_FT_wrapper.xml" />
    <devices file="wrappers/MAIS/left_hand_mais_wrapper.xml" />
    <devices file="wrappers/MAIS/right_hand_mais_wrapper.xml" />
    <devices file="hardware/FT/left_arm_strain.xml" />
    <!-- <devices file="hardware/FT/left_leg_strain.xml" />  -->
    <devices file="hardware/FT/right_arm_strain.xml" />
    <devices file="hardware/FT/right_leg_strain.xml" />
    <devices file="hardware/MAIS/left_hand_mais.xml" />
    <devices file="hardware/MAIS/right_hand_mais.xml" />

    <!-- <devices file="wrappers/FT/left_foot_FT_wrapper.xml" /> -->
    <devices file="wrappers/FT/right_foot_FT_wrapper.xml" />
    <!-- <devices file="hardware/FT/left_foot_strain.xml" />  -->
    <devices file="hardware/FT/right_foot_strain.xml" />

    <!-- 6SG SENSORS -->
<!--    <devices file="wrappers/6SG/left_arm_SG_wrapper.xml" />
    <devices file="wrappers/6SG/left_lower_leg_SG_wrapper.xml" />
    <devices file="wrappers/6SG/left_upper_leg_SG_wrapper.xml" />
    <devices file="wrappers/6SG/right_arm_SG_wrapper.xml" />
    <devices file="wrappers/6SG/right_lower_leg_SG_wrapper.xml" />
    <devices file="wrappers/6SG/right_upper_leg_SG_wrapper.xml" />
    <devices file="wrappers/6SG/torso_SG_wrapper.xml" />
    <devices file="hardware/6SG/left_arm_SG.xml" />
    <devices file="hardware/6SG/left_lower_leg_SG.xml" />
    <devices file="hardware/6SG/left_upper_leg_SG.xml" />
[... 350 lines cut ...]
yarp: Receiving input from /icub/right_arm/analog:o to /wholeBodyDynamicsTree/ftSens/r_arm_ft_sensor:i using udp
yarp: Port /wholeBodyDynamicsTree/ftSens/r_leg_ft_sensor:i active at tcp://10.0.0.40:11094
yarp: Datagram packet size set to 65000
yarp: Receiving input from /icub/right_leg/analog:o to /wholeBodyDynamicsTree/ftSens/r_leg_ft_sensor:i using udp
yarp: Port /wholeBodyDynamicsTree/ftSens/r_foot_ft_sensor:i active at tcp://10.0.0.40:11095
yarp: Datagram packet size set to 65000
yarp: Receiving input from /icub/right_foot/analog:o to /wholeBodyDynamicsTree/ftSens/r_foot_ft_sensor:i using udp
yarp: Port /wholeBodyDynamicsTree/imu/imu_frame:i active at tcp://10.0.0.40:11096
yarp: Datagram packet size set to 65000
yarp: Receiving input from /icub/inertial to /wholeBodyDynamicsTree/imu/imu_frame:i using udp
[INFO]Launching wholeBodyDynamicsThread with name :  wholeBodyDynamicsTree  and robotName  icub  and period  10 
[INFO] TorqueEstimationTree constructor: loaded urdf with 26dofs and 4 fts ( 4) 
[FATAL]Assertion failure (dof_serialization.size() == serial.getNrOfDOFs())
/usr/local/src/robot/yarp/build-x86_64/lib/libYARP_OS.so.1(_ZNK4yarp2os3Log5fatalEPKcz+0x143) [0x7fdffce36173]
/usr/local/src/robot/codyco-superbuild/build/install/lib/x86_64-linux-gnu/libidyntree-yarp.so.0.3.5(_ZN4iCub8iDynTree20TorqueEstimationTree27TorqueEstimationConstructorERN3KDL4TreeESt6vectorIN13kdl_format_io12FTSensorDataESaIS7_EES5_ISsSaISsEESB_RN4yarp3sig6VectorESF_Ssj+0x553) [0x7fdffa8fdebb]
/usr/local/src/robot/codyco-superbuild/build/install/lib/x86_64-linux-gnu/libidyntree-yarp.so.0.3.5(_ZN4iCub8iDynTree20TorqueEstimationTreeC1ESsSt6vectorISsSaISsEES4_Ssj+0x44e) [0x7fdffa8fd284]
wholeBodyDynamicsTree(_ZN23wholeBodyDynamicsThreadC2ESsSsiPN7yarpWbi20yarpWholeBodySensorsERN4yarp2os8PropertyEbSsb+0xc48) [0x4338d8]
wholeBodyDynamicsTree(_ZN23wholeBodyDynamicsModule9configureERN4yarp2os14ResourceFinderE+0x21b4) [0x468a54]
/usr/local/src/robot/yarp/build-x86_64/lib/libYARP_OS.so.1(_ZN4yarp2os8RFModule9runModuleERNS0_14ResourceFinderE+0xa) [0x7fdffceb489a]
wholeBodyDynamicsTree(main+0x7b4) [0x4275a4]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf5) [0x7fdff6d87b45]
wholeBodyDynamicsTree() [0x42848e]
icub@icub-desktop:~$ 

```

traversaro, 2015-06-23:
Ah.. sorry, I forgot that I had hardcoded that assumption that the number of joint used for estimation is the same of the URDF.. this should be definitely be fixed (ref: https://github.com/robotology-playground/idyntree/issues/48 ). 

The easiest workaround for now is to change the type of all the joints belonging to the left leg in the urdf used by the yarpWholeBodyInterface (in https://github.com/robotology-playground/yarp-wholebodyinterface/tree/master/app/robots/iCubParis02) from `revolute` to `fixed`.

rlober, 2015-06-23:
Ok so I finally got it working and yes running `wholeBodyDynamicsTree --autocomplete` fixed the `HARWARE FAULT` issue.

**Summary of the solution:**
Given that parts have been commented out in `icub_all.xml`
-  In `yarpWholeBodyInterface.ini` remove these parts in `ROBOT_DYNAMIC_MODEL_JOINTS` and in `ROBOT_MAIN_FTS`
- In `wholeBodyDynamicsTree.ini` remove the parts in `WBD_OUTPUT_TORQUE_PORTS` and `WBD_SUBTREES`
- In `model.urdf` change the joints of the part from `revolute` to `fixed`. As @traversaro mentioned - this will not be necessary in the future.

Below I have put the files for your reference. The files that are loaded are typically located under `$CODYCO_SUPERBUILD_ROOT/build/install/share/codyco/robots/[your_robot_name]/`. So you need to modify their sources for this to take permanent effect.

You can see the files I have modified for `iCubParis02` here:

[`wholeBodyDynamicsTree_no_left_leg.ini`](https://github.com/robotology/codyco-modules/blob/eaaf7b32e1b0b6c005acda5d83e0d297faee2230/src/modules/wholeBodyDynamicsTree/app/robots/iCubParis02/wholeBodyDynamicsTree_no_left_leg.ini)

[`yarpWholeBodyInterface_no_left_leg.ini`](https://github.com/robotology-playground/yarp-wholebodyinterface/blob/3e38a69e263e40c5258208eee5c543b5da7382e0/app/robots/iCubParis02/model_fixed_left_leg.urdf)

[`model_fixed_left_leg.urdf`](https://github.com/robotology-playground/yarp-wholebodyinterface/blob/3e38a69e263e40c5258208eee5c543b5da7382e0/app/robots/iCubParis02/yarpWholeBodyInterface_no_left_leg.ini)

Once you have rebuilt superbuild with these files you just run:
```
wholeBodyDynamicsTree --from wholeBodyDynamicsTree_no_left_leg.ini --autoconnect
```

closed 2015-06-23 by rlober