robotology/icub-main #176: Documentation on installation from sources on Windows is broken

traversaro, 2015-07-16:
The tutorial at http://wiki.icub.org/wiki/Windows:_installation_from_sources was broken by a recent modification by @lornat75 [1], that removed the instruction on how to install `icub-main` dependencies. 
I can revert the old version, but it seems it make sense to remove some YARP related installation instructions and leave just the iCub ones. 

@lornat75 can I proceed and restore the iCub dependencies installation or you were planning to put them somewhere else, like the YARP one?

cc @fstramandinoli

[1] http://wiki.icub.org/index.php?title=Windows:_installation_from_sources&diff=prev&oldid=21166

lornat75, 2015-07-17:
Should be fixed now.

traversaro, 2015-07-17:
Yes, thanks!

closed 2015-07-17 by traversaro