robotology/icub-main #180: robotInterface warning cleanup

randaz81, 2015-07-20:
when robotInterface is launched, there are several warning due to the fact that most of the robots in icub-main/app do not have updated configuration files.
On of the most annoying is the one due to recent change in: `yarp\src\libYARP_dev\src\modules\AnalogWrapper`:
which generates the message:
`
    yWarning() <<   " AnalogServer device:\n"
                    "************************************************************************************************\n"
                    "  AnalogServer is using deprecated parameters for port name! It should be:\n"
                    "       name:         full name of the port, like /robotName/deviceId/sensorType:o\n"
                    "       period:       refresh period of the broadcasted values in ms (optional, default 20ms)\n"
                    "************************************************************************************************";`
Please close this issue when all the robot configuration files have been updated.

Tobias-Fischer, 2015-09-28:
Hi @randaz81 and @davidetome,
in addition to the above mentioned warning, which we also see, we also have the following warnings in `iCubLondon01`. It would be great if they could get fixed as well:

```312		2.450918		WARNING		maxPosStep: Using default MaxPosStep=10 degs``` (several times)
```328		2.663231		WARNING		SharedCanBus [1] using default thread period = 10ms``` (also for [5], [2], [6], and [0])
```678		3.961549		WARNING		Set baud rate not allowed from CanBusAccessPoint implementation```
```983		6.270142		WARNING		CanBusSkin id list contains more than one entry -> devices will be merged.  ```
```1057		8.733364		WARNING		Device  left_arm_mc is not implementing a yarp::dev::IRemoteCalibrator, therefore it cannot attach to a Calibrator device. ``` (also for right_arm_mc, left_hand_mc, right_hand_mc)

Thank you!
Tobi and @maxime-petit

Tobias-Fischer, 2015-12-11:
Hi @randaz81 and @davidetome,
are there any news on this issue?

Best,
Tobias

barbalberto, 2015-12-11:
About the last point regarding calibration, I think the calibrator config file in the 'action' part sould look like:
```
	<action phase="startup" level="10" type="calibrate">
	    <param name="target">left_arm_mc_wrapper</param>
	</action>

	<action phase="interrupt1" level="1" type="park">
	    <param name="target">left_arm_mc_wrapper</param>
	</action>

	<action phase="interrupt3" level="1" type="abort" />
```

i.e. the calibrator should attach to the wrapper, not to the motion control device.
Try this change and see if the waarning goes away.

Tobias-Fischer, 2015-12-13:
Hi @barbalberto,
Thanks for the hint! Unfortunately, the robot did not start up properly if `left_arm_mc` was changed to `left_arm_mc_wrapper` (and the same for `right_arm`, `left_hand` and `right_hand`). The calibration for the arms didn't start in this case. I did not investigate it further, as it's "just" a warning at the end, but it doesn't seem to work out of the box.

Thanks again! Tobi

barbalberto, 2015-12-14:
Which error did you get after the change?

Tobias-Fischer, 2015-12-14:
I didn't even check the error messages .. it was just a quick test, and we needed to do some "actual" work just afterwards ;).
We'll give it a try again in the new year or so.

Best, Tobi

Tobias-Fischer, 2016-01-26:
This issue has been open for over 6 months now .. Are there any plans to fix the robot config files? Also see https://github.com/robotology/icub-main/issues/273 which reports errors additionally to the warnings reported in this issue.

Best, Tobias
/cc @randaz81 @davidetome

Tobias-Fischer, 2016-05-04:
Other related issues:
https://github.com/robotology/icub-main/commit/387e03c21d32cd53f6fabadeb72443fb04888193
https://github.com/robotology/QA/issues/108#issuecomment-214743789

As it seems that only the configuration files of some robots are being updated, can we have a central place to collect all the necessary changes? In that way, the maintainer of the individual robots can keep their config files up-to-date without having to go through the git history (as I do every now and then ...).

Best, Tobi

/cc @randaz81 @davidetome @drdanz

randaz81, 2016-05-04:
we are tracking all this changes in http://wiki.icub.org/wiki/ChangeLog @drdanz @pattacini

pattacini, 2016-05-04:
Hope that the new workflow will be out very soon and properly advertised to catch/avoid all these glitches.

Tobias-Fischer, 2016-05-04:
Cool thanks this is good to know :)

pattacini, 2016-09-05:
Issue moved to [robotology/yarp #902](https://github.com/robotology/yarp/issues/902) via [**ZenHub**](https://www.zenhub.com/)

closed 2017-03-28 by pattacini