robotology/icub-main #192: Move pf3d* modules into icub-basic-demos repository

pattacini, 2015-09-02:
`pf3dBottomup` and `pf3dTracker` modules can be conveniently moved into https://github.com/robotology/icub-basic-demos repository.

/cc @gsaponaro @alecive

gsaponaro, 2015-09-02:
OK, cool.
I can also clean up and update a bit the code (i.e., use RFModule and latest yInfo macros).
What does IFI mean?

pattacini, 2015-09-02:
@gsaponaro great, you're willing to help clean those modules, you're more than welcome. Go ahead with a **PR** then, and I'll take care of moving them to their final destination.

To find out more on the **IFI Day** events, visit [image] and if you like it you can ask to join us :smile:

gsaponaro, 2015-09-10:
Hi @pattacini, regarding **pf3dBottomup** which you just upgraded (cool), I just wanted to mention that there are issues with that module stability, for me it crashes upon ctrl+c, seemingly due to a memory leak in function ```calc_hist_model_2D()```. I'm not sure whether this hinders moving the module to another repo, probably not. We can fix it in the new location too.

pattacini, 2015-09-10:
We don't use `pf3dBottomup` module so I can't help.
However it can be fixed conveniently in the new location, as you said.

pattacini, 2015-09-11:
https://github.com/robotology/icub-main/commit/6aace36329f884c2b8fe0c41259fa9665eeb7f89 moves **`pf3d*`** modules into [**`icub-basic-demos`**](https://github.com/robotology/icub-basic-demos) with https://github.com/robotology/icub-basic-demos/commit/7cc8bebccd5310e1958a143366afea13298653bc

closed 2015-09-11 by pattacini