robotology/icub-main #193: Refining the content of the repository

pattacini, 2015-09-02:
In the attempt of reducing the amount of components, I suggest to:

- [ ] move `pf3d*` modules in [**`icub-basic-demos`**](https://github.com/robotology/icub-basic-demos), see #192.
- [ ] move `canBusSniffer` in [**`icub-tests`**](https://github.com/robotology/icub-tests).
- [ ] move `focusTool` in [**`icub-tests`**](https://github.com/robotology/icub-tests).
- [ ] move `testStereoMatch` in [**`icub-tests`**](https://github.com/robotology/icub-tests).
- [ ] move `joystickControl` and `joystickCheck` in [**`ikart`**](https://github.com/robotology/ikart).
- [ ] move `robotInterface` in [**`yarp`**](https://github.com/robotology/yarp).
- [ ] move `iCubInterfaceGuiLib` and `iCubInterfaceGui` in [**`robotology-legacy`**](https://github.com/orgs/robotology-legacy).
- [ ] get rid finally of all `gtk` based modules.
- [ ] get rid finally of the `robotMotorGui`.
- [ ] merge together `camCalib` and `dualCamCalib`.

lornat75, 2015-09-02:
About testStereoMatch, canBusSniffer and focusTool: these are not automatic tests but guis. I would see more as test tools or something like that.

I like the idea of moving stuff out of icub-main; but we need to:
1) discuss what should remain in icub-main, i.e. what is the content of this repository? and move everything else out
2) set-up a superbuild that allows us to perform automatic tests on a dashboard otherwise we end up missing many errors

pattacini, 2015-09-02:
I was proposing `icub-tests` as just first thought. Is there any more suitable place for them? Don't know.

1. **"What is the content of icub-main?"** Well, very difficult to answer indeed. At time being the only guidelines I tried to follow were in the direction to move out _superseded_ modules, clearly _self-contained_ libraries, or _never/very rarely_ used components (in terms of applications we know about). Of course, we can brainstorm about the _mission_ of our beloved `icub-main`.
2. Fine with me.

pattacini, 2017-09-16:
@drdanz what's the status of **gtk modules**?
Can we declare them finally banned?

drdanz, 2017-09-19:
@pattacini I don't know...

* We should have a new qt firmware tool but I've never used it

Perhaps we should schedule a quick meeting on the topic?

By the way
* icub-cluster was included in yarpmanager
* We now have an sdl joypad device in YARP
* We are considering moving the usb camera (and perhaps check if we have other generic devices) in YARP

pattacini, 2017-09-19:
Thank you @drdanz for the quick update.
I'll highlight above the further steps you outlined to keep track.

marcoaccame, 2017-09-20:
Hi @drdanz and @pattacini, about the `We should have a new qt firmware tool but I've never used it `:
we have had *FirmwareUpdater* in devel since december 2016 and in master since june. It replaces the GTK-based *canLoader* and *ethLoader*. See https://github.com/robotology/QA/issues/240.

pattacini, 2017-09-20:
Great @marcoaccame 
So, shall we get shot of GTK once and for good? 😄

marcoaccame, 2017-09-20:
For what regards *canLoader* and *ethLoader*: yes we can.

See however issue https://github.com/robotology/icub-main/issues/473 which I have just created to help  P&M (production & maintenance ) dropping the use of *canLoader* from an internal procedure of theirs.

traversaro, 2017-09-20:
Probably this is worth having its own issues, but there are several points of the documentation that should be updated:
* https://github.com/robotology/icub-main/blob/cd2028daac5afcddb220f3b6a7bfe6caa4abdf7b/src/libraries/icubmod/canBusSkin/CanBusSkin.cpp#L964
* https://github.com/robotology/icub-firmware-shared/search?utf8=%E2%9C%93&q=canLoader&type=
* https://github.com/robotology/icub-firmware-shared/search?utf8=%E2%9C%93&q=ethLoader&type=
* https://github.com/robotology/icub-firmware-build/search?utf8=%E2%9C%93&q=canLoader&type=
* https://github.com/robotology/icub-firmware-build/search?utf8=%E2%9C%93&q=ethLoader&type=
* https://github.com/robotology/icub-firmware/search?utf8=%E2%9C%93&q=ethLoader&type=
* https://github.com/drdanz/icub-firmware/search?utf8=%E2%9C%93&q=canLoader&type=
* http://wiki.icub.org/wiki/CanLoader
* http://wiki.icub.org/index.php?title=Special%3ASearch&profile=default&search=CanLoader&fulltext=Search
* http://wiki.icub.org/wiki/Firmware

cc @lornat75

marcoaccame, 2017-09-21:
I had a look at the links. There are several cases, hence I write some comments in here as an help to do proper changes when we remove GTK and related GUIs. 

1. The `icub-main/src/tools/canLoader/canLoaderLib` and `icub-main/src/tools/ethLoader/ethLoaderLib` libraries will remain because they are used also by the new FirmwareUpdater.
2. Name of variables/type in C/C++ code: in most cases it is wrong to modify them because they refer to protocols of the canLoaderLib / ethLoaderLib.
3. Online prints, such as yWarning(),  telling a robot's user to check vs a particular FW version: I would change the string `canLoader` with `FW maintenance tools`. In this way the user can refer to the proper section of the wiki (see later point).
4. Comments in C/C++ code: we could do the same string substitution as in point 3 but I don't see it so important to do.
5. Documentation on wiki: there will be robots with GTK GUI (only) for a while, hence it is not correct to remove any canLoader/ethLoader section. Now the wiki already mentions the double possibility of 
 use of canLoader/ethLoader or FirmwareUpdater. What I would do is to edit the wiki to add a section *FW maintenance tools* where more focus is given to operations with FirmwareUpdater but which still contains references to canLoader+ethLoader.

traversaro, 2019-09-27:
@pattacini for more fine-grained handling of this issue, I converted this issue in an epic and opened separate issues for each element left.

closed 2023-07-30 by pattacini