robotology/icub-main #196: Improve switching to torque control mode

francesco-romano, 2015-09-09:
There are currently to issues when switching the a control board into torque mode:

- initial references.
It seems that at the current state a reference of zero torque is set. While this is ok with position and velocity control, the choice of an initial reference for a torque control is more subtle. A proposal is to keep as a reference the same torque read at the moment of switching.
- delay in the change of control mode.
We are experiencing a big (and visible) delay while we switch control mode. The nature of this delay is still to be determined. 
The code responsible of the switch at high level (balancing) is the following: https://github.com/robotology-playground/yarp-wholebodyinterface/blob/master/src/yarpWholeBodyActuators.cpp#L372-L426

cc: @randaz81 @DanielePucci @traversaro

traversaro, 2015-09-09:
We should use the multijoint version of setControlMode and setRefTorque to see if this delay is reduced.

alecive, 2015-09-09:
+1 @traversaro

pattacini, 2016-09-02:
@francesco-romano any update happened to these two points in the meantime?

francesco-romano, 2016-09-02:
The first issue has been solved by @ale-git  almost one year ago.
For the second one, we need to test again everything, so I think it is safe to close this issue for now, and reopen it if needed.

closed 2016-09-02 by francesco-romano