robotology/icub-main #205: Shaking iCub - update of config files

Tobias-Fischer, 2015-10-07:
Hi all,

as reported in https://github.com/robotology/icub-support/issues/30, the config files of various robots have to be updated. Firstly, all robots should use `TORQUE_CONTROL`, and the old source for the other parameters should be deleted (or at least the bug should be corrected). Furthermore, for robots mounted on the iKart, the parameters `Kp`, `Ki`, and `Kd` need to be changed. It would be great if @jypuigbo could report the values which are used on `iCubBarcelona01` now.

Best,
Tobi

pattacini, 2015-10-07:
I've marked this as **urgent**.
I think @randaz81 knows how to deal with it.

super-ste, 2015-10-09:
In order not to change the PID paramenters ('ki', 'kp', 'kd') you can lower the power supply voltage to a value closer to 38.5 -- 39 V and exclude the battery from the setup

You will have to lower the PIDs if you plan to use the iKart autonomously and with the battery. In such case the voltage is set by the battery (>53V when fully charged).

@randaz81 @pattacini @jypuigbo correct me if I'm wrong

Tobias-Fischer, 2015-10-10:
Hi @super-ste, as we are using the iKart with the battery every now and then, changing the power supply voltage is unfortunately not an option for us.

@jypuigbo could you please report the values you use for `ki`, `kp` and `kd`?

Thanks, Tobi

pattacini, 2015-10-10:
@Tobias-Fischer the PID output is given in terms of machine ticks in the range `[-max_output, max_output]`, where `max_output` (historically set equal to `1333` ticks) corresponds to the voltage full scale, i.e. `39 V` or `53 V` depending on the current power supply setting.

Therefore, multiplying the values of all your PID gains by a factor of `39/53` should do the job in a way that has at least a bit of rationale, whatever rules others might have followed to cut their gains. Hence, I would recommend you to start with this.

Further, I would also suggest to modify directly those values in the repositories (now you can), to avoid future problems.

Tobias-Fischer, 2015-10-12:
Hi,

I just pushed the modified configuration files for `iCubLondon01` (see https://github.com/robotology/icub-main/commit/900b82e4484a3917218c091d3267311428f543a5). The robot seems to run fine with the new values.

However, I found that all values of `ki` and `kd` are set to 0, which makes the PID controller to a P controller, isn't it? Or are default values used if the values are 0?

I'll leave this ticket open until the `TRQ_PIDS` issue has been resolved.

Best, Tobi

pattacini, 2015-10-12:
@Tobias-Fischer 
I'd say that also `POS_PIDS` should be affected by the gain reduction.

Apparently, when in torque mode there's only a `P` controller active, yep.

Tobias-Fischer, 2015-10-12:
Hi Ugo @pattacini, thanks for pointing that out! I totally missed that these values need to be changed for the `POS_PIDS` as well, as I've only discovered the bug for `TRQ_PIDS`. I'll change the config files accordingly tomorrow.

Tobias-Fischer, 2015-10-13:
Fixed in https://github.com/robotology/icub-main/commit/eaa9fb95d40a0c89e0d3a498b4b3edf06ab1a4f0. Thanks @pattacini!

randaz81, 2015-10-13:
@pattacini This topic will be discussed in IFI Day3.

randaz81, 2015-10-22:
Hi @Tobias-Fischer, i'm going to revert your modifications to `icub_all.xml`.
That file has a special meaning, it's not only parsed by robotInterface but also by a tool which updates the iit database. I am thus pushing to the repo the following solution: I created a new file `icub_ikart.xml` which disables the legs, and I modified the pointer contained in `robotInterface.ini`
Btw, we encourage to use the pull request workflow for critical directories such as `icub-main/app/robots`. Thank you!

closed 2015-10-22 by randaz81

Tobias-Fischer, 2015-10-22:
Hi @randaz81,
Thanks for your changes. We've had this setup locally for a long, long time, and I didn't know about this tool you are talking about. We'll make sure to make our changes in `icub_ikart.xml` in the future.

Thanks,
Tobias