robotology/icub-main #208: in xml file rename maxPwm in maxoutput

valegagge, 2015-10-14:
In Pid configuration maxpwm means max pid output.
whats about limPWM param???

pattacini, 2015-10-14:
As per #205 (just an example) or the experience I had yesterday along with @drdanz, @barbalberto, @vtikha and `Bluey`, changing robot config files without warning users is very likely to become a **wasps' nest**.

We should handle this with extreme care.

randaz81, 2015-10-14:
`limPWM` means limited PWM.
if `useLimitedPWM` is set to `true` in `general.xml`, motion control will send to the boards this value instead of `maxPwm`.
When parsing the xml file, motion control is currently checking that `limPWM` is < `maxPwm` and produce an error otherwise.
This functionality was meant to replace the old *safe*.ini files of iCubInterface but was never used in practice.
It's ok to rename also this parameter (carefully, as Ugo said) but it is mostly likely that this functionality will be suppressed or refactored (indeed, it's better to limit the PWM than the PID output. In this case, the parameter is not only renamed but also moved to the limits section, as it acts on the motor and not on the pid outputs)

valegagge, 2015-10-28:
Randaz: "New spec: `limPWM` will not be renamed since it will act ot the motor (units will % of max PWM) and it will be moved from from the PID section to Limits"

valegagge, 2016-01-26:
Renamed only maxPwm in maxOutput

Added "motor pwm limit" in LIMITS group params

closed 2016-01-26 by valegagge