robotology/icub-main #222: [MacOS] icub_SIM windows is not displayed

iron76, 2015-10-23:
While installing YARP and iCub on El Capitan I am experiencing a strage behaviour. Even if the installation was successful, when running the icub_SIM everything works properly but the window with the 3D view of the iCub is not popping out. Cameras and control boards can be accessed. If I right click on the dock-icon I can see that the process is not responding, but I think that this was the case also when the simulator was properly working.

iron76, 2015-10-27:
This has been replicated in another machine different from mine, therefore labelled replicable. Details of the installation (e.g. dependencies) are in the following scripts:

- YARP: https://github.com/iron76/Teaching/blob/master/scripts/mac-os-yarp-install.sh
- iCub: https://github.com/iron76/Teaching/blob/master/scripts/mac-os-icub-install.sh

claudiofantacci, 2016-01-02:
Good evening,
I'm a YARP/iCub beginner on MacOSX El Capitan (10.11.2) and I've followed the installation guide in http://wiki.icub.org/wiki/Homebrew_Mavericks.
As iron76 already reported, the iCub_SIM simulator is correctly compiled, but fails to show any GUI while the CPU stuck between 120-140% on an Intel i7 processor with 4 physical cores (8 logical).

After running the iCub_SIM from command line, the simulator seems to be stuck after opening the *face/eyelids* port. The console debug log stops providing any messages after reporting
```
yarp: Port /icubSim/face/eyelids active at tcp://192.168.1.12:10049
```

Here is the **WindowServer** log once the application is being run:
```
02/01/16 21:08:13,168 WindowServer[188]: Received display connect changed for display 0x4280642
02/01/16 21:08:13,179 WindowServer[188]: Found 27 modes for display 0x04280642 [27, 0]
02/01/16 21:08:13,192 WindowServer[188]: Received display connect changed for display 0x3f003f
02/01/16 21:08:13,193 WindowServer[188]: Found 1 modes for display 0x003f003f [1, 0]
02/01/16 21:08:13,193 WindowServer[188]: Received display connect changed for display 0x3f0040
02/01/16 21:08:13,193 WindowServer[188]: Found 1 modes for display 0x003f0040 [1, 0]
02/01/16 21:08:13,193 WindowServer[188]: Received display connect changed for display 0x3f0041
02/01/16 21:08:13,193 WindowServer[188]: Found 1 modes for display 0x003f0041 [1, 0]
02/01/16 21:08:13,194 WindowServer[188]: MPAccessSurfaceForDisplayDevice: Set up page flip mode on display 0x04280642 device: 0x7fab4141db00  isBackBuffered: 1 numComp: 3 numDisp: 3
02/01/16 21:08:13,225 WindowServer[188]: CGXMuxAcknowledge: Posting glitchless acknowledge
02/01/16 21:08:13,293 WindowServer[188]: Received display connect changed for display 0x4280642
02/01/16 21:08:13,294 WindowServer[188]: Found 1 modes for display 0x04280642 [1, 0]
02/01/16 21:08:14,341 WindowServer[188]: disable_update_timeout: UI updates were forcibly disabled by application "iCub_SIM" for over 1.00 seconds. Server has re-enabled them.
02/01/16 21:08:28,341 WindowServer[188]: disable_update_likely_unbalanced: UI updates still disabled by application "iCub_SIM" after 15.00 seconds (server forcibly re-enabled them after 1.00 seconds). Likely an unbalanced disableUpdate call.
```
followed by
```
02/01/16 21:11:03,539 WindowServer[188]: _CGXRemoveWindowFromWindowMovementGroup: window 0xd5 is not attached to window 0xe1
02/01/16 21:11:05,039 WindowServer[188]: _CGXRemoveWindowFromWindowMovementGroup: window 0xd5 is not attached to window 0xe1
02/01/16 21:11:08,690 WindowServer[188]: _CGXRemoveWindowFromWindowMovementGroup: window 0xd5 is not attached to window 0xe1
02/01/16 21:11:10,307 WindowServer[188]: _CGXRemoveWindowFromWindowMovementGroup: window 0xd5 is not attached to window 0xe1
02/01/16 21:11:16,774 WindowServer[188]: _CGXRemoveWindowFromWindowMovementGroup: window 0xd5 is not attached to window 0xe1
```
sometimes I also get
```
02/01/16 20:41:52,111 WindowServer[188]: common_reenable_update: UI updates were finally reenabled by application "iCub_SIM" after 19.48 seconds [0.05fps] (server forcibly re-enabled them after 1.00 seconds [1.00fps])
```
but nothing really happens, the GUI does not show off.

tkelestemur, 2016-05-11:
I also have same problem with El Capitan. This is what I get when I run iCub_SIM:

```
yarp: Port /icubSim/face/eyelids active at tcp://192.168.110.102:10049
[WARNING]the simulation is too slow to run in real-time, you should increase the timestep in ode_params.ini (current value: 25, suggested value: 28)
[WARNING]the simulation is too slow to run in real-time, you should increase the timestep in ode_params.ini (current value: 25, suggested value: 31)
[WARNING]the simulation is too slow to run in real-time, you should increase the timestep in ode_params.ini (current value: 25, suggested value: 32)
[WARNING]the simulation is too slow to run in real-time, you should increase the timestep in ode_params.ini (current value: 25, suggested value: 31)
[WARNING]the simulation is too slow to run in real-time, you should increase the timestep in ode_params.ini (current value: 25, suggested value: 31)
```

Is anyone working on that bug?

closed 2017-03-28 by pattacini