robotology/icub-main #223: Immediately after positionMove, checkMotionDone returns true

valegagge, 2015-10-23:
Running MotorTest I found this bug:

the test sends  position move command and immediately after checks if checkMotionDone's return value is false. This check fails because checkMotionDone returns the cached value updated by broadcasted stream.

I fixed this bug in this way: checkMotionDone function asks to firmware if motion is done and returns fw's response; therefore checkMotionDone is blocking, that is blocks the called thread until fw returns a response regardless of its value (true o false).

This change is on branch embOMC_checkMotionDone_blocking.
 
Currently I test this change on headV3 and it works properly. 
Now i'm going to test it on blu robot.

valegagge, 2015-10-23:
I tested the changes on icubBlu and don't work property on joints controlled by mc4 because motiondone variable is not a proxy variable.

In conlcusion, I may merge branch in master in order to fix the bug on ems and mc4plus board.

To fix bug for joints controlled by mc4 board more work is necessary.

valegagge, 2015-10-23:
merge done with SHA 130b8c2fb015f860d19aa9750f748f5021716e41

marcoaccame, 2015-11-06:
I will open an issue in icub-firmware.

valegagge, 2016-07-28:
The solution of issue robotology/icub-firmware#32  fixed this problem.

closed 2016-07-28 by valegagge