robotology/icub-main #232: problem in installing icub-main

azadm, 2015-10-30:
This is my brew config:
````
HOMEBREW_VERSION: 0.9.5
ORIGIN: https://github.com/Homebrew/homebrew.git
HEAD: 67f14d41dc3c47f622e70f0e9a76751671b86a20
Last commit: 25 hours ago
HOMEBREW_PREFIX: /usr/local
HOMEBREW_REPOSITORY: /usr/local
HOMEBREW_CELLAR: /usr/local/Cellar
HOMEBREW_BOTTLE_DOMAIN: https://homebrew.bintray.com
CPU: 8-core 64-bit haswell
OS X: 10.9.5-x86_64
Xcode: 6.2
CLT: N/A
Clang: 6.0 build 600
X11: 2.7.7 => /opt/X11
System Ruby: 2.0.0-p481
Perl: /usr/bin/perl
Python: /usr/bin/python
Ruby: /usr/bin/ruby => /System/Library/Frameworks/Ruby.framework/Versions/2.0/usr/bin/ruby
Java: 1.6.0_65-b14-462
````

and this the error that I get:

````
[ 48%] Built target iCubSkinGui-gtk
Scanning dependencies of target frameGrabberGui2-gtk
[ 48%] Building CXX object src/tools/frameGrabberGui2-gtk/CMakeFiles/frameGrabberGui2-gtk.dir/src/fggui2_main.cpp.o
In file included from /Users/azadm/Applications/icub-main/src/tools/frameGrabberGui2-gtk/src/fggui2_main.cpp:42:
In file included from /usr/local/include/gtkmm-2.4/gtkmm.h:87:
In file included from /usr/local/include/glibmm-2.4/glibmm.h:87:
In file included from /usr/local/include/glibmm-2.4/glibmm/thread.h:49:
In file included from /usr/local/include/glibmm-2.4/glibmm/error.h:23:
In file included from /usr/local/include/glibmm-2.4/glibmm/exception.h:25:
/usr/local/include/glibmm-2.4/glibmm/ustring.h:267:13: error: expected ';' at end of declaration list
  ~ustring() noexcept;
            ^
            ;
/usr/local/include/glibmm-2.4/glibmm/ustring.h:881:48: warning: deleted function definitions are a C++11 extension [-Wc++11-extensions]
  FormatStream(const ustring::FormatStream&) = delete;
                                               ^
/usr/local/include/glibmm-2.4/glibmm/ustring.h:882:59: warning: deleted function definitions are a C++11 extension [-Wc++11-extensions]
  FormatStream& operator=(const ustring::FormatStream&) = delete;
                                                          ^
/usr/local/include/glibmm-2.4/glibmm/ustring.h:894:18: error: expected ';' at end of declaration list
  ~FormatStream() noexcept;
                 ^
                 ;
/usr/local/include/glibmm-2.4/glibmm/ustring.h:1273:45: warning: deleted function definitions are a C++11 extension [-Wc++11-extensions]
  Stringify(const ustring::Stringify<T>&) = delete;
                                            ^
/usr/local/include/glibmm-2.4/glibmm/ustring.h:1274:59: warning: deleted function definitions are a C++11 extension [-Wc++11-extensions]
  Stringify<T>& operator=(const ustring::Stringify<T>&) = delete;
                                                          ^
/usr/local/include/glibmm-2.4/glibmm/ustring.h:1290:51: warning: deleted function definitions are a C++11 extension [-Wc++11-extensions]
  Stringify(const ustring::Stringify<ustring>&) = delete;
                                                  ^
/usr/local/include/glibmm-2.4/glibmm/ustring.h:1291:71: warning: deleted function definitions are a C++11 extension [-Wc++11-extensions]
  Stringify<ustring>& operator=(const ustring::Stringify<ustring>&) = delete;
                                                                      ^
/usr/local/include/glibmm-2.4/glibmm/ustring.h:1309:55: warning: deleted function definitions are a C++11 extension [-Wc++11-extensions]
  Stringify(const ustring::Stringify<const char*>&) = delete;
                                                      ^
/usr/local/include/glibmm-2.4/glibmm/ustring.h:1310:75: warning: deleted function definitions are a C++11 extension [-Wc++11-extensions]
  Stringify<ustring>& operator=(const ustring::Stringify<const char*>&) = delete;
                                                                          ^
/usr/local/include/glibmm-2.4/glibmm/ustring.h:1328:51: warning: deleted function definitions are a C++11 extension [-Wc++11-extensions]
  Stringify(const ustring::Stringify<char[N]>&) = delete;
                                                  ^
[... 189 lines cut ...]
                    ^
                    ;
/usr/local/include/sigc++-2.0/sigc++/functors/slot.h:642:14: warning: rvalue references are a C++11 extension [-Wc++11-extensions]
  slot2(slot2&& src) noexcept
             ^
/usr/local/include/sigc++-2.0/sigc++/functors/slot.h:642:21: error: expected ';' at end of declaration list
  slot2(slot2&& src) noexcept
                    ^
                    ;
/usr/local/include/sigc++-2.0/sigc++/functors/slot.h:737:14: warning: rvalue references are a C++11 extension [-Wc++11-extensions]
  slot3(slot3&& src) noexcept
             ^
/usr/local/include/sigc++-2.0/sigc++/functors/slot.h:737:21: error: expected ';' at end of declaration list
  slot3(slot3&& src) noexcept
                    ^
                    ;
/usr/local/include/sigc++-2.0/sigc++/functors/slot.h:835:14: warning: rvalue references are a C++11 extension [-Wc++11-extensions]
  slot4(slot4&& src) noexcept
             ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
34 warnings and 20 errors generated.
make[2]: *** [src/tools/frameGrabberGui2-gtk/CMakeFiles/frameGrabberGui2-gtk.dir/src/fggui2_main.cpp.o] Error 1
make[1]: *** [src/tools/frameGrabberGui2-gtk/CMakeFiles/frameGrabberGui2-gtk.dir/all] Error 2
make: *** [all] Error 2
````

Tobias-Fischer, 2015-10-30:
Hi, see https://github.com/robotology/yarp/issues/598

Best, Tobias

azadm, 2015-10-30:
Thanks @Tobias-Fischer 

I added `-std=c++11` to `CMAKE_CXX_FLAGS` but now I get this error:

````
[ 93%] Built target camCalib
Scanning dependencies of target camCalibWithPose
[ 93%] Building CXX object src/modules/camCalibWithPose/CMakeFiles/camCalibWithPose.dir/src/main.cpp.o
In file included from /Users/azadm/Applications/icub-main/src/modules/camCalibWithPose/src/main.cpp:131:
/Users/azadm/Applications/icub-main/src/modules/camCalibWithPose/include/iCub/CalibToolFactory.h:78:13: warning: delete called on
      'CalibToolFactory' that is abstract but has non-virtual destructor [-Wdelete-non-virtual-dtor]
            delete group[i];
            ^
In file included from /Users/azadm/Applications/icub-main/src/modules/camCalibWithPose/src/main.cpp:134:
/Users/azadm/Applications/icub-main/src/modules/camCalibWithPose/include/iCub/CamCalibModule.h:49:10: error: no template named 'map' in
      namespace 'std'; did you mean 'max'?
    std::map<double, yarp::os::Bottle> h_encs_map;
    ~~~~~^~~
         max
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/algorithm:2600:1: note: 'max'
      declared here
max(const _Tp& __a, const _Tp& __b, _Compare __comp)
^
In file included from /Users/azadm/Applications/icub-main/src/modules/camCalibWithPose/src/main.cpp:134:
/Users/azadm/Applications/icub-main/src/modules/camCalibWithPose/include/iCub/CamCalibModule.h:49:10: error: non-friend class member 'max'
      cannot have a qualified name
    std::map<double, yarp::os::Bottle> h_encs_map;
    ~~~~~^
/Users/azadm/Applications/icub-main/src/modules/camCalibWithPose/include/iCub/CamCalibModule.h:49:10: error: C++ requires a type specifier for
      all declarations
    std::map<double, yarp::os::Bottle> h_encs_map;
    ~~~  ^
/Users/azadm/Applications/icub-main/src/modules/camCalibWithPose/include/iCub/CamCalibModule.h:49:13: error: expected ';' at end of
      declaration list
    std::map<double, yarp::os::Bottle> h_encs_map;
            ^
            ;
/Users/azadm/Applications/icub-main/src/modules/camCalibWithPose/include/iCub/CamCalibModule.h:50:10: error: no template named 'map' in
      namespace 'std'; did you mean 'max'?
    std::map<double, yarp::os::Bottle> t_encs_map;
    ~~~~~^~~
         max
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/algorithm:2600:1: note: 'max'
      declared here
max(const _Tp& __a, const _Tp& __b, _Compare __comp)
^
In file included from /Users/azadm/Applications/icub-main/src/modules/camCalibWithPose/src/main.cpp:134:
/Users/azadm/Applications/icub-main/src/modules/camCalibWithPose/include/iCub/CamCalibModule.h:50:10: error: non-friend class member 'max'
      cannot have a qualified name
    std::map<double, yarp::os::Bottle> t_encs_map;
    ~~~~~^
/Users/azadm/Applications/icub-main/src/modules/camCalibWithPose/include/iCub/CamCalibModule.h:50:10: error: C++ requires a type specifier for
      all declarations
    std::map<double, yarp::os::Bottle> t_encs_map;
    ~~~  ^
/Users/azadm/Applications/icub-main/src/modules/camCalibWithPose/include/iCub/CamCalibModule.h:50:13: error: expected ';' at end of
      declaration list
    std::map<double, yarp::os::Bottle> t_encs_map;
            ^
            ;
/Users/azadm/Applications/icub-main/src/modules/camCalibWithPose/include/iCub/CamCalibModule.h:51:10: error: no template named 'map' in
      namespace 'std'; did you mean 'max'?
    std::map<double, yarp::os::Bottle> imu_map;
    ~~~~~^~~
         max
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/algorithm:2600:1: note: 'max'
      declared here
max(const _Tp& __a, const _Tp& __b, _Compare __comp)
^
In file included from /Users/azadm/Applications/icub-main/src/modules/camCalibWithPose/src/main.cpp:134:
[... 36 lines cut ...]
/Users/azadm/Applications/icub-main/src/modules/camCalibWithPose/include/iCub/CamCalibModule.h:83:83: error: use of undeclared identifier
      'h_encs_map'
    void setHeadEncoders(double time, const yarp::os::Bottle &h_encs) { m.lock(); h_encs_map[time] = h_encs; m.unlock(); }
                                                                                  ^
/Users/azadm/Applications/icub-main/src/modules/camCalibWithPose/include/iCub/CamCalibModule.h:84:75: error: use of undeclared identifier
      'imu_map'
    void setImuData(double time, const yarp::os::Bottle &imu) { m.lock(); imu_map[time] = imu; m.unlock(); }
                                                                          ^
/Users/azadm/Applications/icub-main/src/modules/camCalibWithPose/src/main.cpp:157:19: error: no matching member function for call to
      'runModule'
    return module.runModule(rf);
           ~~~~~~~^~~~~~~~~
/Users/azadm/Applications/yarp/src/libYARP_OS/include/yarp/os/RFModule.h:89:17: note: candidate function not viable: no known conversion from
      'CamCalibModule' to 'yarp::os::RFModule' for object argument
    virtual int runModule(yarp::os::ResourceFinder &rf);
                ^
/Users/azadm/Applications/yarp/src/libYARP_OS/include/yarp/os/RFModule.h:80:17: note: candidate function not viable: requires 0 arguments, but
      1 was provided
    virtual int runModule();
                ^
1 warning and 19 errors generated.
make[2]: *** [src/modules/camCalibWithPose/CMakeFiles/camCalibWithPose.dir/src/main.cpp.o] Error 1
make[1]: *** [src/modules/camCalibWithPose/CMakeFiles/camCalibWithPose.dir/all] Error 2
make: *** [all] Error 2
````

Tobias-Fischer, 2015-10-30:
1) Do not just add `-std=c++11` to `CMAKE_CXX_FLAGS`, but use `YARP_EXPERIMENTAL_CXX11` instead (or something similar for iCub, if available).

2) Some of the errors might go away by adding `#include <map>`, for others I have no idea. Maybe @drdanz might be able to help?

Best, Tobias

azadm, 2015-10-30:
Unfortunately, I do not see any flag similar to `YARP_EXPERIMENTAL_CXX11` (including `CXX11`) in the list.

jeljaik, 2015-10-30:
First press `t` to see the advanced options, then you'll find `YARP_EXPERIMENTAL_CXX11`. Hitting `?` you can search for a particular option.

Tobias-Fischer, 2015-10-30:
@azadm: Did you try adding `#include <map>` after 
```
 // std
#include <stdio.h>
```
in `src/modules/camCalibWithPose/include/iCub/CamCalibModule.h`? Can you please report the error after adding this line?

azadm, 2015-10-30:
Thanks @Tobias-Fischer 
Yes adding that line made it work.

Tobias-Fischer, 2015-10-30:
Fixed in https://github.com/robotology/icub-main/commit/26f11b1448f9299e06539bd21ca8ce95f3648a9f

How about adding an `ICUB_EXPERIMENTAL_CXX11` to fix the other issue as well?

Best, Tobi

CC @drdanz @pattacini @lornat75

closed 2015-10-30 by Tobias-Fischer

drdanz, 2015-10-31:
@Tobias-Fischer I'm working on a solution in YARP that will fix the issues with GTK and c++11 without enabling c++11 in all the code but it will be enabled only when required (unfortunately c++11 is yet quite untested and therefore the "experimental" flag).
I already merged a patch in `FindGTK2.cmake` upstream, that will be available with CMake 3.5, and that will fix the issue when using the targets. Unfortunately, even importing the updated `FindGTK2.cmake`, it won't work for us for 2 different reasons: 1) We are not using the targets but the variables and 2) `target_compile_features()` and `INTERFACE_COMPILE_FEATURES` property were added in CMake 3.1 and even CMake will not fail, c++11 will not be enabled with older versions (we are still depending on 2.8.9, see [YARP Supported Distributions](http://wiki.icub.org/wiki/YARP_Supported_Distributions)).

Unfortunately, everything GTK related has a very low priority, since we are moving to Qt5 everything, anyway I'm very close to a solution that will work with all CMake versions that we support (but it requires another patch to `FindGTK2.cmake` that should be ported upstream, and a few other changes around the code. As soon as it is ready, I will apply it to icub-main as well.

 cc @lornat75, robotology/yarp#598

drdanz, 2015-10-31:
@Tobias-Fischer thanks for the fix to the camCalibWithPose module by the way...

drdanz, 2015-11-02:
GTK2 and c++ should no longer be an issue, c++11 does not require to be explicitly enabled anymore