robotology/icub-main #233: Command 'iCub_SIM' giving error

muhammadalidildar, 2015-10-30:
Dear all, I want to test iCub and trying command 'iCub_SIM' in a terminal window. It gives the following error:
"iCub_SIM: symbol lookup error: iCub_SIM: undefined symbol: _ZTIN4yarp3dev22ImplementMotorEncodersE"

Whereas YARP is running fine. Please help me a little to figure out the problem.
Thank you very much in advance.

pattacini, 2015-11-10:
No one has still tackle this issue to help @muhammadalidildar.
This is a good test for young people to help the community.

@massimoregoli @GiuliaP @tanismar @jeljaik @Ommac @bhigy please take this up.

jeljaik, 2015-11-11:
@muhammadalidildar assuming you're on Linux can you post the output of `ldd iCub_SIM` to see against which shared libraries it's trying to link?

muhammadalidildar, 2015-11-11:
Yes I am running it on Ubuntu. "ldd iCub_SIM" is giving the following error:
ldd: ./iCub_SIM: No such file or directory

Tobias-Fischer, 2015-11-11:
Try `ldd "$(which iCub_SIM)"`

jeljaik, 2015-11-11:
Yes sorry, forgot to say you have to do that in the directory where iCub_SIM has been installed, so, either in:
`$ICUB_DIR/bin`
or wherever you installed icub-main

muhammadalidildar, 2015-11-11:
I am very much thankful to all of you. I guess I did something wrong while installing. I installed YARP from sources and iCub from binaries. So I uninstalled and reinstalled both from binaries. Now they are working fine.

closed 2015-11-11 by muhammadalidildar

pattacini, 2015-11-11:
@muhammadalidildar glad to know that you found a solution, even though in principle there shouldn't be any problem with the setting `yarp(sources)` + `icub(binaries)`.

thanks @jeljaik for your commitment! :+1: