robotology/icub-main #234: check setControlMode() duration

randaz81, 2015-11-02:
@DanielePucci @francesco-romano
it seems that following piece of code:
```
for all_joints  
{
   setControlMode (j)
}
for all_joints 
{
   setRefTorque (j)
}
```
causes a glitch on the robot.

The following piece of code could fix the issue:
```
for all_joints 
{
   setControlMode (j)
   setRefTorque (j)
}
```
however, it's better to check what is happening at the low level.

lornat75, 2015-11-03:
It seems to be a race condition in the firmware that happens when time passes between setControlMode() and setRefTorque(), but not if the two are called in tight sequence. I would add a test before fixing it.

barbalberto, 2015-11-03:
As I always stress it is always better to use the 'all joint' version or use the 'group version' instead of for loop.

With for loop (first case) the delay between the first setControlMode and the first setRefTorque is ***njoints * "time of one rpc call"*** which is known to be slow since it is blocking.
With torque mode it is particularly critic since the the firmeware goes in hardware fault if it does not get any reference soon enough.

So my opinion from this snippet, this is not a problem of how much time the rpc takes, it is just wrong user code.
The habit of using the all joints will remove those and similar problems from root

ale-git, 2015-11-03:
Moreover, the multiple call for coupled joints in the waist or shoulders is redundant.

francesco-romano, 2015-11-03:
We were told that `setControlMode`, even in the group call, is anyway blocking, due to the role it has.
Thus calling multiple or single does not solve the issue.

If this is not true, then of course the code must be updated to use the multi-call

DanielePucci, 2015-11-10:
@francesco-romano  and I have recently carried on some quantitative tests on the `setControlMode` implemented as @randaz81 has suggested, i.e.

```
for all_joints 
{
   setControlMode (j)
   setRefTorque (j)
}
```

These are the results, which show really bad performances:
~~~
*************************************************
Test 1 (from position to torque)
*************************************************

SetControlMode[Body part: 0, Axis: 2] 6404  us
SetControlMode[Body part: 0, Axis: 1] 4730  us
SetControlMode[Body part: 0, Axis: 0] 3161  us
SetControlMode[Body part: 1, Axis: 0] 19845 us
SetControlMode[Body part: 1, Axis: 1] 6489  us
SetControlMode[Body part: 1, Axis: 2] 7339  us
SetControlMode[Body part: 1, Axis: 3] 19128 us
SetControlMode[Body part: 2, Axis: 0] 5029  us
SetControlMode[Body part: 2, Axis: 1] 4296  us
SetControlMode[Body part: 2, Axis: 2] 5418  us
SetControlMode[Body part: 2, Axis: 3] 3565  us
SetControlMode[Body part: 3, Axis: 0] 19940 us
SetControlMode[Body part: 3, Axis: 1] 19870 us
SetControlMode[Body part: 3, Axis: 2] 19494 us
SetControlMode[Body part: 3, Axis: 3] 19245 us
SetControlMode[Body part: 3, Axis: 4] 21044 us
SetControlMode[Body part: 3, Axis: 5] 8404  us
SetControlMode[Body part: 4, Axis: 0] 20546 us
SetControlMode[Body part: 4, Axis: 1] 22529 us
SetControlMode[Body part: 4, Axis: 2] 18310 us
SetControlMode[Body part: 4, Axis: 3] 19139 us
SetControlMode[Body part: 4, Axis: 4] 6373  us
SetControlMode[Body part: 4, Axis: 5] 18563 us
-------------------------------------------------
Total = 299 ms
Mean  =  13 ms

*************************************************
SetControlMode[Body part: 0, Axis: 2] 23721 us
SetControlMode[Body part: 0, Axis: 1] 4866  us
SetControlMode[Body part: 0, Axis: 0] 9804  us
SetControlMode[Body part: 1, Axis: 0] 22082 us
SetControlMode[Body part: 1, Axis: 1] 3638  us
SetControlMode[Body part: 1, Axis: 2] 4526  us
SetControlMode[Body part: 1, Axis: 3] 18947 us
SetControlMode[Body part: 2, Axis: 0] 20154 us
SetControlMode[Body part: 2, Axis: 1] 2727  us
SetControlMode[Body part: 2, Axis: 2] 6413  us
SetControlMode[Body part: 2, Axis: 3] 20867 us
SetControlMode[Body part: 3, Axis: 0] 19608 us
SetControlMode[Body part: 3, Axis: 1] 19579 us
SetControlMode[Body part: 3, Axis: 2] 19740 us
SetControlMode[Body part: 3, Axis: 3] 20365 us
SetControlMode[Body part: 3, Axis: 4] 19933 us
SetControlMode[Body part: 3, Axis: 5] 19785 us
SetControlMode[Body part: 4, Axis: 0] 19925 us
SetControlMode[Body part: 4, Axis: 1] 17563 us
SetControlMode[Body part: 4, Axis: 2] 18648 us
SetControlMode[Body part: 4, Axis: 3] 19875 us
SetControlMode[Body part: 4, Axis: 4] 19482 us
SetControlMode[Body part: 4, Axis: 5] 17895 us
-------------------------------------------------
Total = 370 ms
Mean  =  16 ms

*************************************************
Test 2 (from torque to position)
*************************************************

SetControlMode[Body part: 0, Axis: 2] 6066  us
SetControlMode[Body part: 0, Axis: 1] 3661  us
SetControlMode[Body part: 0, Axis: 0] 5384  us
SetControlMode[Body part: 1, Axis: 0] 19694 us
SetControlMode[Body part: 1, Axis: 1] 5506  us
SetControlMode[Body part: 1, Axis: 2] 4767  us
SetControlMode[Body part: 1, Axis: 3] 7883  us
SetControlMode[Body part: 2, Axis: 0] 19844 us
SetControlMode[Body part: 2, Axis: 1] 5484  us
SetControlMode[Body part: 2, Axis: 2] 4163  us
SetControlMode[Body part: 2, Axis: 3] 20722 us
SetControlMode[Body part: 3, Axis: 0] 19362 us
SetControlMode[Body part: 3, Axis: 1] 19373 us
SetControlMode[Body part: 3, Axis: 2] 22051 us
SetControlMode[Body part: 3, Axis: 3] 19039 us
SetControlMode[Body part: 3, Axis: 4] 18428 us
SetControlMode[Body part: 3, Axis: 5] 20941 us
SetControlMode[Body part: 4, Axis: 0] 19612 us
SetControlMode[Body part: 4, Axis: 1] 20974 us
SetControlMode[Body part: 4, Axis: 2] 20018 us
SetControlMode[Body part: 4, Axis: 3] 19708 us
SetControlMode[Body part: 4, Axis: 4] 19882 us
SetControlMode[Body part: 4, Axis: 5] 19907 us
-------------------------------------------------
Total = 342 ms
Mean  =  15 ms

*************************************************
SetControlMode[Body part: 0, Axis: 2] 17488 us
SetControlMode[Body part: 0, Axis: 1] 4513  us
SetControlMode[Body part: 0, Axis: 0] 4770  us
SetControlMode[Body part: 1, Axis: 0] 19371 us
SetControlMode[Body part: 1, Axis: 1] 6072  us
SetControlMode[Body part: 1, Axis: 2] 3859  us
SetControlMode[Body part: 1, Axis: 3] 19584 us
SetControlMode[Body part: 2, Axis: 0] 20276 us
SetControlMode[Body part: 2, Axis: 1] 5130  us
SetControlMode[Body part: 2, Axis: 2] 3520  us
SetControlMode[Body part: 2, Axis: 3] 8044  us
SetControlMode[Body part: 3, Axis: 0] 18738 us
SetControlMode[Body part: 3, Axis: 1] 23294 us
SetControlMode[Body part: 3, Axis: 2] 16874 us
SetControlMode[Body part: 3, Axis: 3] 23136 us
SetControlMode[Body part: 3, Axis: 4] 19107 us
SetControlMode[Body part: 3, Axis: 5] 20675 us
SetControlMode[Body part: 4, Axis: 0] 19448 us
SetControlMode[Body part: 4, Axis: 1] 20086 us
SetControlMode[Body part: 4, Axis: 2] 20317 us
SetControlMode[Body part: 4, Axis: 3] 18893 us
SetControlMode[Body part: 4, Axis: 4] 19721 us
SetControlMode[Body part: 4, Axis: 5] 20092 us
-------------------------------------------------
Total = 353 ms
Mean  =  15 ms
~~~

@traversaro @iron76 @lornat75 @randaz81 @barbalberto @ale-git @marcoaccame

lornat75, 2015-11-10:
I think there are two separate issues here, 

The original issue mentions a "glitch"; this is a bug irrespective of the time it take by setControlMode to process the request.

A separate problem is the evaluation of the time taken by setControlMode.

DanielePucci, 2015-11-10:
Yes, we do agree. The `glitch` has been solved by the fixes at the firmware level made by @ale-git. Before setting the control mode, a reading of the torques is done, and then this reading is set as torque reference. The performance are incredibly good. We no longer note the switch from position to torque control mode @francesco-romano.

traversaro, 2015-11-10:
@randaz81 to do a recap, the current  policy to set the initialized the desired "quantity" in the low level control loop is:  

|  Control Mode |  Initial desired value                    | 
|----------|------------------|
| Position                 | Measured encoder value             |
| Velocity                 | 0                                                | 
| Torque                  | Measured torque value                | 
| OpenLoop             | 0 | 

There is an open issue in gazebo plugins to match the behavior of the real robots in simulaton with respect to this : https://github.com/robotology/gazebo-yarp-plugins/issues/111#issuecomment-58874062 .

lornat75, 2015-11-10:
@ale-git @DanielePucci it is worth mentioning the commit from this issue then.

Can we change the title of this issue to "glitch in setControlMode" and open another one about the duration of the rpc call. I have the feeling the reported duration is in line with the periods of the involved thread but we should check more carefully.

ale-git, 2015-11-10:
@lornat75 
The commit that fixes the glitch is 23a4b700509aa346f84dd5722e1aedade24f5d3b

barbalberto, 2015-11-11:
@ale-git is it in yarp or where?

jeljaik, 2015-11-16:
I believe @ale-git meant:
https://github.com/robotology/icub-firmware/commit/23a4b700509aa346f84dd5722e1aedade24f5d3b

barbalberto, 2016-08-04:
Can we close this issue?

pattacini, 2016-09-05:
Feel free to open it again if there's news.

closed 2016-09-05 by pattacini