robotology/icub-main #235: Skin ports remain up after closing robotInterface

DanielePucci, 2015-11-04:
I am currently working on the purpley (iCubGenova02), and after closing `robotInterface` some ports related to the skin stay up.

Feel free to close this issue if it is a duplicate of some other issue.

[image]

@marcoaccame @pattacini @drdanz @francesco-romano

DanielePucci, 2015-11-04:
@barbalberto

pattacini, 2015-11-11:
Another issue which seems suitable for young researchers. It would be helpful to exercise their debugging abilities and also for them to get more feeling with the gut of our system.

@tanismar @massimoregoli @GiuliaP @bhigy @giuliavezzani @jeljaik and whoever wants to be remembered as the quest's winner.

pattacini, 2015-11-16:
@bhigy would you be willing to take up this task?

bhigy, 2015-11-18:
I'll try to have a look when I find a moment.

pattacini, 2015-11-18:
Great, thanks.

lornat75, 2015-11-22:
The first thing I would check is if ports are created dynamically or statically. If they are created dynamically (i.e. using new) check that close (and destructors) call delete.

giuliavezzani, 2015-11-23:
If @bhigy is too busy in this period, I can try to fix the problem the next
month,  as soon as all the bureaucratic tasks will be finished :)

barbalberto, 2015-11-26:
If I remember correctly, the skin wrapper create dinamically a set of analogServers using ``new``.
Maybe something is not right in the cleanup.
Both classes are located in the icub-main/icubmod, if someone wants to address this issue I can help giving a couple of hints for a quick start.

commit 9764e6bd (2015-12-07): Merge pull request #250 from giuliavezzani/master

fixed #235 on closing Skin ports
closed 2015-12-07 by pattacini