robotology/icub-main #237: Red ball demo failing on iCubGenova01

jeljaik, 2015-11-09:
Hello guys,

The red ball demo is failing everytime `demoRedBall` is launched with the arms going into hardware fault. It works for a couple of seconds before the arms crumble. The following links are for the logs of the `robotInterface`, `demoRedBall` and `gravityCompensator`. The latter, due to some messages it displays regarding a slow network.

https://drive.google.com/file/d/0B6zDGh11iY6ocnkzUUVrWWNuUnM/view?usp=sharing
https://drive.google.com/file/d/0B6zDGh11iY6oV1hkeTRuTHloSWc/view?usp=sharing
https://drive.google.com/file/d/0B6zDGh11iY6obS04XzQxNzVaSFE/view?usp=sharing

Any help would be greatly appreciated!

julijenv, 2015-11-10:
Hi @jeljaik ,
I didnt see the demo running so i dont know if the hardwre fault you mentionned can be seen on yarpmotorgui only, but on the log of robotInterface there isnt any hardware fault mentionned.
The only redundant error message is :
_AnalogWrapper: left_foot: Sensor returned overflow error (code 2)_
but we also have one instance or 2 of those folowing messagess:
_AnalogWrapper: right_leg: Sensor returned error with code 1_
and
_AnalogWrapper: right_arm: Sensor returned timeout error (code 3)_
and
_AnalogWrapper: left_arm: Sensor returned error with code 1_
I dont know if those specific error shutdown the motors of the arms?
if the software engineer can say why both right and left foot pop this error message and what are supposed to be the behaviour of the robot when the message is boradcast, it would be helpful!

jeljaik, 2015-11-10:
I strongly believe there's some network issue in that setup as reported by `gravityCompensator`. With @vtikha we proceeded to launch the Red Ball Demo opening just one camera (left eye), no skin and it was working fine for a while. After stopping the demo and restarting after a few minutes the problem appeared again with both arms going in hardware fault.

alecive, 2015-11-10:
I agree with @jeljaik . It is something that showed up frequently in the last month or so, and basically it makes the robot unusable.
If I recall correctly, the arms go to `HW FAULT` because the `wholeBodyDynamics` doesn't get data for more than `100ms`. Is it true @jeljaik ?

iron76, 2015-11-10:
This is super-serious since the black iCub has always been our flagship for demos. I recommend opening a suitable issue in the right channels (cub-support) and addressing the issue as soon as possible.

barbalberto, 2015-11-10:
Putting togheter all error/warning messages I saw today, namely
gravityCompensator: network delay
AnalogWrapper: error reading from sensor
can buff overrun

my guess of what it's happening is the following:

the flow of data from the robot to the wholebody to gravitycomp then back to the robot is somewhere taking too much time and the torque references on the firmware are not updated with the correct frequency, so the watchdog triggers and the joints go in hardware fault.

To verify this hypotesis, is the firmware sending some special error code if it goes in HW FAULT for this reason? @randaz81 @ale-git 

If this hypotesis is true then I see 2 possible causes:
1) the CFW2 not dispatching the can messages fast enough filling the queue and losing messages or creatnig delays
2) network doesn't work well creating big queue of packets somwhere and then sending burst of them. This will create black spot (triggering the watchdog) followed by burst of messages (causing the overrun on CAN)

vtikha, 2015-11-10:
Maybe it is worth a try as a first step to check the cisco router and see if all the machines have full bandwidth and not slowing down things.

francesco-romano, 2015-11-11:
@DanielePucci and I tried a `iperf` two weeks ago (at Festival della Scienza) and everything was running at 1Gbps.

pattacini, 2015-11-11:
@francesco-romano `iperf` is not sufficient unfortunately. The proper test is to connect to the router and check the status of each single port.

barbalberto, 2015-11-11:
Also, I think an effect similar to network delay could be caused by the process not scheduled fast enough, so delaying the reading of packets

iron76, 2015-11-11:
Just as an additional debugging info: the problem was experienced by @francesco-romano even at the Festival della Scienza, using another network switch.

ale-git, 2015-11-11:
@barbalberto I can talk about ethernet robots, where if a joint is in compliant interaction mode or torque control mode, and the torque measure is missing for more than 200 ms, then the joint goes in hardware fault state and the error is signaled.

vtikha, 2015-11-12:
Yesterday I managed to check the full system of iCubGenova01, installation on all machines, specially looking at network performances etc and found out that the problem (at least one of them) seems to be coming from the `pc104`. When the robot is in motion, at some point it loses the ethernet connection very briefly (~1s) which can been seen visually as a blockage of the stream of images but also blocks while reading the `state:o` ports. This drop can also be seen in the following graph:

[image]

This brief network drop naturally causes the `wholeBodyDynamics` to go in a sort of `safe mode` which then faults the arms. 
I have spoken with @julijenv and we will have a look at all relevant connections (crimps etc) on the `pc104` asap.

jeljaik, 2015-11-12:
Thank you @vtikha

pattacini, 2015-11-12:
:+1: great debugging!

iron76, 2015-11-12:
This seems promising but not yet a solution. What's next? Should we check the PC104 connection? The ethernet cable has been often problematic. For our demos this sudden lack of feedback is problematic.

vtikha, 2015-11-12:
@iron76 this is part of the solution but not a complete one otherwise I would have closed the issue. This is obviously important and still in debugging but I reported one of the problems and eliminated many others. `iCubGenova01` has been travelling a great deal and problems kept adding up, and he obviously not yet in a good shape. We have seen similar problems with the ethernet cable especially on the `iCubGenova04` but that has now been fixed. We will re-crimp all the ethernet connections and do further tests to insure stability (at least on the ethernet part). This sudden lack of feedback is problematic for all of us.

vtikha, 2015-11-13:
Work has been finished on the `pc104` to prevent the network drop mentioned above. @julijenv blocked the ethernet cable on the `pc104` and made sure that all crimps are tight on each pin. He noticed that some where loose which resulted i bad connections. I have tested again the demo on the `iCubGenova01` making extreme movements and playing with the connector and no drops happened. 

I am closing this issue as I am satisfied with this mornings performance but I would gently ask @jeljaik to conduct his tests on balancing and report if he notices any drops.

jeljaik, 2015-11-13:
Thank you so much @vtikha and @julijenv. I will do this first hour on morning :+1:

pattacini, 2015-11-19:
Unfortunately, the replacement of the network cable doesn't seem to fix the problem. Today we have experienced several network communication timeout of the `iKinGazeCtrl`.

@julijenv is afraid that somehow the `cfw2` board might be responsible for this.

pattacini, 2015-11-20:
Some new insights: it might be that the network overload is caused at the `CAN` level by lots of error messages we spotted out coming from the legs boards.

For the time being we have therefore disabled the legs.

pattacini, 2015-12-18:
I think this issue can be considered sorted out.
Hence closing for the meantime...

closed 2015-12-18 by pattacini