robotology/icub-main #240: Update documentation describing Fine Calibration

pattacini, 2015-11-18:
Parameters for [`fine calibrating`](http://wiki.icub.org/wiki/Manual#Three._Calibration) robot's parts have recently changed.
We should then update the documentation accordingly.

See https://github.com/robotology/icub-main/commit/82b56c63df0a99b8cd2246142c1d70e9fc69516d#commitcomment-14458176

randaz81, 2015-11-18:
We should disambiguate between `ETH` and `CAN` robots, because parameters names and procedures are different.

traversaro, 2016-08-26:
cc @gabrielenava @fjandrad @TeoKa @Yeshasvitvs .

traversaro, 2016-08-26:
Example commit : https://github.com/robotology/icub-main/commit/2edb509d1400d1aacbc7d897933dec9bd25e8b99 .

traversaro, 2017-08-22:
cc @giuliavezzani @claudiofantacci

claudiofantacci, 2017-08-22:
@giuliavezzani and I have recently fine calibrated the Black iCub due to server disruption and restoration.

We had some problem figuring out (thanks @traversaro) that the correct files to edit are the XMLs files in `robots-configuration/iCubGenova01/hardware/motorControl`, i.e. [line 35 for the torso](https://github.com/robotology/robots-configuration/blob/master/iCubGenova01/hardware/motorControl/icub_torso.xml#L35), [line 35 for the left arm](https://github.com/robotology/robots-configuration/blob/master/iCubGenova01/hardware/motorControl/icub_left_arm.xml#L37), [line 35 for the right arm](https://github.com/robotology/robots-configuration/blob/master/iCubGenova01/hardware/motorControl/icub_right_arm.xml#L35) and [line 35 for the head](https://github.com/robotology/robots-configuration/blob/0238f1661f053536689df7bf2ce0881b22131cdb/iCubGenova01/hardware/motorControl/icub_head.xml#L35). Not also that only the `Zeros` tag is present in these files.

I suggest to give room in wiki to this clarification.

pattacini, 2023-07-30:
The documentation shuld be aligned: https://icub-tech-iit.github.io/documentation/icub_robot_calibration/icub-robot-calibration-v2.x/?h=fine+calibration#legs-calibration.

closed 2023-07-30 by pattacini