robotology/icub-main #241: gravityCompensator doesn't launch on iCubSim

gregoire-pointeau, 2015-11-18:
Hello. I just updated yarp and icub-main, and when I try to launch `gravityCompensator` on the iCubSim,
`iCum_SIM` `simCartesianControl` `iKinCartesianSolver` `iKinGazeCtrl` and `wholeBodyDynamics` run without problem, but when launching `gravityCompensator` I have the following error:

||| configuring
||| no policy found
yarp: Port /gravityCompensator/head/client/rpc:o active at tcp://10.69.200.182:1
0206
yarp: Port /gravityCompensator/head/client/command:o active at tcp://10.69.200.1
82:10207
yarp: Port /gravityCompensator/head/client/state:i active at tcp://10.69.200.182
:10208
yarp: Port /gravityCompensator/head/client/stateExt:i active at tcp://10.69.200.
182:10209
[ERROR]Problem connecting to /icub/head/rpc:i, is the remote device available?
[ERROR]Problem connecting to /icub/head/command:i, is the remote device availabl
e?
[ERROR]Problem connecting to /icub/head/state:o from /gravityCompensator/head/cl
ient/state:i, is the remote device available?
[ERROR]*** Extended port /icub/head/stateExt:o was not found on the controlBoard
Wrapper I'm connecting to. Falling back to compatibility behaviour
[WARNING]Updating to newer yarp and the usage of controlBoardWrapper2 is suggest
ed***
yarpdev: ***ERROR*** driver <remote_controlboard> was found but could not open

I'm on windows 7, and I got the feeling that it comes from the way I configured my cmake, but it was working just last week before updating my system.

Thank you for your help.

vtikha, 2015-11-19:
It is trying to connect to `/icub/head/....` Are you running it with the correct parameters? Eg: `--robot icubSim`

kt10aan, 2015-11-19:
On my weekly-update test machine, ``gravityCompensator`` works fine with the simulator using the ``--robot icubSim``

pattacini, 2015-11-19:
'icubSim' really? Easy then.
That's my second oversight in a row; does it mean I need a vacation? :smirk:

gregoire-pointeau, 2015-11-19:
Yep, my bad, it seems that it was some kind of problem of iKinGaze that crashed and didn't restart correctly for an unknown reason.
Now it's working.

Problem solved.
Thanks !

closed 2015-11-19 by gregoire-pointeau