robotology/icub-main #244: Error during compiling icub-main

wainting, 2015-11-21:
Hi to all,

I'm trying to install icub-main on my pc following these guidelines: http://wiki.icub.org/wiki/Linux:Installation_from_sources , but after I set ccmake ../ , when i run make comes out this error
 
[  0%] Copying manager.py to /home/daniele/robsoft/icub-main/build/bin/.
[  0%] Built target python-scripts
[  1%] Copying icub-cluster scripts to /home/daniele/robsoft/icub-main/build/bin/.
[  1%] Built target cluster-scripts
[  2%] Built target iCubIDLClients
[  2%] Built target iCubDev
[  5%] Built target ctrlLib
[  7%] Built target skinDynLib
[  8%] Built target iKin
[ 10%] Built target iDyn
[ 16%] Built target learningMachine
[ 18%] Built target perceptiveModels
[ 18%] Built target actionPrimitives
[ 19%] Built target icubmod
[ 19%] Built target cartesiancontrollerclient
[ 20%] Built target cartesiancontrollerserver
[ 20%] Built target gazecontrollerclient
[ 21%] Built target debugInterfaceClient
[ 22%] Built target optimization
[ 23%] Built target ICUB_tinyxml
[ 26%] Built target robotInterface
[ 27%] Built target emotionInterface
[ 27%] Built target controlBoardDumper
[ 28%] Built target simpleClient
[ 28%] Built target testStereoMatch
[ 28%] Built target fingersTuner
[ 28%] Built target imuFilter
[ 29%] Built target ethLoaderLib
[ 29%] Built target ethLoader
[ 30%] Automoc for target QtICubSkinGuiPlugin
[ 30%] Built target QtICubSkinGuiPlugin_automoc
[ 34%] Built target QtICubSkinGuiPlugin
[ 34%] Automoc for target iCubSkinGui-qt
[ 34%] Built target iCubSkinGui-qt_automoc
[ 35%] Built target iCubSkinGui-qt
[ 36%] Automoc for target frameGrabberGui2-qt
[ 36%] Built target frameGrabberGui2-qt_automoc
[ 40%] Built target frameGrabberGui2-qt
[ 40%] Automoc for target skinManagerGui-qt
[ 40%] Built target skinManagerGui-qt_automoc
[ 44%] Built target skinManagerGui-qt
[ 48%] Built target iCubSkinGui-gtk
[ 48%] Built target frameGrabberGui2-gtk
[ 48%] Built target skinManagerGui-gtk
[ 49%] Built target joystickCtrl
[ 49%] Built target joystickCheck
[ 50%] Built target focusTool
[ 50%] Automoc for target iCubGui
[ 50%] Built target iCubGui_automoc
[ 55%] Built target iCubGui
[ 56%] Built target stereoCalib
[ 58%] Built target depth2kin
[ 59%] Built target simCartesianControl
Linking CXX executable ../../../bin/iCub_SIM
/usr/bin/ld: /usr/local/lib/libode.a(atomic.o): undefined reference to symbol 'pthread_mutexattr_init@@GLIBC_2.2.5'
/lib/x86_64-linux-gnu/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [bin/iCub_SIM] Error 1
make[1]: *** [src/simulators/iCubSimulation/CMakeFiles/iCub_SIM.dir/all] Error 2
make: *** [all] Error 2

Someone can help me? Thanks 

Daniele

barbalberto, 2015-11-27:
which OS are you using?
Did you install yarp on system folders?

wainting, 2015-11-27:
Hi Alberto,

Thanks for your answer , I 've solved this problem. Simply there was a conflict between different ODE versions, anyway I'm using ubuntu trusty.

closed 2015-11-27 by wainting